stack-of-tasks/pinocchio #747: Index types

gabrielebndn, 2019-03-25:
I feel the index types in Pinocchio are unnecessarily complicated.

Right now, [src/multibody/fwd.hpp](https://github.com/stack-of-tasks/pinocchio/blob/devel/src/multibody/fwd.hpp) defines `Index`, `JointIndex`, `GeomIndex`, `FrameIndex`, and `PairIndex`, all typedefs of `std::size_t`. This is fine, I guess. Although they are all the same, it is nice to denote clearly which index stands for what, especially for `JointIndex` vs `FrameIndex`, I think.

Then `Model` and `Data` define their own class-scoped typedefs, which are simply aliases of the above. The same does `Frame` with `JointIndex`, and `GeometryModel` with `PairIndex`. All these aliases are explicitly referred to within the code, making it simply heavier to read, and sometimes needlessly complicating function signatures.
Also, [LieGroupBase](https://github.com/stack-of-tasks/pinocchio/blob/devel/src/multibody/liegroup/liegroup-base.hpp) typedefs its own `Index` to `int`, while joint classes define it as`Eigen::DenseIndex`.

Do we really need all this granularity? Why the class-scoped typedefs? I think we can get rid of them. This would also go in the same direction as [what Eigen did](http://eigen.tuxfamily.org/index.php?title=3.3#Index_typedef).

Which raises the question of choosing the ideal index type. This might be related to the old issue #323.

jcarpent, 2019-03-26:
@gabrielebndn You're totally right. But unfortunately, it would require a long time and hard work to uniformize this part of the API.
If you are willing to do this operation, I will really appreciate.

So if you do so, then I will suggest using the std::size_t everywhere with a typedef in order to comply with the std::vector.
Another good solution is to rely on a signed type to detect anomalies when unrolling a vector from the tail to the head for instance.

gabrielebndn, 2019-03-26:
Eigen defaults to `std::ptrdiff_t`. If we were to go for a signed type, that would be the right way to do it, I think. But I would rather stick to `std::size_t`.
Also, Eigen makes it globally configurable through the macro `EIGEN_DEFAULT_INDEX_TYPE`. I like that.

jmirabel, 2019-03-26:
> Also, Eigen makes it globally configurable through the macro EIGEN_DEFAULT_INDEX_TYPE. I like that.

The idea may look nice but:
- I don't think it's a good idea to changes this. You would have to make sure all libraries using Eigen do the same.
- I don't know how tested it is. Eigen::Dynamic is set to -1 so you may have a hard time...

My advice on this is: keep it simple.

gabrielebndn, 2019-03-26:
> I don't think it's a good idea to changes this. You would have to make sure all libraries using Eigen do the same.

Not following you. I am definitely not suggesting to play with EIGEN_DEFAULT_INDEX_TYPE. I would just define a new macro called PINOCCHIO_DEFAULT_INDEX_TYPE, which would be e.g. `std::size_t` if not user-defined

> I don't know how tested it is

Agreed, it might be very complicated to test

> Eigen::Dynamic is set to -1 so you may have a hard time

This has nothing to do with what we are saying. That concerns `Matrix` template parameters `RowsAtCompileTime` and `ColsAtCompileTime`, which are `int` (see [here](https://eigen.tuxfamily.org/dox/group__TutorialMatrixClass.html))

jmirabel, 2019-03-26:
I wasn't clear then, sorry.

> I would just define a new macro called PINOCCHIO_DEFAULT_INDEX_TYPE, which would be e.g. std::size_t if not user-defined

Don't do that. Just choose a type (signed or not signed) and stick to it.

I think it is pointless to do that. It adds an untested features, which may add *hard to find* bugs for someone who takes the risk of using this (as I believe it is the case for Eigen).

jcarpent, 2019-03-31:
I will now close this issue. @gabrielebndn feel free to reopen it if needed.

closed 2019-03-31 by jcarpent