stack-of-tasks/pinocchio #748: Installation from source fails on master

sheim, 2019-03-26:
Hi, when building from source for release on ubuntu 16.04, I get the following error on the `master` branch (the `devel` branch builds fine though...):

```
In file included from /home/heim/Repositories/src_sot2/pinocchio/build/include/pinocchio/parsers/urdf.hpp:268:0,
                 from /home/heim/Repositories/src_sot2/pinocchio/unittest/value.cpp:23:
/home/heim/Repositories/src_sot2/pinocchio/build/include/pinocchio/parsers/urdf/geometry.hxx:294:51: error: ‘pinocchio::fcl::MeshLoaderPtr’ has not been declared
                                              fcl::MeshLoaderPtr& meshLoader,
                                                   ^
/home/heim/Repositories/src_sot2/pinocchio/build/include/pinocchio/parsers/urdf/geometry.hxx:375:34: error: ‘pinocchio::fcl::MeshLoaderPtr’ has not been declared
                             fcl::MeshLoaderPtr& meshLoader,
                                  ^
In file included from /home/heim/Repositories/src_sot2/pinocchio/build/include/pinocchio/parsers/urdf.hpp:268:0,
                 from /home/heim/Repositories/src_sot2/pinocchio/unittest/value.cpp:23:
/home/heim/Repositories/src_sot2/pinocchio/build/include/pinocchio/parsers/urdf/geometry.hxx:411:37: error: ‘pinocchio::fcl::MeshLoaderPtr’ has not been declared
                                fcl::MeshLoaderPtr meshLoader)
                                     ^
/home/heim/Repositories/src_sot2/pinocchio/build/include/pinocchio/parsers/urdf/geometry.hxx:429:37: error: ‘pinocchio::fcl::MeshLoaderPtr’ has not been declared
                                fcl::MeshLoaderPtr meshLoader)
```

jcarpent, 2019-03-26:
The problem has been solved as explained in #740. @sheim Thanks for this report anyway.

jcarpent, 2019-03-26:
@sheim I will provide a new release of the devel tomorrow. Hope it will be helpful for you too.

jcarpent, 2019-03-27:
The devel branch has been merged in the master one. Issue solved.

closed 2019-03-27 by jcarpent