stack-of-tasks/pinocchio #749: Mac pinocchio import failure

ddliugit, 2019-03-26:
I found the problem like the following when I tried to import the Pinocchio library:
I have installed the boost, I am wondering why this happened.

$ python
Python 2.7.16 (default, Mar  4 2019, 09:01:38) 
[GCC 4.2.1 Compatible Apple LLVM 10.0.0 (clang-1000.11.45.5)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import pinocchio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/site-packages/pinocchio/__init__.py", line 6, in <module>
    from pinocchio.robot_wrapper import RobotWrapper
  File "/usr/local/lib/python2.7/site-packages/pinocchio/robot_wrapper.py", line 5, in <module>
    from . import libpinocchio_pywrap as pin
ImportError: dlopen(/usr/local/lib/python2.7/site-packages/pinocchio/libpinocchio_pywrap.so, 2): Symbol not found: __ZN5boost6system6detail20generic_category_ncxEv
  Referenced from: /usr/local/opt/hpp-fcl/lib/libhpp-fcl.dylib
  Expected in: /usr/local/opt/boost/lib/libboost_system-mt.dylib
 in /usr/local/opt/hpp-fcl/lib/libhpp-fcl.dylib

jcarpent, 2019-03-27:
I think you simply  need to reinstall `hpp-fcl`:
`brew reinstall hpp-fcl`

ddliugit, 2019-03-27:
Same issue:

$brew reinstall hpp-fcl
==> Tapping brewsci/science
Cloning into '/usr/local/Homebrew/Library/Taps/brewsci/homebrew-science'...
remote: Enumerating objects: 439, done.
remote: Counting objects: 100% (439/439), done.
remote: Compressing objects: 100% (436/436), done.
remote: Total 439 (delta 1), reused 129 (delta 1), pack-reused 0
Receiving objects: 100% (439/439), 400.98 KiB | 2.52 MiB/s, done.
Resolving deltas: 100% (1/1), done.
Tapped 420 formulae (459 files, 1.2MB).
==> Reinstalling gepetto/gepetto/hpp-fcl 
==> Downloading https://github.com/humanoid-path-planner/hpp-fcl/releases/downlo
Already downloaded: /Users/dongdong/Library/Caches/Homebrew/downloads/6d8c8e604ca0a47222d1ff471ebaf21f0597701a7dd83742eb7e7e9338c8bf96--hpp-fcl-1.0.1.mojave.bottle.1.tar.gz
==> Pouring hpp-fcl-1.0.1.mojave.bottle.1.tar.gz
🍺  /usr/local/Cellar/hpp-fcl/1.0.1: 1,666 files, 13.9MB
10-18-70-186:~ dongdong$ python
Python 2.7.16 (default, Mar  4 2019, 09:01:38) 
[GCC 4.2.1 Compatible Apple LLVM 10.0.0 (clang-1000.11.45.5)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import pinocchio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/site-packages/pinocchio/__init__.py", line 6, in <module>
    from pinocchio.robot_wrapper import RobotWrapper
  File "/usr/local/lib/python2.7/site-packages/pinocchio/robot_wrapper.py", line 5, in <module>
    from . import libpinocchio_pywrap as pin
ImportError: dlopen(/usr/local/lib/python2.7/site-packages/pinocchio/libpinocchio_pywrap.so, 2): Symbol not found: __ZN5boost6system6detail20generic_category_ncxEv
  Referenced from: /usr/local/opt/hpp-fcl/lib/libhpp-fcl.dylib
  Expected in: /usr/local/opt/boost/lib/libboost_system-mt.dylib
 in /usr/local/opt/hpp-fcl/lib/libhpp-fcl.dylib

jcarpent, 2019-03-27:
brew reinstall pinocchio too

ddliugit, 2019-03-27:
Thanks. It worked well! @jcarpent

closed 2019-03-27 by ddliugit

jcarpent, 2019-03-27:
Your welcome!