stack-of-tasks/pinocchio #753: Using CodeGen utilities

francofusco, 2019-03-28:
Hello,
I successfully built Pinocchio and played a little with the algorithms, and I have few questions about the generation of code. Specifically, I would like to understand which is the intended procedure to generate and use a given algorithm.

I gave it a try myself, considering RNEA, and wrote the following:
```
pinocchio::CodeGenRNEA<double> rnea(model);
rnea.initLib();
rnea.loadLib();
```
After compiling and running, I can see the generated library in the same folder where I run the program. However, neither `CodeGenBase` nor `CodeGenRNEA` expose the function in any way. In addition, `evalFunction`/`evalJacobian` methods update only protected variables, without making the result of the computation directly accessible. Therefore my question is: how am I supposed to access the generated functionalities? I could load the library myself, but I guess this is not the intended way... Perhaps I completely misunderstood the usage, and the code is directly generated and called by "standard" algorithms behind the scenes?

jcarpent, 2019-03-29:
The first question I need you to answer is: why do you need code generation? Are the current algorithms not fast enough for your current setup?

jcarpent, 2019-03-29:
Concerning your current issue, it is true that the `CodeGenX` classes are not exposing their results in the current version. I will modify this behavior as soon as possible.
The current implementation has been done for benchmarking the code generation mostly but never used for a dedicated application. 
When is your deadline in terms of using this code generation features? According to your answer, I will try to do my best for meeting your requirement.

Finally, if you need another example on how to use code generation with Pinocchio, you can have a look at this file which is a simple but working unit test: 
https://github.com/stack-of-tasks/pinocchio/blob/904d7f9cfab86269487cd59ce9efd3f640fad8e1/unittest/cppadcg-algo.cpp#L27

jcarpent, 2019-03-29:
> Perhaps I completely misunderstood the usage, and the code is directly generated and called by "standard" algorithms behind the scenes?

The idea of the current CodeGen classes is to replace a direct call to the standard algorithms and to remove any notion of *Model* and *Data* structure. The compiler is making additionnal simplifications which allows to improve (a lot) the performances. See this paper https://hal.laas.fr/hal-01866228 for the benchmarks values.

francofusco, 2019-03-29:
Thanks for such a quick reply!

> why do you need code generation? Are the current algorithms not fast enough for your current setup?

I had already read the paper about Pinocchio and that is why I was really interested in the code generation feature. I have an idea for my work that involves Nonlinear Model Predictive Controls, and I will need to evaluate some kinematics/dynamics quantities (and their derivatives) many times. I do not really know how fast I need to get, but while looking for a complete library providing all the required tools I found Pinocchio and thought that it was worth giving it a try! So there is no specific need to use CodeGen right now, I just thought it was interesting and useful to improve performances.

> When is your deadline in terms of using this code generation features? According to your answer, I will try to do my best for meeting your requirement.

There is no rush, thank you! I already implemented two "workarounds". The first one consists in inheriting from `CodeGenRNEA`. Since the values I need are protected variables, I cheat by adding public methods which return them. This looks like the following (as a reference for anyone else needing a quick-and-dirty solution):
```c++
template<typename _Scalar>
struct CodeGenRNEAaccess : public CodeGenRNEA<_Scalar>
{
  typedef CodeGenRNEA<_Scalar> RNEA;
  typedef typename RNEA::VectorXs VectorXs;

  CodeGenRNEAaccess(
    const Model & model,
    const std::string& function_name = "rnea",
    const std::string& library_name = "cg_rnea_eval"
  ) : RNEA(model, function_name, library_name)
  { }

  inline VectorXs result() const { return res; }

protected:
  using RNEA::res;
};
```

The second way is a utility class that loads the library in the constructor and "replicates" the methods `evalFunction` and `evalJacobian`, but also giving the output directly as an additional parameter. This takes longer but I have finer control on library loading - since with this class there is no guarantee that the generated code will be "visible" at runtime.

I am looking forward an official "patch" but with these workarounds I am fine for now! Actually, if this helps in any way I can propose the patch myself by adding the required methods. As I see it, it is just a matter of adding few lines to each struct so it will not take long...

jcarpent, 2019-03-29:
At first glance, I would suggest you use the current API without code generation for getting your derivatives.
At the current stage of your developments, you do not need the code generation but we already have a lot of algorithms that implement the derivatives you need and with a very efficient implementation (you should expect to divide by two the computations without code generation).
You can even use the Python bindings for code prototyping.

Concerning your workaround, it is perfect. I will modify the API to make it compliant with your approach and your needs.

If you need to talk about your project offline, do not hesitate to send me a private email, and I will be happy to provide you some guidance.

francofusco, 2019-03-29:
>  I would suggest you use the current API without code generation for getting your derivatives

Sure, I was planning to make a first version without CodeGen utils and eventually to use them if I need a little boost.

> If you need to talk about your project offline, do not hesitate to send me a private email, and I will be happy to provide you some guidance.

I would be glad to! I will go on with my idea and write you in case of need, and as soon as something is up it would be nice to get some ideas to imprve it. So, thanks again for everything!

closed 2019-03-29 by francofusco

jcarpent, 2019-03-29:
@francofusco No problem. Good luck with your project!