stack-of-tasks/pinocchio #756: How to add ground in pinocchio?

littleggghost, 2019-04-02:
Hi, @cmastalli 
I have seen your new progress about quadruped through youtube https://www.youtube.com/watch?v=C0h-NLdMVuc. 
Here I have a question, considering you use gepetto in this video, then is that done by pinocchio?
It has been a long time that I tried to add ground collision in pinocchio.
Thanks.

cmastalli, 2019-04-02:
As you said, I displayed the planned motion using Gepetto viewer (GV). Planned motions are generated by an optimal control solvers that I've working on. And Pinocchio is used to model the robot's dynamics and its derivatives.

On the other hand, we use RobotWrapper from displaying the robot posture. Inside it wraps some GV functionalities. And you can't add other objects with it.

I would suggest to learn more about GV. Unfortunatelu there isn't a tutorial yet for that. Nevertheless it's not too difficult to understand it.

In that video I used the following Python code for adding the floor:

```python
window_id = robot.viewer.gui.getWindowID('python-pinocchio')
robot.viewer.gui.addFloor('hpp-gui/floor')
robot.viewer.gui.setScale('hpp-gui/floor', [0.5, 0.5, 0.5])
robot.viewer.gui.setColor('hpp-gui/floor', [0.7, 0.7, 0.7, 1.])
robot.viewer.gui.setLightingMode('hpp-gui/floor', 'OFF')
```

and this one for setting the white background

```python
robot.viewer.gui.setBackgroundColor1(window_id, [1., 1., 1., 1.])
robot.viewer.gui.setBackgroundColor2(window_id, [1., 1., 1., 1.])
```
Feel free to re-open it, if you have further questions.

closed 2019-04-02 by cmastalli

littleggghost, 2019-04-03:
Hello @cmastalli 
I know how to add the floor in "visual", but here my focus is on the ground collision with foot. I really want to know how to set the collision property of floor.
Thanks in advance!

cmastalli, 2019-04-03:
I'm not the best person to ask this, you might want to ask @jmirabel.

littleggghost, 2019-04-03:
Hi, @jmirabel 
What is your advice about ground collision?

jmirabel, 2019-04-04:
You can represent the floor with a box of the right shape for collision detection. But it won't make your example work. You have to handle yourself the different phases (free drop and contact with ground) with a suitable contact model.

The practical exercise "Snap your finger" in the doc may help you.

littleggghost, 2019-04-08:
Hi, @jmirabel  Sorry for my late reply!
What I am concern here is only about the collision. The walk phase is not needed, but I want a stick or a box or a boll just contact with **ground box**.
About the exercise you mentioned, I have asked an issue before: https://github.com/stack-of-tasks/pinocchio/issues/731. 
And, how many **collisionPairs** should be added? Because the **ground box** will be huge and don't know where the collision will happen.
Hope you can explain it in more detail!

littleggghost, 2019-04-08:
Hello @cmastalli 
I really think this issue should be re-opened.