stack-of-tasks/pinocchio #757: Gepetto support broken

jviereck, 2019-04-03:
Hi folks,

building current master, visualizing the robot using gepetto does not work properly for me on macOS. The mesh_scale for the visual nodes are all 1.

I think this comes from commit ccf91776cde0aba005f75e8e907faee6d6912b1b:

- There is a `BUILD_WITH_HPP_FCL_SUPPORT` variable defined, but above there is an optional variable `BUILD_WITH_COLLISION_SUPPORT`
- When I fix this and enable building with HPP_FCL, the visualization work again

Does loading the URDF + mesh files require HPP_FCL (I am not 100% sure but I think the URDF loader requires HPP_FCL to load the scaling).

From the perspective of a user, it would be nice to get an error / warning message when loading an URDF with meshes and properly loading the mesh (e.g. scaling information) is not supported by the current installed pinocchio library.

Best,
julian

jcarpent, 2019-04-03:
This seems to be fixed on the devel branch by @jmirabel. Can you confirm it?

jviereck, 2019-04-03:
@jcarpent Two things:

1) building the current devel branch gives me the error shown below
2) if you know that master is broken, why wasn't the fix ported back to master already? My understanding is that `master` is supposed to be stable. Spending 4 hours to track down a bug and getting told it is already fixed on devel is frustrating.

```
[ 54%] Building CXX object unittest/CMakeFiles/model.dir/model.cpp.o
cd /Users/jviereck/dev/tools/pinocchio/build26/unittest && /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++  -DPINOCCHIO_URDFDOM_TYPEDEF_SHARED_PTR -DPINOCCHIO_URDFDOM_USE_STD_SHARED_PTR -DPINOCCHIO_WITH_URDFDOM -I/Users/jviereck/dev/tools/pinocchio/build26 -I/Users/jviereck/dev/tools/pinocchio/build26/include -I/Users/jviereck/dev/tools/pinocchio/include -I/Users/jviereck/dev/tools/pinocchio/unittest -isystem /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/python2.7 -isystem /usr/local/include -isystem /usr/local/Cellar/eigen/3.3.7/include/eigen3 -isystem /usr/local/Cellar/urdfdom/1.0.3/include -isystem /usr/local/Cellar/urdfdom_headers/1.0.3/include -isystem /usr/local/Cellar/console_bridge/0.4.3/include  -pedantic -Wno-long-long -Wall -Wextra -Wcast-align -Wcast-qual -Wformat -Wwrite-strings -Wconversion  -O3 -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk    -DBOOST_TEST_DYN_LINK -DBOOST_TEST_MODULE=modelTest -o CMakeFiles/model.dir/model.cpp.o -c /Users/jviereck/dev/tools/pinocchio/unittest/model.cpp
In file included from /Users/jviereck/dev/tools/pinocchio/unittest/model.cpp:6:
In file included from /Users/jviereck/dev/tools/pinocchio/build26/include/pinocchio/algorithm/model.hpp:54:
/Users/jviereck/dev/tools/pinocchio/build26/include/pinocchio/algorithm/model.hxx:205:17: error: no member
      named 'addCollisionPair' in 'pinocchio::GeometryModel'
      geomModel.addCollisionPair (CollisionPair (go1, go2));
      ~~~~~~~~~ ^
/Users/jviereck/dev/tools/pinocchio/build26/include/pinocchio/algorithm/model.hxx:211:17: error: no member
      named 'addCollisionPair' in 'pinocchio::GeometryModel'
      geomModel.addCollisionPair (CollisionPair (go1, go2));
      ~~~~~~~~~ ^
/Users/jviereck/dev/tools/pinocchio/build26/include/pinocchio/algorithm/model.hxx:221:21: error: no member
      named 'addCollisionPair' in 'pinocchio::GeometryModel'
          geomModel.addCollisionPair(CollisionPair(go1, go2));
          ~~~~~~~~~ ^
```

jcarpent, 2019-04-03:
To be honest, the things are that currently, I'm trying to be at best as I can available for Pinocchio but it is not always possible mainly due to professional reasons. I'm really sorry that you spent so many times on this error which has been introduced by me.

Second, for the first point, can you confirm that you are not compiling with the COLLISION SUPPORT?

jcarpent, 2019-04-03:
I have found the error number 1, which has been introduced by @jmirabel in a recent commit.

jcarpent, 2019-04-03:
@jviereck I've proposed a fix in #758. Hope it will be useful for you tonight!

jviereck, 2019-04-04:
Hi @jcarpent, I am sorry if my previous comment in point 2) was not constructive and upset you. I appreciate the effort you put into this library!

Thanks for providing the PR #758. As noted in the PR, there is still a problem when compiling with COLLISION support enabled. In addition, when the COLLISION is not enabled, the mesh visualization is still wrong.

jmirabel, 2019-04-04:
> This seems to be fixed on the devel branch by @jmirabel. Can you confirm it?

Yes, I confirm.

gabrielebndn, 2019-04-04:
I think the fix should be cherry-picked to master. Either this or a new release should be made. It is bad to have build failures on the master branch.

jcarpent, 2019-04-04:
@gabrielebndn We cannot go faster than the music, I will try to fix correctly all the remaining bugs revealed by @jviereck first.

jmirabel, 2019-04-04:
@gabrielebndn do you talk about Pinocchio or GV ? There is no build failure on any master branch. There is a build failure on devel branch of Pinocchio. And there is a bug at execution time for GV on the master branch.

gabrielebndn, 2019-04-04:
@jmirabel, @jcarpent, sorry, bad wording on my part. I meant the original problem found by @jviereck, which was fixed on the devel branch but not on the master branch. It is not a build failure, but still, if I understand well, the whole part of Pinocchio related to HPP-FCL is not building anymore on the master branch. This problem manifested itself as an execution bug for GV, but it will be a build failure for any C++ code attempting to use HPP-FCL functionalities. I think the fix provided by commit 8a02e77a6af3ed386d8fd5c64cd197bc64a521b8 could be easily cherry-picked. This unless @jcarpent is planning a new release very soon, but apparently he does not, and I understand that. This is what I meant, sorry for the confusion.

jmirabel, 2019-04-04:
> > This seems to be fixed on the devel branch by @jmirabel. Can you confirm it?
> 
> Yes, I confirm.

I misread something. I think the bug here has nothing to do with GV:
> The mesh_scale for the visual nodes are all 1.

jcarpent, 2019-04-04:
@jviereck Anyway, we need feedbacks as yours in order to make Pinocchio stable and robust. Next time, it will be even faster if you directly post a new issue here directly.

@gabrielebndn I will make a new release fix very soon indeed, to stabilize everything.

@nim65s Will you be available for publishing the next release in robotpkg?

jcarpent, 2019-04-04:
Solved by #758.

closed 2019-04-04 by jcarpent

jcarpent, 2019-04-05:
The new release with fixes is now out.