stack-of-tasks/pinocchio #762: cmake: configure error on archlinux / cmake 3.14

dicta, 2019-04-10:
Pinocchio version: tag v2.1.2 (dbcfbcb283eea486e1f5279c5e40cecdb6e87fad)
CMake version: 3.14.0

The following error is generated when trying to run cmake to configure a build:

```
-- eigenpy >= 1.4.4 is required.
-- Checking for module 'eigenpy >= 1.4.4'
--   Found eigenpy , version 1.5.0
-- Pkg-config module eigenpy v1.5.0 has been detected with success.
CMake Error: failed to create symbolic link '/home/dicta/repos/pinocchio.git/build/include/pinocchio/bindings/python/multibody/fcl/mesh-loader.hpp': no such file or directory
-- Doxygen rendering: using MathJax backend
-- Configuring done
```

Reproduction steps:
```
git clone https://github.com/stack-of-tasks/pinocchio
cd pinocchio
git checkout v2.1.2
git submodule init --update
mkdir build
cd build
cmake ..
```

I believe this is caused by the target directory not existing in the SYMLINK_AND_INSTALL_HEADERS() macro. The following one-line patch appears to resolve things for me:

```
diff --git a/bindings/python/CMakeLists.txt b/bindings/python/CMakeLists.txt
index 69c175b6..657b7bb0 100644
--- a/bindings/python/CMakeLists.txt
+++ b/bindings/python/CMakeLists.txt
@@ -8,6 +8,7 @@ MACRO(SYMLINK_AND_INSTALL_HEADERS HEADERS SUBPATH)
   FOREACH(header ${HEADERS})
     GET_FILENAME_COMPONENT(headerName ${header} NAME)
     GET_FILENAME_COMPONENT(headerPath ${header} PATH)
+    file(MAKE_DIRECTORY ${${PROJECT_NAME}_BINARY_DIR}/include/${PROJECT_NAME}/${SUBPATH}/${headerPath})
     EXECUTE_PROCESS(COMMAND ${CMAKE_COMMAND} -E ${LINK}
       ${CMAKE_CURRENT_SOURCE_DIR}/${header}
       ${${PROJECT_NAME}_BINARY_DIR}/include/${PROJECT_NAME}/${SUBPATH}/${header})
```

jcarpent, 2019-04-11:
The problem has been solved in #763. A file was not ignored while it should have been.
@dicta Thanks a lot for this report.

I have then opened a PR on the devel branch. You will then need to sync on it in order to recover a compiling version of Pinocchio.

nim65s, 2019-04-11:
Hi @dicta

Just for information, Pinocchio is available on ArchLinux both on [robotpkg](http://robotpkg.openrobots.org/install.html) and on [AUR](https://aur.archlinux.org/packages/pinocchio/). The first one is beeing upadted to v2.1.2, and the second one is not really used for now, so it is lagging behind, but if we get more interest from Arch users I will gladly improve AUR support.

jcarpent, 2019-04-11:
Solved by #763

closed 2019-04-11 by jcarpent