stack-of-tasks/pinocchio #764: Strange behavior of geom.removeCollisonPair

nmansard, 2019-04-11:
I don't understand what happens here:

p=gmodel.collisionPairs[1]  
print(p)    # display    collision pair (4,0)
gmodel.removeCollisionPair(gmodel.collisionPairs[0])
print(p)   # display collision pair (6,0)
print( gmodel.collisionPairs[0]) # display collision pair (4,0)
print(gmodel.collisionPairs[1]) # display collision pair (6,0)

Is p a fake index on the collision-pair array?

jmirabel, 2019-04-11:
Did you check that pair (4,0) is not duplicated in the beginning ?

nmansard, 2019-04-11:
It is not (the first is (0,3)). Even if it was, I don't see how this would explain the behavior.

gabrielebndn, 2019-04-12:
The problem is in the way the bindings of `collisionPairs` are implemented. When you do
```
p=gmodel.collisionPairs[1]
```
`p` is a *pointer* to the location of `gmodel.collisionPairs[1]`.

Then, you do
```
gmodel.removeCollisionPair(gmodel.collisionPairs[0])
```
Here, you are removing element zero from `gmodel.collisionPairs`. `collisionPairs` is a `std::vector`. Internally, it stores its data in an array. To execute this operation, it shifts all elements of the array back one position.

```
print(p) # display collision pair (6,0)
```
Here, `p` is still pointing at element 1 of the array. This position is now filled by what used to be element 2, before `removeCollisionPair` was called. In this case, `(6,0)`

```
print( gmodel.collisionPairs[0]) # display collision pair (4,0)
```
Element zero now holds what used to be element 1, before `removeCollisionPair` was called, i.e. `(4,0)`.

```
print(gmodel.collisionPairs[1]) # display collision pair (6,0)
```
Element 1 now holds what used to be element 2, before `removeCollisionPair` was called, i.e. `(6,0)`.

It gets worse: after a certain number of removals, the `std::vector` could decide to completely reallocate the array in order to free some memory. In this case, any previously-taken pointer would be pointing at garbage.

To make this clearer: what you are doing in Python is equivalent to the following C++ code
```
CollisionPair & p = gmodel.collisionPairs[1] // notice the reference
cout << p << endl; // display collision pair (4,0)
gmodel.removeCollisionPair(gmodel.collisionPairs[0])
cout << p << endl; // display collision pair (6,0)
cout << gmodel.collisionPairs[0] << endl; // display collision pair (4,0)
cout << gmodel.collisionPairs[1]) << endl; // display collision pair (6,0)
```
while what you want is
```
CollisionPair p = gmodel.collisionPairs[1] // WITHOUT reference - makes a copy
cout << p << endl; // display collision pair (4,0)
gmodel.removeCollisionPair(gmodel.collisionPairs[0])
cout << p << endl; // STILL display collision pair (4,0)
cout << gmodel.collisionPairs[0] << endl; // display collision pair (4,0)
cout << gmodel.collisionPairs[1]) << endl; // display collision pair (6,0)
```

I don't think there is an easy fix to this unwanted behavior. The return-by-pointer policy is desirable, in order to be able to easily write inside the array. In order to avoid problems in Python, you should always *explicitly* make a copy, like this
```
p=pin.CollisionPair(gmodel.collisionPairs[1].first,gmodel.collisionPairs[1].second)
print(p) # display collision pair (4,0)
gmodel.removeCollisionPair(gmodel.collisionPairs[0])
print(p) # display collision pair (4,0)
print( gmodel.collisionPairs[0]) # display collision pair (4,0)
print(gmodel.collisionPairs[1]) # display collision pair (6,0)
```
I will make `CollisionPair`copyable so you can replace the first line with
````
p=gmodel.collisionPairs[1].copy()
```

gabrielebndn, 2019-04-25:
So, it was decided in #782 to maintain the current behavior.
My recommendation is to always explicitly make a copy when retrieving collision pairs, using the `copy()` method introduced by #769, as in
```
p=gmodel.collisionPairs[1].copy()
```
@jcarpent can we close this issue?

closed 2019-04-29 by jcarpent