stack-of-tasks/pinocchio #765: Release version 2.1.2 on robotpkg

jviereck, 2019-04-11:
hi folks,

the current version (Date 11 April 2019) of pinocchio at openrobots

  http://robotpkg.openrobots.org/robotpkg/math/pinocchio/index.html

is 2.1.0. Is there a chance to release version 2.1.2 there?

Best,
Julian

\cc @nim65s

nim65s, 2019-04-11:
Hi @jviereck 

I'm currently on on it, the binary will be available soon :)

jcarpent, 2019-04-12:
@nim65s it should be nice if you can provide a clear deadline for this release, in such a way everybody is aware of it.

jviereck, 2019-04-12:
Could it be that the current robotpkg-pinocchio package in version 2.1.0 is also broken? When I try to install it the python files are not installed. When I look at the content of the package, no python files are listed :/

```
$ dpkg-query -L robotpkg-pinocchio 
/.
/opt
/opt/openrobots
/opt/openrobots/share
/opt/openrobots/share/doc
/opt/openrobots/share/doc/debian
/opt/openrobots/share/doc/debian/pinocchio
/opt/openrobots/share/doc/debian/pinocchio/+BUILD_VERSION
/opt/openrobots/share/doc/debian/pinocchio/+DESC
/opt/openrobots/share/doc/debian/pinocchio/+SIZE_PKG
/opt/openrobots/share/doc/debian/pinocchio/control
/opt/openrobots/share/doc/debian/pinocchio/+CONTENTS
/opt/openrobots/share/doc/debian/pinocchio/+BUILD_INFO
/opt/openrobots/share/doc/debian/pinocchio/+SIZE_ALL
/opt/openrobots/share/doc/debian/pinocchio/+COMMENT
/opt/openrobots/lib
/opt/openrobots/lib/pkgconfig
/opt/openrobots/lib/pkgconfig/pinocchio.pc
/opt/openrobots/lib/libpinocchio.so
/opt/openrobots/include
/opt/openrobots/include/pinocchio
/opt/openrobots/include/pinocchio/multibody
/opt/openrobots/include/pinocchio/multibody/frame.hpp
/opt/openrobots/include/pinocchio/multibody/model.hxx
/opt/openrobots/include/pinocchio/multibody/data.hpp
/opt/openrobots/include/pinocchio/multibody/fcl.hpp
/opt/openrobots/include/pinocchio/multibody/joint
/opt/openrobots/include/pinocchio/multibody/joint/joint-base.hpp
/opt/openrobots/include/pinocchio/multibody/joint/joint-spherical-ZYX.hpp
/opt/openrobots/include/pinocchio/multibody/joint/joint-composite.hxx
/opt/openrobots/include/pinocchio/multibody/joint/joint-translation.hpp
/opt/openrobots/include/pinocchio/multibody/joint/joint-generic.hpp
/opt/openrobots/include/pinocchio/multibody/joint/joint-revolute-unbounded.hpp
/opt/openrobots/include/pinocchio/multibody/joint/joint-composite.hpp
/opt/openrobots/include/pinocchio/multibody/joint/joint-prismatic.hpp
/opt/openrobots/include/pinocchio/multibody/joint/joint-revolute.hpp
/opt/openrobots/include/pinocchio/multibody/joint/joint-planar.hpp
/opt/openrobots/include/pinocchio/multibody/joint/joint-basic-visitors.hpp
/opt/openrobots/include/pinocchio/multibody/joint/joint-collection.hpp
/opt/openrobots/include/pinocchio/multibody/joint/joint-free-flyer.hpp
/opt/openrobots/include/pinocchio/multibody/joint/joint-revolute-unaligned.hpp
/opt/openrobots/include/pinocchio/multibody/joint/joint-prismatic-unaligned.hpp
/opt/openrobots/include/pinocchio/multibody/joint/joint-spherical.hpp
/opt/openrobots/include/pinocchio/multibody/joint/fwd.hpp
/opt/openrobots/include/pinocchio/multibody/joint/joint-basic-visitors.hxx
/opt/openrobots/include/pinocchio/multibody/fcl.hxx
/opt/openrobots/include/pinocchio/multibody/data.hxx
/opt/openrobots/include/pinocchio/multibody/force-set.hpp
/opt/openrobots/include/pinocchio/multibody/constraint.hpp
/opt/openrobots/include/pinocchio/multibody/liegroup
/opt/openrobots/include/pinocchio/multibody/liegroup/vector-space.hpp
/opt/openrobots/include/pinocchio/multibody/liegroup/liegroup-variant.hpp
/opt/openrobots/include/pinocchio/multibody/liegroup/liegroup-variant-visitors.hpp
/opt/openrobots/include/pinocchio/multibody/liegroup/liegroup.hpp
/opt/openrobots/include/pinocchio/multibody/liegroup/liegroup-base.hxx
/opt/openrobots/include/pinocchio/multibody/liegroup/cartesian-product.hpp
/opt/openrobots/include/pinocchio/multibody/liegroup/liegroup-variant-visitors.hxx
/opt/openrobots/include/pinocchio/multibody/liegroup/special-orthogonal.hpp
/opt/openrobots/include/pinocchio/multibody/liegroup/liegroup-generic.hpp
/opt/openrobots/include/pinocchio/multibody/liegroup/liegroup-base.hpp
/opt/openrobots/include/pinocchio/multibody/liegroup/liegroup-collection.hpp
/opt/openrobots/include/pinocchio/multibody/liegroup/liegroup-algo.hpp
/opt/openrobots/include/pinocchio/multibody/liegroup/cartesian-product-variant.hpp
/opt/openrobots/include/pinocchio/multibody/liegroup/liegroup-algo.hxx
/opt/openrobots/include/pinocchio/multibody/liegroup/special-euclidean.hpp
/opt/openrobots/include/pinocchio/multibody/liegroup/fwd.hpp
[... 146 lines cut ...]
/opt/openrobots/include/pinocchio/algorithm/compute-all-terms.hpp
/opt/openrobots/include/pinocchio/algorithm/kinematics-derivatives.hxx
/opt/openrobots/include/pinocchio/algorithm/contact-dynamics.hxx
/opt/openrobots/include/pinocchio/algorithm/crba.hxx
/opt/openrobots/include/pinocchio/algorithm/centroidal.hpp
/opt/openrobots/include/pinocchio/algorithm/geometry.hpp
/opt/openrobots/include/pinocchio/algorithm/rnea.hpp
/opt/openrobots/include/pinocchio/algorithm/jacobian.hxx
/opt/openrobots/include/pinocchio/algorithm/geometry.hxx
/opt/openrobots/include/pinocchio/algorithm/default-check.hpp
/opt/openrobots/include/pinocchio/algorithm/frames.hxx
/opt/openrobots/include/pinocchio/deprecation.hpp
/opt/openrobots/include/pinocchio/deprecated-macros.hpp
/opt/openrobots/include/pinocchio/codegen
/opt/openrobots/include/pinocchio/codegen/code-generator-base.hpp
/opt/openrobots/include/pinocchio/codegen/code-generator-algo.hpp
/opt/openrobots/include/pinocchio/deprecated.hpp
/opt/openrobots/include/pinocchio/deprecated-namespaces.hpp
/opt/openrobots/include/pinocchio/fwd.hpp
/opt/openrobots/include/pinocchio/macros.hpp
/usr
/usr/share
/usr/share/doc
/usr/share/doc/robotpkg-pinocchio
```

olivier-stasse, 2019-04-12:
@jviereck The python files are installed in a different package 
```
robotpkg-py27-pinocchio
```
because they are tested against several version of python.
We are preparing the switch to python 3 which will be mandatory in 2020.
All our binaries packages are trying to follow this policy.
So if you wish you can use
```
robotpkg-py35-pinocchio
```

jviereck, 2019-04-15:
Thanks a lot @olivier-stasse !

nim65s, 2019-04-16:
Done

closed 2019-04-16 by nim65s