stack-of-tasks/pinocchio #773: Compute jacobians in local world oriented frame

jviereck, 2019-04-15:
Hi folks,

at this point the jacobians can be computed in the world or local frame. I propose to add another reference frame that is the local_cartesian_oriented frame. This is useful when dealing with point contacts.

The code to compute this boils down to:

```
        jac = se3.getFrameJacobian(self.model, self.data, index, se3.ReferenceFrame.LOCAL)
        world_R_joint = se3.SE3(self.data.oMf[index].rotation, zero(3))
        return world_R_joint.action.dot(jac)
```

As this is a convenient method to have, I would like to add it to the pinocchio se3/RobotWrapper Jacobian's C++ API but would like to get feedback of adding a PR with this feature has changes to getting accepted.

Best,
Julian

jcarpent, 2019-04-16:
This feature is mainly worthy when you want to track a velocity field expressed at the origin of the moving frame, but with quantities expressed in the world axis.
It can then be worthy to have it, but it will also means that we will lose the fondamental relations that:
Jworld_body_i = oMi.act(Jlocal_body_i)
Then, the documentation has to be pretty clear on this point.

gabrielebndn, 2019-04-16:
I would love to see more support for the traditional Cartesian way of understanding frames.

jcarpent, 2019-04-18:
@jviereck Are you still willing to implement this feature?

jviereck, 2019-04-18:
@jcarpent, if someone else jumps on it, I would not mind. Otherwise I want to make a first few steps on Friday.

jcarpent, 2019-04-18:
From my side, I may help by reviewing the code and providing some hints if needed.
Maybe @gabrielebndn would like to do it?

gabrielebndn, 2019-04-18:
I like the idea and I approve of it fully, but I am quite busy with a lot of stuff at the moment :)
I have just a few recommendations:
- if you do it for the frames, do not forget doing it for the joints too
- I think the best way to do it would be adding another element to the `enum` called `ReferenceFrame` and take the new value into account inside the existing functions

jcarpent, 2019-05-01:
Solved by #781

PR #781 (2019-05-01): Add support for LOCAL_CARTESIAN_ORIENTED in jacobian computation
This is to fix #773. 

This is NOT done yet but there are a few points I thought it's better to get feedback on earlier then later.

I haven't tested the code yet. The code compute the frame orientation I've copied from some other internal code we have, but I don't know if the jointJacobian code is working as expected.

A few things:
- Does the name `LOCAL_CARTESIAN_ORIENTED` make sense?
- I have to use a temporary `J_tmp` to avoid allocations (this was a problem we ran into with tsid). Though I am not 100% sure yet if using a lo [... cut]
closed 2019-05-01 by jcarpent