stack-of-tasks/pinocchio #774: Broken 2.1.2 binaries on robotpkg?

gabrielebndn, 2019-04-16:
Robotpkg-pinocchio updated itself to 2.1.2. Now the Python bindings do not work
If I just do:
```
import pinocchio
```
this is the error I get
```
ImportError                               Traceback (most recent call last)
<ipython-input-1-6ede4ff3d689> in <module>()
----> 1 import pinocchio as pin

/opt/openrobots/lib/python2.7/site-packages/pinocchio/__init__.py in <module>()
      4 
      5 import numpy as np
----> 6 from .robot_wrapper import RobotWrapper
      7 from libpinocchio_pywrap import __version__
      8 

/opt/openrobots/lib/python2.7/site-packages/pinocchio/robot_wrapper.py in <module>()
      3 #
      4 
----> 5 from . import libpinocchio_pywrap as pin
      6 from . import utils
      7 from .deprecation import deprecated

ImportError: /opt/openrobots/lib/python2.7/site-packages/pinocchio/libpinocchio_pywrap.so: undefined symbol: _ZN9pinocchio6python9exposeFCLEv
```

What I have tried:
```
sudo apt-get install --reinstall robotpkg-py27-pinocchio
sudo apt-get install --reinstall robotpkg-hpp-fcl
sudo apt-get install --reinstall robotpkg-py27-pinocchio
```
to no avail

gabrielebndn, 2019-04-16:
I am not the only one experiencing this. I know @Fannyris at least in the same situation. @nim65s clues on this?

proyan, 2019-04-16:
By the way, just for info, c++filt is a useful tool to decipher the jumbled symbols

`c++filt _ZN9pinocchio6python9exposeFCLEv`
`pinocchio::python::exposeFCL()`

jcarpent, 2019-04-16:
This is because by default, Pinocchio is not looking for FCL when generating the makefile. I think @nim65s did not force this on robotipkg.

gabrielebndn, 2019-04-16:
> By the way, just for info, c++filt is a useful tool to decipher the jumbled symbols
`c++filt _ZN9pinocchio6python9exposeFCLEv`
`pinocchio::python::exposeFCL()`

Well, that wasn't so difficult to decipher visually, but thanks for the suggestion ^^

gabrielebndn, 2019-04-16:
> This is because by default, Pinocchio is not looking for FCL when generating the makefile. I think @nim65s did not force this on robotpkg.

@jcarpent ou mean there is no way for me to fix it other than switching to installation from source?

jcarpent, 2019-04-16:
In fact, I do not understand why this symbol is missing while it is not known at compile time because the support of FCL has been deactivated.
This should be investigated deeper.

@gabrielebndn I think you have to switch a compile from source version while @nim65s is fixing the bug. Sorry for that.

gabrielebndn, 2019-04-16:
Ok, not a big deal. Thanks!

nim65s, 2019-04-16:
Fixed. A revision is on its way.

nim65s, 2019-04-16:
An update is available

closed 2019-04-16 by nim65s