stack-of-tasks/pinocchio #776: Questionable hard-coded joint limit value in urdf-parsers for "continous" joints.

markusgft, 2019-04-17:
https://github.com/stack-of-tasks/pinocchio/blob/43e43293033b7f0b239869241c6cf780917164cf/src/parsers/urdf/model.hxx#L312

Dear all,
why is the joint limit value set to 1.01 ?

jcarpent, 2019-04-17:
There is no good reason for that. It just to have something greater than one. Because continuous revolute joints do not have bounds.

cmastalli, 2019-04-17:
I would expect to have the greatest value that it's possible, e.g.:

```c++
const ConfigVector_t::Scalar u = numeric_limits<double>::max()
```

jcarpent, 2019-04-17:
@cmastalli This is indeed one solution.

jmirabel, 2019-04-17:
If I remember well, I think I am the one who set it to 1.01. The motivations were:
- to help tracking errors. Since then, we added a isNormalized, which is the proper way of tracking an error.
- to easily (i.e. without visitor) know whether a joint is bounded or not. If you put the max double, then you lose the information that this value is bounded.

The bounds are on the configuration vector parameters, and not on the rotation angle. The ideal bound ought to be 1, and not max double. But then, there will be numerical issues... I admit 1.01 is too big.

gabrielebndn, 2019-04-17:
I don't think max double makes sense. A continuous joint is never expected to exceed `. It should be as close as possible to 1, but not exactly 1, just slightly greater.

Also, putting max double will likely create problems to `randomConfiguration(model)`, which uses the limits

gabrielebndn, 2019-04-17:
> Also, putting max double will likely create problems to randomConfiguration(model), which uses the limits

Actually, I've just read the code computing a random configuration for an unbounded joint. It seems the bounds are ignored. You can find it in [src/multibody/liegroup/special-orthogonal.hpp](https://github.com/stack-of-tasks/pinocchio/blob/devel/src/multibody/liegroup/special-orthogonal.hpp). Method `randomConfiguration_impl` does not employ the bounds. Is this right? I know it does not make a lot of sense to put bounds on "unbounded" joints, but are we OK just ignoring them?

gabrielebndn, 2019-04-17:
Also, I've noticed an incoherence: when *parsing* an URDF model, the default limit is 1.01. But when you use `addJoint`, *without* specifying the limits, then the limits are indeed max double.

(also, `addJoint` with limits is not exposed in Python, I'll fix this)

cmastalli, 2019-04-17:
@jmirabel thanks for the explanation.

Do you think the following code makes sense?

```c++
const ConfigVector_t::Scalar u = 1. + numeric_limits<double>::min()
```

gabrielebndn, 2019-04-17:
> @jmirabel thanks for the explanation.
> 
> Do you think the following code makes sense?
> 
> ```c++
> const ConfigVector_t::Scalar u = 1. + numeric_limits<double>::min()
> ```

No. That expression is equal to 1. The resolution of `double` around 1 is not enough.

```
#include <limits>
#include <iostream>

using std::cout;
using std::endl;

int main() {
    double u1 = 1.0;
    double u2 = 1 + std::numeric_limits<double>::min();

    cout << u1 << endl;
    cout << u2 << endl;
    cout << (u1-u2) << endl;
    cout << (u1==u2) << endl;
}
```

gabrielebndn, 2019-04-17:
The way to obtain a floating-point value that is *just smaller* or *just larger* than another one is using [nextafter](http://www.cplusplus.com/reference/cmath/nextafter/).

Just tested it
```
#include <cmath>
#include <iostream>

using std::cout;
using std::endl;

int main() {
    double u1 = 1.;
    double u2 = nextafter(1.,2.);

    cout << "------" << endl;
    cout << u1 << endl;
    cout << u2 << endl;
    cout << (u2-u1) << endl;
    cout << (u2>u1) << endl;
    cout << (u1==u2) << endl;

    cout << "------" << endl;
    double u3 = -1.;
    double u4 = nextafter(-1.,-2.);
    cout << u3 << endl;
    cout << u4 << endl;
    cout << (u3-u4) << endl;
    cout << (u3>u4) << endl;
    cout << (u3==u4) << endl;
}
```
On my PC, it seems to work even for types other than `double`, but I am not sure it is a good idea using it, it might reduce portability and/or complicate the code.

jmirabel, 2019-04-18:
@gabrielebndn `randomConfiguration(model)` samples uniformly an angle theta in [-pi, pi] and set the configuration to `cos(theta), sin(theta)`.

> @jmirabel thanks for the explanation.
> 
> Do you think the following code makes sense?
> 
> ```c++
> const ConfigVector_t::Scalar u = 1. + numeric_limits<double>::min()
> ```

@cmastalli What matters is the coherence between different functions in Pinocchio (integrate, difference...). I can't come up with a good method to determine the value. IMHO, we shouldn't touch it. I would rather add a comment in the code to justify it.

jcarpent, 2019-04-18:
I will close this issue as it seems no obvious solutions are coming out. Feel free to reopen it if needed.

closed 2019-04-18 by jcarpent

gabrielebndn, 2019-04-18:
> @gabrielebndn `randomConfiguration(model)` samples uniformly an angle theta in [-pi, pi] and set the configuration to `cos(theta), sin(theta)`.

Yes, I've seen it. So, it does not impose limits