stack-of-tasks/pinocchio #777: lowerPositionLimit and upperPositionLimit in pinocchio::Model have unexpected dimensions for unbounded joints - how to read them correctly?

markusgft, 2019-04-17:
https://github.com/stack-of-tasks/pinocchio/blob/904d7f9cfab86269487cd59ce9efd3f640fad8e1/src/multibody/joint/joint-revolute-unbounded.hpp#L23

Dear all, 

a quick question for understanding. Can you give a hint why the dimension of the config-vector for an unbounded joint is 2?

The question is motivated as follows: it leads to (at least for me) **unexpected behaviour when reading joint limits from the model**. For example, when parsing a 6 DoF fix-base robot from a URDF where the joints are defined as "continuous", the resulting joint limit vector will have size 2*6 = 12.
In other word, the field lowerPositionLimit and upperPositionLimit in pinocchio::Model will have twice the dimension as expected.

Probably it is my fault and I am doing something wrong, since you may have algorithmic reasons for assigning that special structure. However, if that is the case, **what would be the correct way for the user to read joint limits from the model, where the dimension of the joint limit vector is equal to the number of non-fixed joints?**

Thanks a lot in advance!

jcarpent, 2019-04-17:
Unbounded joints are single dof joints, but without bounds. Which means that the velocity vector is represented by \dot{theta} and the configuration vector is of dimension two with [cos(theta);sin(theta)] as representation.
The idea of having this representation is very similar to the Quaternion convention, but here, it is just limited by the fact you are rotating around a fixed axis.

Then, if you want to work with the bounds, you need to use the operation `difference` which will provide the distance in the tangent space of the configuration space between two configuration vectors. 
`v = difference(model,q1,q2)`
which has as reciprocal:
`q2 = integrate(model,q1,v)`

Is it clear enough?

gabrielebndn, 2019-04-17:
My bit: what we call "unbounded" joint in Pinocchio is the equivalent of what is called "continuous" joint in URDF convention.
I believe the name "unbounded" is a bit misleading and "continuous" would have been better. I also believe it is not worth changing the name.

jcarpent, 2019-04-17:
Following the precepts of @jmirabel, the naming of the standard URDF is not always the best imho.

gabrielebndn, 2019-04-17:
> Following the precepts of @jmirabel, the naming of the standard URDF is not always the best imho.

Yeah, I don't like it either

markusgft, 2019-04-18:
Hi guys, your response times are amazing, thank you! :) Your explanations, and the definition of the continuous joint make a lot of sense.
We will check out the "difference" method for computing the joint limits in minimal coordinates. Thanks again!

closed 2019-04-18 by markusgft