stack-of-tasks/pinocchio #784: Documentation not being displayed properly in Chrome - 16.04

olivier-stasse, 2019-04-25:
Chrome: Version 72.0.3626.96
Pb: After 
```
make doc
```
when going to 
```
doc/doxygen-html/md_doc_a-features_a-spatial.html
```
I got the following blank page:
[image]

Where as it is displayed properly on Firefox.
Any idea why ?

gabrielebndn, 2019-04-25:
This is surprising because the doc is working properly in Chrome on my (old) Macbook (El Capitan).

jmirabel, 2019-04-30:
The developer tools contains a tab that lists the errors. Can you provide them ?

olivier-stasse, 2019-05-02:
Nothing more than this:
```
MathJax.js:32 WARNING: cdn.mathjax.org has been retired. Check https://www.mathjax.org/cdn-shutting-down/ for migration tips.
```
Not sure that this is really related.

jcarpent, 2019-05-17:
I will close this issue. @olivier-stasse feel free to reopen it if needed.

closed 2019-05-17 by jcarpent