stack-of-tasks/pinocchio #786: Build error with C++17

aelkhour, 2019-04-29:
When trying to build pinocchio 2.1.1 with gcc-8 and the flag `-std=c++17`, I get a lot of build errors like this one:

```
pinocchio/serialization/archive.hpp:39:60: error: ISO C++17 does not allow dynamic exception specifications
                              const std::string & filename) throw (std::invalid_argument)
                                                            ^~~~~
pinocchio/serialization/archive.hpp:66:58: error: ISO C++17 does not allow dynamic exception specifications
                            const std::string & filename) throw (std::invalid_argument)
                                                          ^~~~~
```

I'm not familiar with dynamic exception specifications, but according to [cppref](https://en.cppreference.com/w/cpp/language/except_spec), they were deprecated since c++11 and removed in c++17. Is it a good idea to remove them from pinocchio ?

jcarpent, 2019-04-29:
Hi @aelkhour, thanks a lot for this feedback.
We indeed need to make our code compliant with C++17 and C++11. It would be interesting to make them silent for the new versions of C++ I think.

jcarpent, 2019-04-30:
Solved by #788.

closed 2019-04-30 by jcarpent

aelkhour, 2019-04-30:
Thanks @jcarpent ! Would it be possible to tag a new release?

jcarpent, 2019-04-30:
@aelkhour Done!