stack-of-tasks/pinocchio #790: Automatically setting PYTHON_DEB_LAYOUT

wxmerkt, 2019-04-30:
Hi,
While this issue technically belongs to https://github.com/jrl-umi3218/jrl-cmakemodules, the discussion upstream has been postponed several times and referred to user configuration (e.g., https://github.com/jrl-umi3218/jrl-cmakemodules/pull/47). 
One of the most common problems users have experienced with the JRL-UMI cmake module is that the Python paths are not set correctly and require manually passing an option (`-DPYTHON_USE_DEB_LAYOUT=1`, cf. [here](https://github.com/jrl-umi3218/jrl-cmakemodules/blob/d4feb8a5220b196a8375ca2ec5fa505272e82c33/python.cmake#L83-L84)) during configuration.

In a catkin environment, the system automatically provides `SETUPTOOLS_DEB_LAYOUT`. Would it be an acceptable option to add the following to eigenpy and pinocchio? [Or, better yet upstream to the cmake modules]:

```cmake
if(SETUPTOOLS_DEB_LAYOUT)
  SET(PYTHON_DEB_LAYOUT ON CACHE BOOL "Python directories for debian distributions")
endif(SETUPTOOLS_DEB_LAYOUT)
```

This would allow to build and package eigenpy/pinocchio etc. easier without the current wrappers (simply by adding a `package.xml`).

What do you think?

jmirabel, 2019-04-30:
A few questions regarding SETUPTOOLS_DEB_LAYOUT:
- Who provides it ? catkin ?
- Is its meaning documented somewhere ?
- Why don't they also provide PYTHON_DEB_LAYOUT along with SETUPTOOLS_DEB_LAYOUT automatically ?

jcarpent, 2019-04-30:
I am not so familiar with ROS, but regarding your comments, I would suggest putting it directly in the JRL Cmake module.

wxmerkt, 2019-04-30:
- Yes, catkin sets it if `find_package(catkin)` is called in a project CMake/or if it's a catkin_make/catkin-tools built workspace. It is set [here](https://github.com/ros/catkin/blob/928c2025206e1a3f0d9d504ef5b26efe1cc95d8b/cmake/python.cmake#L10-L14) and was introduced in https://github.com/ros/catkin/pull/114
- I am not too familiar with this - is `PYTHON_DEB_LAYOUT` a standard or is it a jrl-standard? The only code I find that uses this variable is based on the jrl-cmake-module. Upstreaming another similar variable into catkin, a core ROS component, is likely unlikely unless it is a documented Python REP.
- I'd be happy if it could be be put directly into the JRL cmake module - something similar was proposed in https://github.com/jrl-umi3218/jrl-cmakemodules/pull/47 but rejected. Perhaps @thomas-moulard could comment whether this auto-config as proposed above would be acceptable to the jrl cmake module?

jmirabel, 2019-05-02:
I understand your point now. Thanks for the clarification.

The option in catkin and in the JRL CMake module are very similar, if not exactly the same. Is there a CMake variable that can be checked for in order to know whether we are in a catkin environment ? As such, I would prefer to encapsulate the check in a sort of IF(CATKIN).

wxmerkt, 2019-05-02:
I looked deeper into it and from what it seems we need to first search for catkin in quiet mode before checking whether any of the variables is defined (e.g., `CATKIN_ENV`, `CATKIN_DEVEL_PREFIX`, `SETUPTOOLS_DEB_LAYOUT`). The issue here is that (at least on 18.04) this throws CMP0048 policy warnings that I have not yet been able to turn off by setting the cmake-policy to old.

```cmake
find_package(catkin QUIET)
if(DEFINED ${SETUPTOOLS_DEB_LAYOUT})
  ...
endif(DEFINED ${SETUPTOOLS_DEB_LAYOUT})
```

But since internally it only calls 

```cmake
if(EXISTS "/etc/debian_version")
  set(enable_setuptools_deb_layout ON)
endif()
```

The question is, whether adding just those lines to the python.cmake with an option to alternately turn it off manually would be acceptable? This way no catkin etc. would need to be added.

jmirabel, 2019-05-02:
Shouldn't it be this snippet instead:
```cmake
find_package(catkin QUIET)
if(DEFINED SETUPTOOLS_DEB_LAYOUT)
...
endif(DEFINED SETUPTOOLS_DEB_LAYOUT)
```

Checking whether `/etc/debian_version` exists will be blocking for the JRL CMake module. It prevents cross-compilation.
I agree with @thomas-moulard when he said
> Introducing distribution dependent rules in a CMake file is wrong.
> ...
> A good pull request would be to add a way to specify where the Python modules should be installed.
Then, it is up to you to set it to the correct value.
>
> The initial PYTHON_DEB_LAYOUT is a mistake

The best for me would be to replace the following line in pinocchio CMakeLists:
```cmake
SET(${PYWRAP}_INSTALL_DIR ${CMAKE_INSTALL_PREFIX}/${PYTHON_SITELIB}/${PROJECT_NAME})
```
Ideally, it shouldn't depend on PYTHON_SITELIB, but on a project dependent variable.

The code snippet you originally proposed would be fine for me if we could include a stricter check for CATKIN. Who triggers the CMP0048 ?

jcarpent, 2019-05-02:
> Checking whether `/etc/debian_version` exists will be blocking for the JRL CMake module. It prevents cross-compilation.

What do you mean here by preventing? I don't get your point sorry.

wxmerkt, 2019-05-02:
It wouldn't prevent cross compilation if it's turn off-able.

> Who triggers the CMP0048 ?

I added `find_package(catkin QUIET)` in `python.cmake` and it triggered CMP0048 from `googletest` which is apparently also searched for.  Setting the cmake-policy for it to OLD didn't change it, alas.

jmirabel, 2019-05-02:
> It wouldn't prevent cross compilation if it's turn off-able.

That's true. But if CMake had been developed with this policy, cross-compilation would be a nightmare. The ideal should be that CMake tells us where to install Python files. But that goes beyond the scope of this discussion.

> I added find_package(catkin QUIET) in python.cmake

I don't think that's what we want. I think we should merely add
```cmake
if(DEFINED CATKIN_ENV AND SETUPTOOLS_DEB_LAYOUT)
  SET(PYTHON_DEB_LAYOUT ON CACHE BOOL "Python directories for debian distributions")
endif(DEFINED CATKIN_ENV AND SETUPTOOLS_DEB_LAYOUT)
```
and let users ensure that catkin is looked for before `FINDPYTHON` is called. This is a bit weak and requires a clear comment to explain the rationale. If we want to make it safer, we may add a check in macro PYTHON_INSTALL_ON_SITE.

jcarpent, 2019-05-13:
@wxmerkt @jmirabel I At the end, what is the final solution to implement in the JRL CMake Module?
Thanks a lot for the great discussion.

jmirabel, 2019-05-13:
I would go for the above code snippet. We may want to use a variable different from CATKIN_ENV to check for catkin. Maybe CATKIN_FOUND ?

wxmerkt, 2019-05-13:
I have tried a few cases to include the above snippet (or similar) in the JRL CMake Module and wasn't fully satisfied with any solution - the reason is without using cache the variables aren't passed down and we would need another `find_package(catkin QUIET)` (they are not defined otherwise or available to projects in the workspace). 

As I don't have more cycles to test right now, I am using a custom packaging/wrapper that fixes the layout at the packaging level. 
Perhaps all that is required is to run `find_package(catkin QUIET)` in target projects (e.g. Pinocchio) before including the cmake module (and from looking further at it, I would have expected `_SETUP_DEBIAN()` which checks for a debian-based system to correctly configure the site-packages path - all current robotpkg debians do not adhere to dist-packages on debian systems for instance, i.e., I don't think the PYTHON_DEB_LAYOUT is even used at the cross-compilation/packaging stage as described).

jmirabel, 2019-05-13:
Then I don't know how to achieve it.

wxmerkt, 2019-05-13:
Me neither - while I still think resolving this issue would make packaging/wrapping a lot easier, I don't see an easy solution out of the impasse. Hence, closing the issue for now.

closed 2019-05-13 by wxmerkt