stack-of-tasks/pinocchio #792: About the Urdf

ddliugit, 2019-05-06:
I am trying to build a simple bipedal robot and using Pinocchio for Jacobian calculation. I import my robot into Pinocchio and cannot display it in gepetto-gui. 
The error is like this:

libpng warning: iCCP: known incorrect sRGB p
rofile
qt.qpa.window: Window position QRect(208,-55
 1023x1009) outside any known screen, using 
primary screen
libpng warning: iCCP: known incorrect sRGB p
rofile
libpng warning: iCCP: known incorrect sRGB p
rofile
libpng warning: iCCP: known incorrect sRGB p
rofile
List of GroupNodes :
   hpp-gui
   python-pinocchio
 does not exist.: unspecified iostream_categ
ory error
 does not exist.: unspecified iostream_categ
ory error

Node "world/pinocchio/collisions/torso_0" does not exist.
Node "world/pinocchio/collisions/arm_l_shou_p_0" does not exist.
Node "world/pinocchio/collisions/arm_l_shou_r_0" does not exist.

Can you help me

jmirabel, 2019-05-06:
I have no idea of what's going on. Can you provide the robot description, whatever method you use ?

jcarpent, 2019-05-06:
@ddliugit It seems that you have not provided the path where the description of your robot is located. It is required to retrieve the full path of the visual and collision objects.

jcarpent, 2019-05-06:
An example is located here: https://github.com/stack-of-tasks/pinocchio/blob/master/examples/python/load-urdf.py

ddliugit, 2019-05-06:
Thanks. I solved it. I didn't setup the base_link in urdf and the directory of meshes is wrong

closed 2019-05-06 by ddliugit