stack-of-tasks/pinocchio #793: About the joint setting of floating body in Urdf

ddliugit, 2019-05-06:
Is there an example of setting up a floating joint in Urdf?

Basically, I want to set up the 6 degrees of freedom for the torso of a humanoid model. 

Currently, the torso is the first link in Urdf, like this:
<link name="torso">
    <inertial>
      <origin
        xyz="-0.0273890624666731 -6.24390539272213E-11 0.0090874604275595"
        rpy="0 0 0" />
      <mass
        value="1.79837714981727" />
      <inertia
        ixx="0.00965569591913053"
        ixy="2.62519057565357E-11"
        ixz="0.000466515313675841"
        iyy="0.00841839017045047"
        iyz="1.1590569581039E-11"
        izz="0.00428669058417288" />
    </inertial>
    <visual>
      <origin
        xyz="0 0 0"
        rpy="0 0 0" />
      <geometry>
        <mesh
          filename="package://humanoid/meshes/torso.STL" />
      </geometry>
      <material
        name="">
        <color
          rgba="0.752941176470588 0.752941176470588 0.752941176470588 1" />
      </material>
    </visual>
    <collision>
      <origin
        xyz="0 0 0"
        rpy="0 0 0" />
      <geometry>
        <mesh
          filename="package://humanoid/meshes/torso.STL" />
      </geometry>
    </collision>
  </link>

Printing out the model shows that the torso is the universe in Pinocchio(joint is 0).

How to setup the floating joint which can be well recognized by Pinocchio?

jcarpent, 2019-05-06:
When loading a model in Pinocchio,
```
using namespace pinocchio
Model model;
urdf::buildModel(filename,model,JointModelFreeFlyer());
```
The last argument allows you to specify the root joint to be a JointModelFreeFlyer for instance.

ddliugit, 2019-05-06:
I am using python.

`robot = pinocchio.RobotWrapper.BuildFromURDF(urdf_model_path,[mesh_dir],pinocchio.JointModelFreeFlyer())`

The above code worked!

closed 2019-05-06 by ddliugit