stack-of-tasks/pinocchio #794: Continuous joint in URDF adds 2 elements to the configuration space

mkatliar, 2019-05-07:
Example:
```c++
auto const xml = R"urdf(
    <robot name="test">
        <link name="base">
        </link>
        <link name="link1">
            <inertial>
            <origin xyz="0. 0. 0."/>
            <mass value="1."/>
            <inertia ixx="1." ixy="0." ixz="0." iyy="1." iyz="0." izz="1."/>
            </inertial>
        </link>
        <joint name="joint0" type="continuous">
            <parent link="base"/>
            <child link="link1"/>
            <axis xyz="0. 0. 1."/>
            <origin rpy="0. 0. 0." xyz="0. 0. 0."/>
        </joint>
    </robot>
)urdf";

pinocchio::Model model;
pinocchio::urdf::buildModelFromXML(xml, model);

assert(model.nq == 1);
```

This code results in `model.nq == 2` and the assert will fail. Changing joint type to "revolute" results in `model.nq == 1`, as expected.

According to http://wiki.ros.org/urdf/XML/joint,
```
type (required)

Specifies the type of joint, where type can be one of the following:
...
continuous - a continuous hinge joint that rotates around the axis and has no upper and lower limits.
```
Therefore I expect a continuous joint to add 1 element to the configuration space, the same way as a revolute joint would do.

Seems also related to this question: https://github.com/stack-of-tasks/pinocchio/issues/735#issuecomment-472506592

jmirabel, 2019-05-07:
For continuous joints, Pinocchio uses the complex representation of SO(2). The 2 elements are thus cos(theta), sin(theta).

jcarpent, 2019-05-07:
We use `cos(theta), sin(theta)` to represent the configuration vector of a Continuous joint while only `theta` for a revolute joint.
This expected because continuous means that there are no bounds on the configuration vectors. 
But for both cases, we have `\dot{theta}`which corresponds to the velocity of the joint.

We also use `cos(theta), sin(theta)` for continuous joints in a similar way we use rely on quaternions to represent the configuration of a spherical joint.

@mkatliar I hope this answer will satisfy your curiosity.

gabrielebndn, 2019-05-07:
This was already the subject of a previous issue: #777

jcarpent, 2019-05-07:
I will close this issue. @mkatliar Feel free to open it again if needed.

mkatliar, 2019-05-07:
Thank you very much for your replies, now it is clear. Would it be possible to add this explanation somewhere in the docs?

jcarpent, 2019-05-07:
Sorry. I was too fast. We will add it in the doc very soon.

jcarpent, 2019-05-07:
@gabrielebndn Can you help me to structure the documentation of the Joint classes inside the Pinocchio doc?