stack-of-tasks/pinocchio #796: sincos unit test fails when compiled with clang-8.0 in Release mode

mkatliar, 2019-05-15:
Build commands:
```bash
cd build
cmake .. -DCMAKE_CXX_COMPILER=/usr/bin/clang++ -DCMAKE_BUILD_TYPE=Release
make sincos
```

Run test:
```bash
unittest/sincos
```

Test output:
```
Running 1 test case...
/home/kotlyar/software/pinocchio/unittest/sincos.cpp(33): [1;31;49merror: in "sincosTest/test_sincos": check sin_value == sin_value_ref has failed[0;39;49m
/home/kotlyar/software/pinocchio/unittest/sincos.cpp(33): [1;31;49merror: in "sincosTest/test_sincos": check sin_value == sin_value_ref has failed[0;39;49m
...
"sincosTest/test_sincos": check sin_value == sin_value_ref has failed[0;39;49m
*** 1287 failures are detected in the test module "sincosTest
```

I added the following lines to the test code to see the values of sin and cos:
```c++
if (sin_value != sin_value_ref)
      std::clog << "alpha = " << alpha << ", sin_value = " << sin_value << ", sin_value_ref = " << sin_value_ref << ", delta = " << sin_value - sin_value_ref << std::endl;    

    if (cos_value != cos_value_ref)
      std::clog << "alpha = " << alpha << ", cos_value = " << cos_value << ", cos_value_ref = " << cos_value_ref << ", delta = " << cos_value - cos_value_ref << std::endl;
```
The output is:
```
...
alpha = 0.824938, cos_value = 0.678602, cos_value_ref = 0.678602, delta = -1.11022e-16
alpha = 0.37224, sin_value = 0.363702, sin_value_ref = 0.363702, delta = 5.55112e-17
alpha = 0.545731, sin_value = 0.519043, sin_value_ref = 0.519043, delta = 1.11022e-16
alpha = 0.88229, sin_value = 0.772196, sin_value_ref = 0.772196, delta = -1.11022e-16
alpha = 0.253306, sin_value = 0.250605, sin_value_ref = 0.250605, delta = -5.55112e-17
alpha = 0.871944, cos_value = 0.643339, cos_value_ref = 0.643339, delta = -1.11022e-16
alpha = 0.374064, cos_value = 0.93085, cos_value_ref = 0.93085, delta = 1.11022e-16
alpha = 0.650994, sin_value = 0.605977, sin_value_ref = 0.605977, delta = 1.11022e-16
...
```
So the difference between the actual and the reference sin and cos values is very small but not 0.

Compiler version:
```
clang version 8.0.0-3 (tags/RELEASE_800/final)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
```

Exact compile command:
```bash
/usr/bin/clang++   -pedantic -Wno-long-long -Wall -Wextra -Wcast-align -Wcast-qual -Wformat -Wwrite-strings -Wconversion  -O3 -DNDEBUG  -rdynamic CMakeFiles/sincos.dir/sincos.cpp.o  -o sincos -Wl,-rpath,/home/kotlyar/software/pinocchio/build/src ../src/libpinocchio.so -lboost_unit_test_framework -lboost_filesystem -lboost_system -lboost_serialization -lurdfdom_sensor -lurdfdom_model_state -lurdfdom_model -lurdfdom_world -lconsole_bridge
```

CPU info: [cpuinfo.txt](https://github.com/stack-of-tasks/pinocchio/files/3184741/cpuinfo.txt)

With g++ as the compiler, the test passes.

mkatliar, 2019-05-15:
In Debug configuration, I get only 1 failure in the test:
```
Running 1 test case...
alpha = 0.7053, cos_value = 0.761417, cos_value_ref = 0.761417, delta = 1.11022e-16
/home/kotlyar/software/pinocchio/unittest/sincos.cpp(34): error: in "sincosTest/test_sincos": check cos_value == cos_value_ref has failed

*** 1 failure is detected in the test module "sincosTest"
```
Compiler command line:
```bash
/usr/bin/clang++   -pedantic -Wno-long-long -Wall -Wextra -Wcast-align -Wcast-qual -Wformat -Wwrite-strings -Wconversion  -g  -rdynamic CMakeFiles/sincos.dir/sincos.cpp.o  -o sincos -Wl,-rpath,/home/kotlyar/software/pinocchio/build/src ../src/libpinocchio.so -lboost_unit_test_framework -lboost_filesystem -lboost_system -lboost_serialization -lurdfdom_sensor -lurdfdom_model_state -lurdfdom_model -lurdfdom_world -lconsole_bridge
```

mkatliar, 2019-05-15:
The issue appears only for Scalar = double. For other types, the match is exact.

mkatliar, 2019-05-15:
Proposed solution: for the `double` type, check that the values are close with ~1e-16 absolute tolerance, instead of exact match.

stonneau, 2019-05-16:
A more general remark is that as far as I know 
BOOST_CHECK_CLOSE is not recommended to compare to 0, but that BOOST_CHECK_SMALL should rather be used. 
See
https://stackoverflow.com/questions/1093453/difference-between-boost-check-close-and-boost-check-close-fraction#1093530

and https://www.boost.org/doc/libs/1_34_0/libs/test/doc/components/test_tools/floating_point_comparison.html

mkatliar, 2019-05-16:
@stonneau we don't compare to 0, we compare two numbers -- the actual and the reference sin/cos value. What I do is
```c++
BOOST_CHECK_CLOSE_FRACTION(sin_value, sin_value_ref, sinCosTolerance<Scalar>());
BOOST_CHECK_CLOSE_FRACTION(cos_value, cos_value_ref, sinCosTolerance<Scalar>());
```

jmirabel, 2019-05-16:
To be sure you never compare to zero, you could replace
```c++
Scalar alpha = (Scalar)std::rand()/(Scalar)RAND_MAX;
```
by
```c++
Scalar alpha = (Scalar)0.1 + (Scalar)std::rand()/(Scalar)RAND_MAX;
```
This way we are sure that both the sine and the cosine are not close to zero.

jcarpent, 2019-05-16:
I've suggested a workaround proposed by Boost.Test itself in #797, which does not require such a trick.

jcarpent, 2019-05-17:
Solved by #797.

closed 2019-05-17 by jcarpent