stack-of-tasks/pinocchio #801: What's the correct way to subscribe to release notifications?

markusgft, 2019-05-23:
Dear all,
i was wondering if you maintain a mailing list or similar, where you announce releases or major merges to the master branch?
I am tracking the pinocchio master branch in one my projects and the latest release (3-4 weeks ago) unfortunately broke some of my interfaces. Of course I am aware that I could simply freeze my version to a specific pinocchio release, but still, I'd like to know if you guys have some kind of mechanism to announce major releases or removal of deprecated interfaces.
Any information would be highly appreciated, and keep up the great work!

jmirabel, 2019-05-23:
Personally, I receive an email when someone adds a new release to github. Is this sufficient for you ?

olivier-stasse, 2019-05-23:
With respect to the interface, Justin is usually trying to specifiy which methods will be deprecated through warnings during compilation from a version to another.
In addition the group is trying the semantic versionning approach which is usually recommended:
MAJOR.MINOR.PATCH
where MINOR means that the API has been changed,
and MAJOR means that the structure has been changed.
Hope that this  helps.

jcarpent, 2019-05-23:
Dear @markusgft,
I'm very sorry if you've encountered some breakings with latest versions of Pinocchio.
We are trying to always make deprecated former signatures, and we avoid to remove them directly. 

Can you tell us which signatures have been broked?

By the way, as mentioned by @jmirabel, all the release are officially published using the GitHub.
Do you think another channel would be useful?

markusgft, 2019-05-24:
Dear all,
I think I have to rephrase this ... actually it appeared to me that all is fine on your side. The issue is rather that the [pinocchio ROS-release](https://github.com/ipab-slmc/pinocchio_catkin) by @wxmerkt was forwarded to the new release, which means by default all changes get installed when doing
`sudo apt-get upgrade`. 
I think we should migrate the discussion to a separate issue within pinocchio-catkin in case @wxmerkt is interested!
Thanks again to everyone!

closed 2019-05-24 by markusgft

wxmerkt, 2019-05-24:
Hi @markusgft,
Apologies for the issues and inconvenience - I did upgrade the ROS release to be inline with release `2.1.3` here - and didn't realise some APIs had been removed from `2.0.x` to `2.1.x`. I'm planning to stay on the `2.1.x` line for Kinetic/Melodic going forward. Happy to hear your feedback and comments - especially in case there is an incompatibility or other underlying issue. [We have to resort to some packaging tricks as `PYTHON_DEB_LAYOUT` is not set directly (cf. #790)  - otherwise ROS packaging would be a simple `package.xml` - and this may cause some issues with `pkg-config`]

jcarpent, 2019-05-24:
@wxmerkt and @markusgft Thanks a lot for this nice collaboration.
@wxmerkt Is `hpp-fcl` available on ROS. In this case, you have to set up the `BUILD_WITH_COLLISION_SUPPORT` variable.

wxmerkt, 2019-05-24:
I’ve released hpp-fcl but the released Pinocchio is without collision support (there have been challenges setting all variables and paths manually - which we are required to do as the python layout isn’t correctly set automatically otherwise).

We are building packages based on (and using the python bindings of) the ROS released versions - is there’s anything broken I’ve missed please open an issue over at Pinocchio-catkin - let’s get it fixed :)