stack-of-tasks/pinocchio #802: End-effector position and rotation

alejandroastudillo, 2019-05-27:
Hi guys, thank you very much for all your effort building this amazing framework.

I just noticed that after building a model from an URDF ([Kinova Gen3 robot](https://github.com/Kinovarobotics/ros_kortex/blob/master/kortex_description/urdf/JACO3_URDF_V11.urdf), KUKA iiwa 14, etc), the variable `model.njoints` is missing one fixed joint which represents the end-effector. Then, for a 7-dof robot, `model.njoints = 8` including the universe, but it should be 9 if the end-effector fixed joint is included. I have checked this with both the kinova gen3 and kuka iiwa14 urdf.

Due to this, I haven't been able to get the pose of the end-effector after using the `pinocchio::forwardKinematics(model,data,q)` function and printing the vector of joint placements `data.oMi[k]`, since the `oMi` vector doesn't contain information about the fixed joint.

Is that a bug, or is there any other way to get the pose of the end-effector (and not of the last actuator)?

jcarpent, 2019-05-27:
@alejandroastudillo 

First of all, thanks a lot for using our tools. 
Concerning your issue, in Pinocchio, we do not append fixed in the vector of joints (model.joints[]).
We want to avoid to have additional computation burdens when adding a fixed joint.

The idea we follow in Pinocchio is to append the fixed joints as a frame (model.frames[]). You can then retrieve its id with `id = model.getFrameId("name")` and then get its parent joint in the kinematic free with `model.frames[id].parent`.

You can also look at the notion of a frame which contains also the fixed transform between a frame and its parent joint.

I hope these details answer your issue.

gabrielebndn, 2019-05-27:
Hi @alejandroastudillo,
as @jcarpent just said, "fixed" joints are not represented as joints in Pinocchio, but as frames.
You can get the corresponding frame id with `frameId = model.getFrameId("name")`. Then, the frame placement is found in `data.oMf[frameId]`.

Notice frame placements are *not* computed by `forwardKinematics`. To compute frame placements, you have thee options:
- *after* running `forwardKinematics(model,data,q)`, run `updateFramePlacements(model,data)`. This will fill `oMf`
- *instead of* `forwardKinematics(model,data,q)`, run `framesForwardKinematics(model,data)`. This is equivalent to  `forwardKinematics` + `updateFramePlacements`
- *after* running `forwardKinematics(model,data,q)`, run `updateFramePlacements(model,data,frameId)`. This will set the correct value for `oMf[frameId]`. It will spare you the computation of the placement for all frames you do not need.

All the above methods are contained in the header `pinocchio/algorithm/frames.hpp`

alejandroastudillo, 2019-05-27:
Thank you very much @jcarpent and @gabrielebndn 
I was able to get the end-effector pose with that.

On the other hand, I have another two questions. I don't know if this is the correct place to ask them or should I open a new issue?

The first one is still related to the forward kinematics function. Since the dimension of the configuration vector representation `nq` is not the same as the dimension of the velocity vector space `nv = Degrees of Freedom`, is it possible to know to which joint does every index of the `q` vector corresponds? (so that one can apply the FK function with a custom `q` vector).

My second question is more about what I am working with, and the possibilities that Pinocchio can give to me. I am trying to interface Pinocchio with CasADi in order to generate symbolic expressions for the FK and FD of a robot based on its URDF, and use those expressions in the solution of an optimal control problem. So far I am scripting the use of custom scalar types in order to use symbolic variables as arguments in some Pinocchio methods. But I would like to know if you have already try to use symbolic variables with Pinocchio (maybe with CppAD)?

gabrielebndn, 2019-05-27:
@alejandroastudillo you can easily find the index of joint `i` in the configuration vector through `model.joints[i].idx_q()`. The index in the velocity vector is `model.joints[i].idx_v()`. `model.joints[i].nq()` and `model.joints[i].nv()` give the size of the configuration and of the velocity for joint `i`.

jcarpent, 2019-05-27:
> My second question is more about what I am working with, and the possibilities that Pinocchio can give to me. I am trying to interface Pinocchio with CasADi in order to generate symbolic expressions [...]

Yes, sure. You can use Pinocchio with CppaAD and CppADCodeGen. 
If you have installed CppAD and the counterpart CppADCodeGen, you can activate the compilation of the tests by setting ON the two Cmake variables: `BUILD_WITH_AUTODIFF_SUPPORT` and `BUILD_WITH_CODEGEN_SUPPORT`

Currently, I do not have concrete examples of how to use it, but the unit tests should be sufficient at first glance. I will try to make concrete the use of Pinocchio with these two frameworks. If you need more details, you can also have a look here, on Pinocchio's paper with benchmarks: https://hal-laas.archives-ouvertes.fr/hal-01866228

alejandroastudillo, 2019-05-28:
thanks for your answers. You really helped me a lot.

closed 2019-05-28 by alejandroastudillo

alejandroastudillo, 2019-07-08:
Hi, one small question to add here: Is it possible to retrieve the joint type ('Continuous, 'Revolute', etc) from the `model` or `data` structs?

jmirabel, 2019-07-08:
There is a one to one mapping between C++ joint types and URDFs.
Have a look here: https://github.com/stack-of-tasks/pinocchio/blob/8e8f0711b1efa18e8dbc3ee0eea5693cb2cdd2b4/src/parsers/urdf/model.hxx#L223

jcarpent, 2019-07-08:
@alejandroastudillo In the field `model.joints[id].shortname()`you can retrieve this information with the high-level interface (`id` is the id of the joint).
Another solution would be to do a dynamic cast, but this is not recommended.

alejandroastudillo, 2019-07-09:
@jmirabel @jcarpent thank you very much. The `shortname` was what I was looking for.

annan-tang, 2021-11-14:
Hi, thanks very much for the great Pinocchio library! I also have some small questions related to the "Frames":
1. when I built a floating-base humanoid with 38 joints, the len(iMf) is 40, but the len(Frames) is 79, they didn't match with each other, why it happened?
2. It seems that the Frames are built **automaticall**y when load from URDF file, but these operations frames aren't the target points which I need, is it possible to **delete them and add my custom Frames**? if yes, I would appreciate a lot if  you can **tell me the methods** or show some examples

jcarpent, 2021-11-14:
@annan-tang could you provide a tiny example to show 1?

annan-tang, 2021-11-14:
Thank you for the reply. Here is my example:
```
import pinocchio as pino
robot = pino.RobotWrapper.BuildFromURDF(filename=urdf_name,
                                                     root_joint=pino.JointModelFreeFlyer())
```
And then, I use the python console in pycharm to find the data structure like this:
in ```robot.model```: ```frames={stdVec_Frame:79}```
in ```robot.data```: ```iMf={stdVec_SE3:40  oMf={stdVec_SE3:79}```
While the joint number of my robot is 38 joints + 1 virtual root joint of 6 Dof + 1 universe = 40.

jcarpent, 2021-11-14:
iMf is not really concerned with Frames in fact. This is an intermediate quantity that is not useful for the end-user in general.
You can easily add a frame to your kinematic tree like that:
`model.addFrame(frame)`

We do not support the removing of frames, but if this feature is needed, we may add it. Do you really need it?

annan-tang, 2021-11-14:
Oh! Thank you very much!! The ```addFrame()``` is really what I want. 

For the iMf, I got the information in [This tutorial](https://github.com/MeMory-of-MOtion/summer-school/blob/master/tutorials/pinocchio/1_geometry_and_dynamics.ipynb): **by iMf the fixed placement of the operational frame with respect to the parent joint frame, the placement of the operational frame with respect to the world is easily computed by oMf(q) = oMi(q) * iMf**

But now, I found the information of specific frame's placement w.r.t its parent frame is stored in the frame object, I can print it easily.

For the removing of frames, I think it is not necessary, because, from the information above, I can only update the placement of the target frame by ```updateFramePlacement(model,data,frameId)``` to seep up the computation with the ignorance of updating other redundant frames. 

By the way, would you please send me a link f there are any channels about Pinocchio in slack or some other talk APPs?

Thank you again for the kindly help!

jcarpent, 2021-11-14:
Your detailed answer looks perfect.
Here it is: https://matrix.to/#/!xlvgaTQeHHVMafeNDX:matrix.org/$XixToKnEsZcnwImJvVPL0HaBcMisAxEJ6MkEMRAQtdI?via=matrix.org&via=laas.fr

Danfoa, 2023-03-10:
> @alejandroastudillo In the field `model.joints[id].shortname()`you can retrieve this information with the high-level interface (`id` is the id of the joint). Another solution would be to do a dynamic cast, but this is not recommended.

This should be mentioned somewhere in the documentation, at least for the python bindings. There are multiple scenarios where it might be required to know the nature of the joint and its DoF.

jcarpent, 2023-03-10:
@Danfoa could you suggest a PR implementing this?

Danfoa, 2023-03-10:
Where exactly would you think this information might go? At first glance I thought about [Examples>Loading the model>python](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/md_doc_b-examples_a-model.html#autotoc_md33)

With a brief mention of detecting the joint types from the short name. But I am not sure if it's appropriate to put it there.

jcarpent, 2023-03-10:
I would say that the best place is here: https://github.com/stack-of-tasks/pinocchio/blob/2e33e8624568d04149707fe3db6ac5d998f5a89a/bindings/python/multibody/joint/joint-derived.hpp#L49

Dantemiwa, 2023-07-31:
HI, I have read the above discussion, but I'm still confused about how to get the position and orientation through Pinocchio.
The model is as followed:
` <joint name="left_joint_1" type="revolute">
    <origin rpy="0 0 0" xyz="-0.001966 0.148956 0.0"/>
    <parent link="base_link"/>
    <child link="left_leg_hip_1"/>
    <limit
      lower="-3.0"
      upper="3.0"
      effort="100"
      velocity="10" />
    <axis xyz="-0.0 1.0 0.0"/>
  </joint>
  <joint name="left_joint_2" type="revolute">
    <origin rpy="0 0 0" xyz="0.0 0.01 -0.2"/>
    <parent link="left_leg_hip_1"/>
    <child link="left_leg_knee_1"/>
    <axis xyz="0.0 1.0 -0.0"/>
    <limit
      lower="-3.0"
      upper="3.0"
      effort="100"
      velocity="10" />
  </joint>
  <joint name="right_joint_1" type="revolute">
    <origin rpy="0 0 0" xyz="-0.001966 -0.151044 0.0"/>
    <parent link="base_link"/>
    <child link="right_leg_hip_1"/>
    <axis xyz="0.0 1.0 0.0"/>
    <limit
      lower="-3.0"
      upper="3.0"
      effort="100"
      velocity="10" />
  </joint>
  <joint name="right_joint_2" type="revolute">
    <origin rpy="0 0 0" xyz="0.0 -0.01 -0.2"/>
    <parent link="right_leg_hip_1"/>
    <child link="right_leg_knee_1"/>
    <axis xyz="0.0 1.0 0.0"/>
    <limit
      lower="-3.0"
      upper="3.0"
      effort="100"
      velocity="10" />
  </joint>`
Which is a base, with two bars and linkages at each side of the base, or a bi-pedal. The length of the bar-link is 0.2m.
I set all my joint angle q = 0, and I get the following
`omi: 
0 0 0
0 0 0
-0.001966 -0.148956         0
-0.001966 -0.158956       0.2
-0.001966  0.151044         0
-0.001966  0.161044       0.2`

`iMf: 
0 0 0
0 0 0
0 0 0
0 0 0
0 0 0
0 0 0`

`omf: 
0 0 0
0 0 0
0 0 0
-0.001966 -0.148956         0
-0.001966 -0.148956         0
-0.001966 -0.158956       0.2
-0.001966 -0.158956       0.2
-0.001966  0.151044         0
-0.001966  0.151044         0
-0.001966  0.161044       0.2
-0.001966  0.161044       0.2`

In which, pinocchio correctly shows the vector of the knee joint of my robot. But I cannot find the postion or orientation of the end effector, which should be shown like `-0.001966  0.161044  0.4`, or something like that.

stephane-caron, 2023-08-07:
@Dantemiwa You should open a [new discussion thread](https://github.com/stack-of-tasks/pinocchio/discussions/new/choose) for your question, as the original question of this issue has been addressed. Make sure to include two more things in your separate posting:

1. Minimal code to reproduce your issue.
2. Details on what in @gabrielebndn's answer https://github.com/stack-of-tasks/pinocchio/issues/802#issuecomment-496210616 does not work for you.

This will maximize your chances of getting useful feedback from the community.