stack-of-tasks/pinocchio #803: Error printing joints

gabrielebndn, 2019-05-27:
The following code:
```
  Model model;
  buildModels::humanoid(model);
  Data data(model);
  std::cout << model.joints[1] << std::endl;
```

prints
```
JointModelFreeFlyer
  index: 18446744073709551615
  index q: -1
  index v: -1
  nq: 7
  nv: 6
```

which is wrong since is should be
```
JointModelFreeFlyer
  index: 1
  index q: 0
  index v: 0
  nq: 7
  nv: 6
```

This is weird because methods `id()`, `idx_q()` and `idx_v()` return correct values. The same happens in Python.
Although I think this problem may be fairly easy to solve, I do not understand why it happens.

jcarpent, 2019-05-27:
I know the origin of this issue. I will solve it asap.
@gabrielebndn Thanks for raising it.

jcarpent, 2019-05-27:
Fixed by #804.

closed 2019-05-27 by jcarpent