stack-of-tasks/pinocchio #807: Serialization : "Boost::serialization" to "Protobuf"

JasonChmn, 2019-06-04:
Should we implement a serialization using protobuf instead of boost ?
This topic was discussed previously and protobuf seems to be the best option.
This will add another dependency to the project but which could also be used in Curves API.

jcarpent, 2019-06-04:
@JasonChmn Maybe, it is important to remind people the topic of the discussion, because it was done in a private channel for a project relying on Pinocchio.

Concerning the support of Protobuf, I think it is a good idea. But we should keep the Boost.Serialization codes because it supports binaries formats, while Protobuf is a mix of both worlds.

nmansard, 2019-06-04:
Thanks for the quick answer.
We are seeking for opinion whether it would be a good choice to migrate our serialization toward protobuf. I would agree that, in a first time, we could keep two serialization options (boost as already implemented, and protobuf). However, it is likely that only one would be kept in the longer term. If we choose to integrate protobuf, I believe it should be to replace boost serialization on this longer term.

Do we have any opinions pro or against protobuf? We yet have these positive feedbacks:
- protobuf is back compatible (i.e. any file generated by protobuf can be loaded with later versions of protobuf), which seems to not be the case of boost::serialization (reference welcome here).
- protobuf seems easier to use, and in particular seems straight foward to use with eigen (again, reference welcome).
On the down side, protobuf seems a more difficult dependency (in particular, knowing that boost at large will remain a dependency of Pinocchio).

nmansard, 2019-06-04:
I would very much like to have the opinions of @nim65s about this issue, and possibly of @wxmerkt if available.

wxmerkt, 2019-06-04:
I have not yet used serialization with Pinocchio and thus cannot comment, I am afraid. What does serialization in Pinocchio do and are there examples? If it's about robot structure etc that does not create reference-counted/pointers, [Cereal](https://uscilab.github.io/cereal/) may be an option as well and header-only.

For the multicontact-API, I believe a structure like protobuf or similar would be preferable (as discussed there). From the discussion there, one of the strongest arguments was by @proyan that boost's serialization is not inter-version-compatible.

jcarpent, 2019-06-04:
>     * protobuf is back compatible (i.e. any file generated by protobuf can be loaded with later versions of protobuf), which seems to not be the case of boost::serialization (reference welcome here).

Boost.Serialization is also back compatible. It is just a matter of dealing with the version flag.
The problem of @proyan was appearing at the XML level, the other formats remain valid.

jcarpent, 2019-06-04:
>     * protobuf seems easier to use, and in particular seems straight foward to use with eigen (again, reference welcome).

Boost.Serialization is very easy to use too. I have done all the major works to deal with Eigen and the different output formats (XML, text or binary).
I think it would not be a mistake to support both of them, as the number of data structure in Pinocchio are not so numerous.

jcarpent, 2019-06-04:
> What does serialization in Pinocchio do and are there examples? If it's about robot structure etc that does not create reference-counted/pointers, [Cereal](https://uscilab.github.io/cereal/) may be an option as well and header-only.

Serialization in Pinocchio is only for data structure, in order to be able to save and load those pieces of information. This is mandatory when one's work on clusters and parallel computations for instance.

Concerning Cereal, it seems that the framework is only C++11 compatible, that we should avoid currently as some external users are still following the C++98/C++03 standards.

jcarpent, 2019-08-12:
I will close this issue as it seems Protobuff is no more the priority for LAAS now. @JasonChmn Feel free to reopen this issue if you need.

closed 2019-08-12 by jcarpent

jmirabel, 2022-05-11:
I came across some issues when trying to load a serialized model. I haven't completely finalized the introspection but I think the issue is that the model is serialized with a newer version of Boost Serialization than the one used for loading the model.
If my hypothesis is true, it is a rather annoying issue for us.

I am writing this here since it bumped into this while searching and I also bumped into one of the missing reference in @nmansard comment above.

> * protobuf is back compatible (i.e. any file generated by protobuf can be loaded with later versions of protobuf), which seems to not be the case of boost::serialization (reference welcome here).

Not true. The reference is https://www.boost.org/doc/libs/1_74_0/libs/serialization/doc/todo.html#backversioning

> ### Back Versioning
> It has been suggested that a useful feature of the library would be the ability to create "older versions" of archives. Currently, the library permits one make programs that are guarenteed the ability to load archives with classes of a previous version. But there is not way to save classes in accordance with a previous version. At first I dismissed this a a huge project with small demand. A cursory examination of the code revealed that this would not be very difficult. It would require some small changes in code and some additional tests. Also it would require special treatment in the documentation - perhaps a case study.

jcarpent, 2022-05-11:
One possible issue is that the class Model evolves with time, and some fields are missing in the serialization file.

jcarpent, 2022-05-11:
@jmirabel If you are ready, you can add the support of protobuff or a better solution in Pinocchio and hpp-fcl.

jmirabel, 2022-05-11:
I just tried a reproducible example.
In a fresh ubuntu:20.04 docker container, I install pinocchio and serialize `pinocchio.buildSampleModelManipulator()`. I then move the file into a fresh ubuntu:18.04 container where I install pinocchio and try to deserialize.
I get the following assertion:
```
In [4]: with open("model.pkl", "rb") as f:
   ...:     model = pickle.load(f)
   ...:     
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
<ipython-input-4-8b77a1eac347> in <module>()
      1 with open("model.pkl", "rb") as f:
----> 2     model = pickle.load(f)
      3 

RuntimeError: unsupported version
```
I suppose this exception is thrown by boost (ref https://www.boost.org/doc/libs/1_66_0/libs/serialization/doc/exceptions.html#unsupported_version)

jmirabel, 2022-05-11:
I don't know whether protobuf would solve the problem.