stack-of-tasks/pinocchio #808: Axis order in JointSphericalZYX

gabrielebndn, 2019-06-05:
JointSphericalZYX represents the angular quantities in the following order, as the name suggests:
- rotation around Z
- rotation around Y
- rotation around X

with the rotation matrix given as

<a href="https://www.codecogs.com/eqnedit.php?latex=R&space;=&space;R_z(q_0)&space;R_y(q_1)&space;R_x(q_2)" target="_blank"><img src="https://latex.codecogs.com/gif.latex?R&space;=&space;R_z(q_0)&space;R_y(q_1)&space;R_x(q_2)" title="R = R_z(q_0) R_y(q_1) R_x(q_2)" /></a>

corresponding to the *moving* ZYX axes.

I find this unusual. The most common convention is roll-pitch-yaw, corresponding to the *fixed* XYZ, which is completely equivalent except for the order in which the angles are given, i.e:
- rotation around X
- rotation around Y
- rotation around Z

with the rotation matrix given as

<a href="https://www.codecogs.com/eqnedit.php?latex=R&space;=&space;R_z(q_2)&space;R_y(q_1)&space;R_x(q_0)" target="_blank"><img src="https://latex.codecogs.com/gif.latex?R&space;=&space;R_z(q_2)&space;R_y(q_1)&space;R_x(q_0)" title="R = R_z(q_2) R_y(q_1) R_x(q_0)" /></a>

This is, for instance, the convention assumed in the Python submodule `pinocchio.rpy` (which is slightly inconsistent). It is also the convention employed by the entities in sot-core.

Why this choice of order in `JointSphericalZYX`? Wouldn't it be better to replace it with a `JointSphericalRPY` which has the inverse order of the three coordinates?

jmirabel, 2019-06-05:
I agree it is misleading. From ZYX, I would have understood your second equation:

<a href="https://www.codecogs.com/eqnedit.php?latex=R&space;=&space;R_z(q_2)&space;R_y(q_1)&space;R_x(q_0)" target="_blank"><img src="https://latex.codecogs.com/gif.latex?R&space;=&space;R_z(q_2)&space;R_y(q_1)&space;R_x(q_0)" title="R = R_z(q_2) R_y(q_1) R_x(q_0)" /></a>

so XYZ seems better to me. This convention is also used in HPP and URDF.

I don't like RPY because it is often used to mean Euler angles with no specific convention, so you never know what convention is behind.

gabrielebndn, 2019-06-05:
> From ZYX, I would have understood your second equation, so XYZ seems better to me. I don't like RPY because it is often used to mean Euler angles with no specific convention, so you never know what convention is behind.

I understand what you mean. The problem is, if we just silently change the convention in `JointSphericalZYX`, it risks to create *very* nasty bugs in code that is currently using it. I do not expect this code to be much (normally, `JointSpherical` is recommended, using a quaternion) but it is something we should avoid.
If we want to change it, the good way to proceed is to find a new name and deprecate the old joint I think. Some built-in sample models have the option of using `JointTranslation` + `JointSphericalZYX` instead of `JointFreeflyer`, this should be replaced, which is also a potential source of bugs for code using the sample models with this option, but as far as I know I expect this kind of code to be virtually non-existent.

jmirabel, 2019-06-05:
I think the people doing bio-engineering are using JointSphericalZYX. At least, they used it.

I suggested to change the name to JointSphericalXYZ.

gabrielebndn, 2019-06-05:
> I suggested to change the name to JointSphericalXYZ

I don't know whether this is more intuitive or not. Whether the appropriate name for the current convention is `JointSphericalXYZ` or `JointSphericalZYX` is subjective, I think. I, for one, think `JointSphericalZYX` is more intuitive because it reflects the order of the joints in the configuration. I would avoid introducing such a heavy change for something which I see dubious. My point was more on the angle ordering itself, with the rotation around z appearing first and the one around x last being highly unconventional, I think.

jmirabel, 2019-06-05:
I understand your point.

ZyYpXr or ZYX_ypr ?

jcarpent, 2019-06-05:
@gabrielebndn The choice originally comes from the book of Featherstone that I followed when implementing Pinocchio. He chose the ZYX convention, which as fine as the XYZ convention.

For me, it makes no difference to use one or the other convention. 

> This is, for instance, the convention assumed in the Python submodule `pinocchio.rpy` (which is slightly inconsistent). It is also the convention employed by the entities in sot-core.

In the same way, we do not have to reflect the choice made for Python for the C++ case. The fact that sot-core has its own convention is internal to this project, and do not have to influence Pinocchio choices.

To be the most generic as Pinocchio is expected to be, I suggest implementing a generic `SphericalEulerAngles<Z,Y,X>` joint which will be compatible with all the usual conventions, as done in Eigen with the Angle Axis convention. In this sense, we will support all the conventions we might need.
As most of the code is already present, the work required to achieve this modification won't take so much time.

gabrielebndn, 2019-06-05:
> He chose the ZYX convention, which is as fine as the XYZ convention.

Of course it's fine, it's absolutely equivalent. I just found the resulting configuration vector to be not intuitive, but if you disagree that's absolutely fine.

> To be the most generic as Pinocchio is expected to be, I suggest implementing a generic SphericalEulerAngles<Z,Y,X> joint

That's a great idea, but I see two drawbacks:
1) Supporting *all* Euler angles means to add 12, if not 24 (fixed+moving axes) concrete joint classes. As the algorithms need to compile for each joint, I would expect the compilation time to rise. Like, a lot. And it's already pretty high
2) The code for `JointSphericalZYX` is really specialized. For instance, this is an excerpt of the code in `calc()`:
```
data.M.rotation () << c0 * c1,
                c0 * s1 * s2 - s0 * c2,
                c0 * s1 * c2 + s0 * s2,
                s0 * c1,
                s0 * s1 * s2 + c0 * c2,
                s0 * s1 * c2 - c0 * s2,
                -s1,
                c1 * s2,
                c1 * c2;
```
writing a generic model would necessarily mean to use generic Eigen code, which will probably lose performance.

jmirabel, 2019-06-05:
> SphericalEulerAngles<Z,Y,X>

I don't think this solves this issue. I see no difference in `SphericalEulerAngles<Z,Y,X>` and `SphericalEulerAnglesZYX`. Both are as obscure regarding to the angle order. Or you must add another template argument...

jmirabel, 2019-06-05:
It may be even worse if you have something like `SphericalEulerAngles<2,1,0>`...

gabrielebndn, 2019-06-05:
> The choice originally comes from the book of Featherstone

O.O Forget me I was wrong! The formula suggested by Featherstone, thus implemented in Pinocchio, is

<a href="https://www.codecogs.com/eqnedit.php?latex=R_x(q_2)R_y(q_1)R_z(q_0)" target="_blank"><img src="https://latex.codecogs.com/gif.latex?R_x(q_2)R_y(q_1)R_z(q_0)" title="R_x(q_2)R_y(q_1)R_z(q_0)" /></a>

which is different from what I wrote above

gabrielebndn, 2019-06-05:
>  I see no difference between SphericalEulerAngles<Z,Y,X> and SphericalEulerAnglesZYX. Both are as obscure regarding to the angle order.

To be completely unambiguous with respect to the order, one should specify whether it is moving axes or rotation axes. For instance, this is what [tf.transformations](http://docs.ros.org/jade/api/tf/html/python/transformations.html) does:

> A triple of Euler angles can be applied/interpreted in 24 ways, which can be specified using a 4 character string or encoded 4-tuple:
> 
> Axes 4-string: e.g. ‘sxyz’ or ‘ryxy’
> 
> first character : rotations are applied to ‘s’tatic or ‘r’otating frame
> remaining characters : successive rotation axis ‘x’, ‘y’, or ‘z’

jcarpent, 2019-06-05:
> To be completely unambiguous with respect to the order, one should specify whether it is moving axes or rotation axes.

To be efficient, this has to be done with respect to the moving rotation axis as suggest in the Featherstone book. We should follow these rules in such a way the concatenation of 3 revolute joints behaves the same.

jcarpent, 2019-06-07:
@gabrielebndn @jmirabel What is the status of this discussion?

1. We need to improve the documentation to reflect the current observations
2. Do we really need to support other Euler Angles types?

gabrielebndn, 2019-06-07:
1. The behavior of `JointSphericalZYX` should be well documented. This can be done in conjunction with the documentation effort for all joints, as suggested in #794 

1.5 I think we should not change the name of `JointSphericalZYX`

2. I do not think we need to support all Euler angles types, it will just make Pinocchio much heavier. If someone *really* wants it, they may still achieve it through a `JointComposite`, for the mobile axis angles

2.5. If anything, we *might*, in my opinion, add a `JointModelSphericalRPY` or `JointModelRPY`, as roll-pitch-yaw is a fairly common configuration which is *not* achievable as a composition of joints, since the order of the products in
<a href="https://www.codecogs.com/eqnedit.php?latex=R_z(q_2)&space;R_y(q_1)&space;R_x(q_0)" target="_blank"><img src="https://latex.codecogs.com/gif.latex?R_z(q_2)&space;R_y(q_1)&space;R_x(q_0)" title="R_z(q_2) R_y(q_1) R_x(q_0)" /></a>
is *opposite* with respect to the one of the joints.

gabrielebndn, 2019-06-07:
If I might add: from what I can understand, Featherstone's joint model is specifical for a spherical socket, i.e. it is equivalent to a sequence JointModelRZ + JointModelRY + JointModelRX. This is the correct way to model it. For the orientation of a rigid object, in contexts where quaternions cannot be used, it is more common to employ the RPY formula I presented above

jcarpent, 2019-07-11:
@gabrielebndn Sorry for my late reply. Go ahead for your proposal. I do agree with your suggestions.

gabrielebndn, 2019-07-12:
@jcarpent, you mean implementing `JointModelSphericalRPY` and/or adding documentation?

jcarpent, 2019-07-12:
Implementing JointModelSphericalRPY ;)

gabrielebndn, 2019-07-12:
Ok. Just to be 100% clear. This post

> O.O Forgive me I was wrong! The formula suggested by Featherstone, thus implemented in Pinocchio, is
> 
> [![](https://camo.githubusercontent.com/0581a5c8e5d9350d4bf7b14f33711a2960450edf/68747470733a2f2f6c617465782e636f6465636f67732e636f6d2f6769662e6c617465783f525f7828715f3229525f7928715f3129525f7a28715f3029)](https://www.codecogs.com/eqnedit.php?latex=R_x(q_2)R_y(q_1)R_z(q_0))

was incorrect. I was mislead by the way the joint is presented in the book. My first intuition was right. The current `JointModelSphericalZYX` implements

<a href="https://www.codecogs.com/eqnedit.php?latex={}^{\lambda(i)}R_i&space;=&space;R_z(q_0)&space;R_y(q_1)&space;R_x(q_2)" target="_blank"><img src="https://latex.codecogs.com/gif.latex?{}^{\lambda(i)}R_i&space;=&space;R_z(q_0)&space;R_y(q_1)&space;R_x(q_2)" title="{}^{\lambda(i)}R_i = R_z(q_0) R_y(q_1) R_x(q_2)" /></a>

which is equivalent to `JointModelRZ` + `JointModelRY` + `JointModelRX`.

Instead, I am going to implement `JointModelSphericalRPY` as

<a href="https://www.codecogs.com/eqnedit.php?latex={}^{\lambda(i)}R_i&space;=&space;R_z(q_2)&space;R_y(q_1)&space;R_x(q_0)" target="_blank"><img src="https://latex.codecogs.com/gif.latex?{}^{\lambda(i)}R_i&space;=&space;R_z(q_2)&space;R_y(q_1)&space;R_x(q_0)" title="{}^{\lambda(i)}R_i = R_z(q_2) R_y(q_1) R_x(q_0)" /></a>

which is basically equivalent except that the position of X and Z is switched, and which is not currently realizable as a sequence of existing joint types.

My workload is pretty high at the moment, but I think I can quickly do it in the spare time by drawing inspiration from `JointModelSphericalZYX`

jcarpent, 2019-07-12:
Yes, I do agree that it should straightforward thanks to `JointModelSphericalZYX`

jcarpent, 2019-12-23:
This issue does not have any activity since a while. @gabrielebndn If you have time, do not hesitate to reopen and solve this issue.

closed 2019-12-23 by jcarpent