stack-of-tasks/pinocchio #810: SINCOS

jcarpent, 2019-06-06:
I see at least two sources of the problem:
1. The non-specialized implementation of `SINCOSAlgo` calls `std::sin()` and `std::cos()`:
    ```c++
    template<typename Scalar>
    struct SINCOSAlgo
    {
      static void run(const Scalar & a, Scalar * sa, Scalar * ca) 
      {   
        (*sa) = std::sin(a); (*ca) = std::cos(a);
      }   
    };
    ```
    It would be more appropriate to just do `sin(a)`, `cos(a)` and let [argument-dependent lookup](https://en.cppreference.com/w/cpp/language/adl) do the job. For `casadi::SX` argument type, the calls would resolve to `casadi::sin()` and `casadi::cos()`. Notice that if we rely on ADL, the following code fragment becomes unnecessary:
    ```c++
    #ifdef PINOCCHIO_WITH_CPPAD_SUPPORT
  
    /// Implementation for CppAD scalar types.
    template<typename Scalar>
    struct SINCOSAlgo< CppAD::AD<Scalar> >
    {
      static void run(const CppAD::AD<Scalar> & a, CppAD::AD<Scalar> * sa, CppAD::AD<Scalar> * ca)
      {
        (*sa) = CppAD::sin(a); (*ca) = CppAD::cos(a);
      }
    };
  
    #endif
    ```
2. The matrix decomposition algorithms in `Eigen::LLT` do inequality comparisons between scalar variables. If those variables are of type `casadi::SX`, the `operator >` is not defined (and can not be defined, because the variables are symbolic variables).

_Originally posted by @mkatliar in https://github.com/stack-of-tasks/pinocchio/issues/809#issuecomment-499509614_

jcarpent, 2019-06-06:
To really take care of both classic scalar types (double, float, ...) as well as overloaded types (CppAD::AD or casadi::SX).
Originally, we were using the sincos instruction to compute `sin` and `cos` at the same time, which is much more efficient than computing the two separated.
Of course, some recent compilers are able to guess it. 
So, maybe it would be worth to just allows ADL to work for each types correctly, by checking if the types are standard or not.

gabrielebndn, 2019-06-06:
There is a function is_fundamental in boost. Maybe the solution is to add a second template argument, which is boolean and computed through is_fundamental

mkatliar, 2019-06-06:
@gabrielebndn why would we need `is_fundamental()`? `SINCOSAlgo` is already specialized for `float`, `double` and `long double`. For other types (like `casadi::SX` of `CppAD::AD<Scalar>`) we can assume that the `sin()` and `cos()` are defined in corresponding namespaces.

gabrielebndn, 2019-06-06:
@mkatliar, I wanted to be generic with respect to any primitive type, but in effect float, double and long double cover all possible types of real numbers in C++ and it would not make sense at all to use integers, so that's probably useless, you are right

jcarpent, 2019-06-07:
Solved by #812.

closed 2019-06-07 by jcarpent