stack-of-tasks/pinocchio #815: Error when building fro source (python bindings)

alejandroastudillo, 2019-06-20:
Hi,
I have been trying to build pinocchio from source both from the `master` and `devel` branches but I haven't succeed. After the `make` command,  I always got the following error:
```
In file included from /home/alejandro/phd_software/pinocchio_source/pinocchio/build/include/pinocchio/bindings/python/multibody/joint/joints-variant.hpp:11:0,
                 from /home/alejandro/phd_software/pinocchio_source/pinocchio/bindings/python/multibody/joint/expose-joints.cpp:7:
/home/alejandro/phd_software/pinocchio_source/pinocchio/build/include/pinocchio/bindings/python/multibody/joint/joints-models.hpp: In function ‘pinocchio::JointModelComposite& pinocchio::python::addJoint_proxy(pinocchio::JointModelComposite&, const JointModelVariant&, const SE3&)’:
/home/alejandro/phd_software/pinocchio_source/pinocchio/build/include/pinocchio/bindings/python/multibody/joint/joints-models.hpp:79:118: error: call of overloaded ‘apply_visitor(pinocchio::python::JointModelCompositeAddJointVisitor, const JointModelVariant&)’ is ambiguous
       return boost::apply_visitor(JointModelCompositeAddJointVisitor(joint_composite,joint_placement), jmodel_variant);
                                                                                                                      ^
```
Do you have any idea on what is causing that error?
Thanks in advance

jcarpent, 2019-06-20:
On which OS are you working on? If you want, they are some instructions here: https://stack-of-tasks.github.io/pinocchio/download.html

jcarpent, 2019-06-20:
By the way, can you provide the complete log of the compilation? It will be easier to identify the compilation error.

alejandroastudillo, 2019-06-20:
Sorry for forgetting to mention the OS. I am working on Ubuntu 16.04.
I did succeed with the installation using `sudo apt install robotpkg-py27-pinocchio` or `sudo apt install robotpkg-py35-pinocchio`, and have been using pinocchio since some weeks ago. But now, I would like to go for the source builds since I would like to be up to date with the `devel` branch, or more specifically with the `topic/casadi` branch from your forked repo @jcarpent.

What I've seen from there is that the only error is the one I already added to my first comment in this issue.
[make_output_print.txt](https://github.com/stack-of-tasks/pinocchio/files/3309852/make_output_print.txt)

jcarpent, 2019-06-20:
Which version of boost are you using now? Because travis is happy when compiling `topic/casadi` under 16.04. It seems the compilation error is coming from your system set up.
Can you try on a clean 16.04 distribution?

alejandroastudillo, 2019-06-20:
These are the versions I am using:
```
14:37 $ cmake .. -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/opt/openrobots
-- eigen3 >= 3.0.5 is required.
-- Checking for module 'eigen3 >= 3.0.5'
--   Found eigen3 , version 3.2.92
-- Pkg-config module eigen3 v3.2.92 has been detected with success.
-- urdfdom >= 0.2.0 is required.
-- Checking for module 'urdfdom >= 0.2.0'
--   Found urdfdom , version 0.4.0
-- Pkg-config module urdfdom v0.4.0 has been detected with success.
-- PythonInterp: /usr/bin/python
-- PythonLibraries: /usr/lib/x86_64-linux-gnu/libpython2.7.so
-- PythonLibraryDirs: /usr/lib/x86_64-linux-gnu
-- PythonLibVersionString: 2.7.12
-- Boost version: 1.58.0
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   filesystem
--   serialization
--   system
--   python
--   unit_test_framework
-- Boost_PYTHON_LIBRARY: /usr/lib/x86_64-linux-gnu/libboost_python.so
-- eigenpy >= 1.4.4 is required.
-- Checking for module 'eigenpy >= 1.4.4'
--   Found eigenpy , version 1.5.1
-- Pkg-config module eigenpy v1.5.1 has been detected with success.
-- Doxygen rendering: using MathJax backend
-- Configuring done
-- Generating done
-- Build files have been written to: /home/alejandro/phd_software/pinocchio_source/pinocchio/build
```

> Can you try on a clean 16.04 distribution?
Not right now, but I will try asap.

BTW, should I remove all `robotpkg-*` packages before trying to build pinocchio from sources?

alejandroastudillo, 2019-06-20:
So, if I try to compile without Python interface (`-DBUILD_PYTHON_INTERFACE=OFF`), it is able to compile and my C++ examples seem to be working fine. Now, I have turned on the variable `BUILD_WITH_CASADI_SUPPORT` but cmake is not able to find my version of casadi. Can you tell me what additional command should I use to specify casadi's location? (sorry if this is a stupid question)

jcarpent, 2019-06-20:
> BTW, should I remove all `robotpkg-*` packages before trying to build pinocchio from sources?

Yes, otherwise you may have conflicts.

jcarpent, 2019-06-20:
> So, if I try to compile without Python interface (`-DBUILD_PYTHON_INTERFACE=OFF`), it is able to compile and my C++ examples seem to be working fine. Now, I have turned on the variable [...]

I have opened an issue in CasADi (https://github.com/casadi/casadi/pull/2446), to add the support of pkgconfig. If you install my version (master branch in https://github.com/jcarpent/casadi), you will be able to find casadi with Pinocchio

alejandroastudillo, 2019-06-20:
Thanks, I will try with that and see what happens.

jcarpent, 2019-06-24:
Any feedback?

alejandroastudillo, 2019-06-25:
I tried with casadi supporting pkgconfig, but when building pinocchio, the `casadi/casadi.hpp` was not found. I added casadi to pinocchio's CMakeLists.txt (using `find_library` and `include_directories`) and it is finding casadi now.
The new error is this one:
```
/home/alejandro/phd_software/pinocchio_source/pinocchio/build/include/pinocchio/math/casadi.hpp:46:16: error: invalid static_cast from type ‘const SX {aka const casadi::Matrix<casadi::SXElem>}’ to type ‘int’
         return static_cast<Scalar>(x);
                ^~~~~~~~~~~~~~~~~~~~~~
```
BTW, I will be able to try in a clean 16.04 distribution later today or tomorrow.

jcarpent, 2019-06-25:
Yes, I would suggest working with a clean Ubuntu environment.
Can you provide me the whole log of the compilation issue (you don't provide the whole information needed to investigate your issue)?

alejandroastudillo, 2019-06-25:
I just pulled the updated version of pinocchio from your topic/casadi branch. Here's the whole compilation log.
[make_output.txt](https://github.com/stack-of-tasks/pinocchio/files/3324174/make_output.txt)

jcarpent, 2019-06-25:
@alejandroastudillo I found the bug which is coming from Eigen in fact. They changed their mind since then. I fixed it in #821. I'm waiting for Travis and it will be merged on devel.

jcarpent, 2019-06-25:
This problem has been solved in #821.

alejandroastudillo, 2019-06-25:
Thank you very much @jcarpent. I was able to build pinocchio's `devel` branch from source (haven't tried with python interface). I think that I will test it tomorrow morning and let you know how it goes.

alejandroastudillo, 2019-06-26:
Dear @jcarpent. I have been testing the aba algorithm and forward kinematics interfacing casadi. Almost everything is working fine. However, since I need to calculate the casadi expressions which define the pose of a robot's end-effector, I must update the frame placements like this:
```
pinocchio::forwardKinematics(casadi_model,casadi_data,q_casadi);
pinocchio::updateFramePlacements(casadi_model,casadi_data);
```
or just: `pinocchio::framesForwardKinematics(casadi_model,casadi_data,q_casadi); `
taking into account that I already defined:
```
  typedef double                              Scalar;
  typedef casadi::SX                          CasadiScalar;

  typedef pinocchio::ModelTpl<Scalar>         Model;
  typedef Model::Data                         Data;

  typedef pinocchio::ModelTpl<CasadiScalar>   CasadiModel;
  typedef CasadiModel::Data                   CasadiData;

  ...

  CasadiModel casadi_model = model.cast<CasadiScalar>();
  CasadiData casadi_data(casadi_model);

  ...

  casadi::SX q_sx = casadi::SX::sym("q", model.nq);
  ConfigVectorCasadi q_casadi(model.nq);
  q_casadi = Eigen::Map<ConfigVectorCasadi>(static_cast< std::vector<CasadiScalar> >(q_sx).data(),model.nq,1);
```

But, whenever I uncomment the `pinocchio::updateFramePlacements(casadi_model,casadi_data)` call, my script is not compiling.

Attached there are both the compilation log and my test script.
* [frames_compilation_error_log.txt](https://github.com/stack-of-tasks/pinocchio/files/3329362/frames_compilation_error_log.txt)
* [test_script.cpp](https://github.com/stack-of-tasks/pinocchio/files/3329367/test_script.txt)

jcarpent, 2019-06-26:
Ok I will have a look and do the appropriate modifications. Thanks for reporting this new issue.

jcarpent, 2019-06-26:
@alejandroastudillo I've just made some fixes in #823.

alejandroastudillo, 2019-06-26:
Thank you very much @jcarpent. Built, tested, and working. 
I'll now try to build with python interface and see what happens.

alejandroastudillo, 2019-07-03:
Dear @jcarpent,

Calling cmake with:
```
cmake .. -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/opt/openrobots -DBUILD_PYTHON_INTERFACE=ON -DBUILD_WITH_CASADI_SUPPORT=ON
```
gave me an error related to not finding `numpy/numpyconfig.h`, but that error was not present when I tried calling cmake with:
```
cmake .. -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/opt/openrobots -DBUILD_PYTHON_INTERFACE=ON -DBUILD_WITH_CASADI_SUPPORT=ON  -DPYTHON_INCLUDE_DIR=/usr/include/python2.7 -DPYTHON_LIBRARY=/usr/lib/libpython2.7.so
```
But now, I am having an error related to the `apply_visitor`:
```
In file included from /home/alejandro/phd_software/pinocchio_source/pinocchio/build/include/pinocchio/bindings/python/multibody/joint/joints-variant.hpp:11:0,
                 from /home/alejandro/phd_software/pinocchio_source/pinocchio/bindings/python/multibody/joint/expose-joints.cpp:7:
/home/alejandro/phd_software/pinocchio_source/pinocchio/build/include/pinocchio/bindings/python/multibody/joint/joints-models.hpp: In function ‘pinocchio::JointModelComposite& pinocchio::python::addJoint_proxy(pinocchio::JointModelComposite&, const JointModelVariant&, const SE3&)’:
/home/alejandro/phd_software/pinocchio_source/pinocchio/build/include/pinocchio/bindings/python/multibody/joint/joints-models.hpp:79:118: error: call of overloaded ‘apply_visitor(pinocchio::python::JointModelCompositeAddJointVisitor, const JointModelVariant&)’ is ambiguous
       return boost::apply_visitor(JointModelCompositeAddJointVisitor(joint_composite,joint_placement), jmodel_variant);
                                                                                                                      ^
In file included from /usr/include/boost/variant/apply_visitor.hpp:16:0,
                 from /usr/include/boost/variant/detail/hash_variant.hpp:23,
                 from /usr/include/boost/variant/variant.hpp:34,
                 from /usr/include/boost/variant.hpp:17,
                 from /home/alejandro/phd_software/pinocchio_source/pinocchio/build/include/pinocchio/multibody/joint/joint-collection.hpp:11,
                 from /home/alejandro/phd_software/pinocchio_source/pinocchio/build/include/pinocchio/bindings/python/multibody/joint/joint-derived.hpp:11,
                 from /home/alejandro/phd_software/pinocchio_source/pinocchio/bindings/python/multibody/joint/expose-joints.cpp:6:
/usr/include/boost/variant/detail/apply_visitor_unary.hpp:82:1: note: candidate: typename Visitor::result_type boost::apply_visitor(const Visitor&, Visitable&) [with Visitor = pinocchio::python::JointModelCompositeAddJointVisitor; Visitable = const boost::variant<pinocchio::JointModelRevoluteTpl<double, 0, 0>, pinocchio::JointModelRevoluteTpl<double, 0, 1>, pinocchio::JointModelRevoluteTpl<double, 0, 2>, pinocchio::JointModelFreeFlyerTpl<double, 0>, pinocchio::JointModelPlanarTpl<double, 0>, pinocchio::JointModelRevoluteUnalignedTpl<double, 0>, pinocchio::JointModelSphericalTpl<double, 0>, pinocchio::JointModelSphericalZYXTpl<double, 0>, pinocchio::JointModelPrismaticTpl<double, 0, 0>, pinocchio::JointModelPrismaticTpl<double, 0, 1>, pinocchio::JointModelPrismaticTpl<double, 0, 2>, pinocchio::JointModelPrismaticUnalignedTpl<double, 0>, pinocchio::JointModelTranslationTpl<double, 0>, pinocchio::JointModelRevoluteUnboundedTpl<double, 0, 0>, pinocchio::JointModelRevoluteUnboundedTpl<double, 0, 1>, pinocchio::JointModelRevoluteUnboundedTpl<double, 0, 2>, boost::recursive_wrapper<pinocchio::JointModelCompositeTpl<double, 0, pinocchio::JointCollectionDefaultTpl> > >; typename Visitor::result_type = pinocchio::JointModelCompositeTpl<double>&]
 apply_visitor(const Visitor& visitor, Visitable& visitable)
 ^~~~~~~~~~~~~
/usr/include/boost/variant/detail/apply_visitor_unary.hpp:160:23: note: candidate: decltype(auto) boost::apply_visitor(const Visitor&, Visitable&, typename boost::disable_if<boost::detail::variant::has_result_type<Visitor> >::type*) [with Visitor = pinocchio::python::JointModelCompositeAddJointVisitor; Visitable = const boost::variant<pinocchio::JointModelRevoluteTpl<double, 0, 0>, pinocchio::JointModelRevoluteTpl<double, 0, 1>, pinocchio::JointModelRevoluteTpl<double, 0, 2>, pinocchio::JointModelFreeFlyerTpl<double, 0>, pinocchio::JointModelPlanarTpl<double, 0>, pinocchio::JointModelRevoluteUnalignedTpl<double, 0>, pinocchio::JointModelSphericalTpl<double, 0>, pinocchio::JointModelSphericalZYXTpl<double, 0>, pinocchio::JointModelPrismaticTpl<double, 0, 0>, pinocchio::JointModelPrismaticTpl<double, 0, 1>, pinocchio::JointModelPrismaticTpl<double, 0, 2>, pinocchio::JointModelPrismaticUnalignedTpl<double, 0>, pinocchio::JointModelTranslationTpl<double, 0>, pinocchio::JointModelRevoluteUnboundedTpl<double, 0, 0>, pinocchio::JointModelRevoluteUnboundedTpl<double, 0, 1>, pinocchio::JointModelRevoluteUnboundedTpl<double, 0, 2>, boost::recursive_wrapper<pinocchio::JointModelCompositeTpl<double, 0, pinocchio::JointCollectionDefaultTpl> > >; typename boost::disable_if<boost::detail::variant::has_result_type<Visitor> >::type = void]
 inline decltype(auto) apply_visitor(const Visitor& visitor, Visitable& visitable,
                       ^~~~~~~~~~~~~

```
The log of this compilation is here: [log_compilation.txt](https://github.com/stack-of-tasks/pinocchio/files/3355602/log_compilation.txt)

jcarpent, 2019-07-03:
A simple question, are you compiling with C++11 or C++14 on? The error coming from `/usr/include/boost/variant/detail/apply_visitor_unary.hpp:160` is raised only if flags C++11 or C++14 are activated. @alejandroastudillo Can you check your current configuration?

alejandroastudillo, 2019-07-04:
I didn't set any additional flag for `C++11` or `C++14`, but I was using `GCC 7.4.0` which needs no flag for enabling C++11 features. I went back to `GCC 5.5.0` and now the compilation with python bindings is completed without errors. Thank you very much.

It is worth clarifying that, with `GCC 7.4.0`, Pinocchio is building without errors only if the python interface is not being built.

jcarpent, 2019-07-04:
The issue here is that GCC >= 7.4.0 is complying by default with C++11 enable. Which is not a correct thing (made in the back of users), IMHO. It seems then the problem is coming from Boost 1.58 (on your computer).
In general, I would suggest people relying on the compiler which comes with the distribution, and not to try the most advanced one on an already old Ubuntu distribution (16.04).

@alejandroastudillo Do you think we might close this issue now your problem is solved?

alejandroastudillo, 2019-07-04:
Ok. Thank you very much for your advice and help.
Sure, I think this issue can be closed now.

closed 2019-07-04 by alejandroastudillo