stack-of-tasks/pinocchio #816: [python bindings] can't import eigenpy and Quaternion within the same script

stonneau, 2019-06-20:
Hi, with the latest version of Pinocchio, the following two lines won't work:

import eigenpy;
from pinocchio import Quaternion;
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-2-cac19b39cca1> in <module>()
----> 1 from pinocchio import Quaternion

ImportError: cannot import name Quaternion

Conversely, the other way around things seem fine. This is not high priority.

Thanks 

Steve

nim65s, 2019-06-20:
Quaternion is in eigenpy, isn't it ?

```
In [1]: import eigenpy

In [2]: eigenpy.Quaternion
Out[2]: eigenpy.Quaternion
```

stonneau, 2019-06-20:
It is, however the import is not mine, but comes from a file that I include.
If your answer is the answer to the issue, then I guess it means Quaternion must be removed from pinocchio don't you think ?

nim65s, 2019-06-20:
No, sorry, I misunderstood the question. Yes this is an issue, and yes this is minor :)

jcarpent, 2019-06-20:
I think the bug is related to EigenPy instead. It is due to the default exposition of Quaternion by the two applications. So if eigenpy is loaded first, then pinocchio will not expose it and the class Quaternion won't appear in the namespace of Pinocchio.

Thanks @stonneau for raising this bug.

jcarpent, 2019-06-20:
Solved by #817

closed 2019-06-20 by jcarpent