stack-of-tasks/pinocchio #824: Make pinocchio pip installable as python package

grozail, 2019-06-29:
Are there any thoughts on making pip-installable package?

nim65s, 2019-06-29:
Hi @grozail,

We already started discussions about that, yes.
This is not trivial, but not impossible, especially since [PEP 571](https://www.python.org/dev/peps/pep-0571/).

If I recall correctly, @jcarpent said that someone from his team wanted to work on that.
Otherwise, I could handle this, but not before fall, or even winter.

jcarpent, 2019-07-01:
Dear @grozail,

We are about to put **pinocchio** in `anaconda`. Does this feature fill your needs? Or `pip` is really your target and needs? Indeed, we are also planning put **pinocchio** inside, but this feature will be achieved later on.

grozail, 2019-07-01:
@nim65s I understand that this task is not trivial at all. Keeping in mind all the required dependencies.
@jcarpent Anaconda would be enough for now.
Maybe I will implement packaging by myself in free time.
Thanks for your work and attention.

jcarpent, 2019-07-15:
@grozail With @lesteve, we are currently working on the integration of Pinocchio inside conda.

jcarpent, 2019-07-23:
@grozail `EigenPy` is now inside conda-force (https://github.com/conda-forge/eigenpy-feedstock/).
Next step is Pinocchio.

grozail, 2019-07-23:
@jcarpent Nice to hear!

jcarpent, 2019-08-08:
@grozail After a long trip inside the twists and turns of Conda, and thanks to the amazing help of @lesteve and @wolfv, Pinocchio can now be installed through `conda`.
Please refer to https://github.com/conda-forge/pinocchio-feedstock#installing-pinocchio for the installation details.

jcarpent, 2019-08-08:
I will now close this issue as it is now solved. @grozail Feel free to reopen it if necessary.

closed 2019-08-08 by jcarpent

junhyeokahn, 2020-09-18:
How difficult it would be to make pip-installable package?
There are many packages which are not installed well with conda in my project so I would love to install pinocchio with pip.

wolfv, 2020-09-18:
you can install packages inside of conda with pip as well @junhyeokahn 

What packages are not working well for you?

jcarpent, 2020-09-18:
In addition, I don't understand what is not working well with conda @junhyeokahn. 
I've been using it for a while even when developping Pinocchio with the full dependencies behind, and it works pretty and allows me to test with isolated environments.

@junhyeokahn Could you elaborate more your issue?

jcarpent, 2020-09-18:
Concerning the pip installation of Pinocchio, there two possible intiatives:
- using the working but not yet finishied [conda-press](https://github.com/conda-incubator/conda-press) module, which will allow to export pip wheel from conda directly. Very smart idea, but I was not able to make it working with Pinocchio. @wolfv Do you have any insight on the current progress?
- @nim65s is currently working on making Pinocchio available on pip using manylinux. Maybe, he may provide some further comments on the progress?

nim65s, 2020-09-19:
This is a high priority tasks for me, and a proof of concept with everything working was already done this summer by @ozon2 : I expect this to be fully tested, integrated and available in a the coming month.

wolfv, 2020-09-20:
no unfortunately I don't know the current state of conda-press. I actually don't know exactly how it works. but it would be great if we could make that workflow work.

As far as I know, building wheels is a truly horrible experience :)

nim65s, 2020-10-13:
ref #1317