stack-of-tasks/pinocchio #831: pkg-config doesn't allow for conda environment

pariterre, 2019-07-15:
Hi there, 

I've spent some time trying to sort this, but it seems that pkg-config won't work with environment like anaconda. However, I am developing other libraries that required specific versions of boost and eigen and therefore I cannot afort to install libraries in my main /usr/ folder!
Is there something I am supposed to do so pkg-config will find the dependencies installed in CMAKE_INSTALL_PREFIX?

Thanks!

jcarpent, 2019-07-15:
Dear @pariterre,

With @lesteve, we are putting some similar efforts in the same direction than you.  See https://github.com/conda-forge/staged-recipes/pull/8820 for instance.
We are making a first trial for EigenPy and we will try Pinocchio next.

To help you, we would need some specific output of your current bugs?

In the meantime, I will add the generation of eigenpyConfig.cmake and pinocchioConfig.cmake to these projects.

Thanks again for your interest in Pinocchio,
Best,

Justin

nim65s, 2019-07-15:
Hi @pariterre 

You can install those libraries anywhere. You just need to set `PKG_CONFIG_PATH` afterwards, eg.:
`export PKG_CONFIG_PATH=~dev/lib/pkgconfig` if you want to install your eigen / boost in `~/dev`.

pariterre, 2019-07-15:
Thanks for your answer
In the mean time, I removed locally the "pkg" calls and changed them for find_package() and it sets the cmake properly (wrong path for boost though, because Boost_INCLUDE_DIRS is indeed not defined, but that is an easy fix)

Thanks for the workaround @nim65s . So far, I was unsuccessful using ```export``` for finding Eigen and Boost, but it did find eigenpy (compiled by hand, I didn't notice there was a conda-forge version!)
*Suggestion* : you could find if the user is on UNIX, and set for them the environment variable of PKG_CONFIG_PATH using ```set(ENV{PKG_CONFIG_PATH} ${CMAKE_INSTALL_PREFIX}/lib/pkgconfig)````

lesteve, 2019-07-16:
Inside a conda environment, it should be reasonably straightforward. For example this works for me for eigenpy:
```sh
# create conda environment
conda create -n test-eigenpy eigen cmake pkg-config -y
conda activate test-eigenpy

# get eigenpy source
git clone https://github.com/stack-of-tasks/eigenpy
cd eigenpy
git submodule update --init --recursive
mkdir build
cd build

# compile
PKG_CONFIG_PATH=$CONDA_PREFIX/share/pkgconfig cmake .. \
               -DCMAKE_BUILD_TYPE=Release -DCMAKE_PREFIX_PATH=$CONDA_PREFIX \
               -DCMAKE_INSTALL_PREFIX=$CONDA_PREFIX
make
make install
```

Note:
- `PKG_CONFIG_PATH` is needed for eigen which puts its `.pc` in a different place than other packages (at least inside a conda environment). The default location for `.pc` inside a conda environment is `$CONDA_PREFIX/lib/pkgconfig` and packages there are found automatically.
- some problems have been fixed quite recently for `pinocchio` (e.g. compilation with boost 1.70 or Python 3) so you may need to try the `devel` branch. For `eigenpy`, everything should be fixed in the last release of `eigenpy` (1.5.4) should be fine.

pariterre, 2019-07-16:
Hi!

From the suggestion of @lesteve, I could actually making things work. However, there is no need to set by hand the PKG_CONFIG_PATH by hand. The relevant lines are : 
```bash
conda activate $NAME_ENVIRONMENT
conda install pkg-config
```
That is to say that pkg-config must be install inside the environment (one must not use the pkg-config installed on the root) **AND** the environment **must** be activated while running cmake. Like this, pkg is properly configured to find Eigen3 without other workaround. I suggest that it is added to your doc, but since it is not in any way related to your code, but from the conda-user's knowledge on how to use pkg-config in the conda environment, maybe this is not necessary...

FYI - I am still getting the following error raised from SEARCH_FOR_BOOST from the devel branch:
```bash
 CMake Error at
 /home/pariterre/Programmation/miniconda3/envs/pinocchio/lib/cmake/Boost-1.70.0
 /BoostConfig.cmake:95 (find_package):
   Found package configuration file:

 /home/pariterre/Programmation/miniconda3/envs/pinocchio/lib/cmake/boost_filesy
 stem-1.70.0/boost_filesystem-config.cmake

   but it set boost_filesystem_FOUND to FALSE so package "boost_filesystem" is
   considered to be NOT FOUND.  Reason given by package:

   No suitable build variant has been found.

 Call Stack (most recent call first):

 /home/pariterre/Programmation/miniconda3/envs/pinocchio/lib/cmake/Boost-1.70.0
 /BoostConfig.cmake:124 (boost_find_dependency)
   /usr/share/cmake-3.12/Modules/FindBoost.cmake:261 (find_package)
   cmake/boost.cmake:104 (FIND_PACKAGE)
   CMakeLists.txt:179 (SEARCH_FOR_BOOST)
```
boost was install from conda-forge channel. 

I don't think it is relevant on this thread though... Do you want me to close this issue and start a new one?

Thanks for your time!

lesteve, 2019-07-16:
Your error looks like you are not using the `pinocchio` `devel` branch, it seems very similar to the one I was having a few weeks ago before things got fixed on `devel` (a few days ago roughly).

Things to try:
* `git submodule --update --init --recursive` to make sure the cmake-related things utilities have the fixed version
* I am a `cmake` newbie so I don't know whether there is a nicer way to do it but I do `git lean -xdf` to make sure I start the build from scratch (`cmake` seems to be caching some steps)

lesteve, 2019-07-16:
And good point about having pkg-config installed, you don't need PKG_CONFIG_PATH anymore, thanks for this, this will simplify the conda recipe.

pariterre, 2019-07-16:
I am using GitKraken which initiate submodules automatically. But in one of my projects (biorbd, ezc3d) I actually run the submodule initiation command if the user didn't run it themselves while they run CMake. I think this is a nice way to ease things to people :)

I am on commit id : 0853bfe732fdf83c17405b919f07871ce3ea9ee8
Do you actually build cmake and run it from the external dependencies? I tried running cmake from the conda-forge channel, but I get the same error message.

lesteve, 2019-07-16:
> I am on commit id : 0853bfe

This commit is 6 days ago (maybe a copy and paste problem from you part). Can you try the latest commit on the `devel` branch i.e. (at the time of writing) https://github.com/stack-of-tasks/pinocchio/commit/0bc84d286cf47afbff7e246fa35271392f77ee60.

To try to tackle a slightly simpler problem, does the snippet I posted for you for eigenpy (rather than pinocchio) in https://github.com/stack-of-tasks/pinocchio/issues/831#issuecomment-511692617 work for you?

lesteve, 2019-07-16:
> To try to tackle a slightly simpler problem, does the snippet I posted for you for eigenpy (rather than pinocchio) in #831 (comment) work for you?

Silly suggestion since boost seems to be the issue and eigenpy is not using boost. Anyway if you still have problems, it would be great if you can post a reproducible snippet similar to https://github.com/stack-of-tasks/pinocchio/issues/831#issuecomment-511692617.

pariterre, 2019-07-16:
Using 
```bash
conda activate pinocchio
cmake .. -DCMAKE_INSTALL_PREFIX=$CONDA_PREFIX
```
works for eigenpy

For the commit, I forgot that I forked pinocchio and that I was on my fork. I am having trouble with Git right now, so in few minutes, I will test using the latest version of pinocchio

pariterre, 2019-07-16:
Allright, 
So with your last commit, everything seems to work :)
I tried using eigenpy compiled by hand (didn't tried the conda-forge channel)
You can close the issue!

jcarpent, 2019-07-16:
Sounds great.
Thanks for using Pinocchio and your user feedback.

closed 2019-07-16 by jcarpent

wilderfield, 2020-02-06:
I install pkg-config in the conda environment, and all problems go away...

lesteve, 2020-02-07:
@wilderfield note `pinocchio` is available from `conda-forge`:
```
conda install -c conda-forge pinocchio
```

so you don't have to compile pinocchio yourself unless of course you have a good reason to (for example, you care about some features in the `devel` branch).

> I install pkg-config in the conda environment, and all problems go away...

This is what I observed as well, e.g. I install `pkg-config` in the conda environment in  https://github.com/stack-of-tasks/pinocchio/issues/831#issuecomment-511692617.