stack-of-tasks/pinocchio #832: Hints on automatic differentiation and code generation

alaurenzi, 2019-07-15:
Hi all, and first off thanks for sharing this work.
I was looking for a multibody kinematics/dynamics library that can be integrated in a symbolic framework (e.g. Casadi) for optimal control purposes.  I have seen that the implementation of pinocchio is fully templatized on the scalar type, which allows to use it for code generation of derivatives via e.g. CppAD, however I couldn't find any example on how to actually do it. 

So, I tried first of all to compile the simple snippet below, where essentially I build a pinocchio model which uses `float` as scalar type, just to get familiar with the Scalar template.

``` c++
int main()
{

    std::string urdf_path = "/some/path/to/urdf.urdf";

    pinocchio::ModelTpl<float> model;

    pinocchio::urdf::buildModel(urdf_path, model);

    VectorXf q;
    q.resize(model.nq);

    pinocchio::DataTpl<float> data(model);

    pinocchio::forwardKinematics(model, data, q);

    return 0;
}
```

This code snipped does not compile, with a very long output from gcc from which I understand that some conversions are failing (e.g. `SE3Tpl<float>` to `SE3Tpl<double>`).

```
/home/.../include/pinocchio/parsers/urdf/model.hxx:541: error: invalid initialization of reference of type ‘const SE3& {aka const pinocchio::SE3Tpl<float, 0>&}’ from expression of type ‘pinocchio::SE3Tpl<double, 0>’
         addFixedJointAndBody(model, 0, SE3::Identity(), "root_joint",
                             ^
```

Is this the expected behavior? Am I missing something? I would be very glad if you could point me to some example on how to use pinocchio for code generation of derivatives!

Thanks in advance!
Arturo

PS: I'm using the latest master on a standard Ubuntu 16.04 system (compiler is GCC)

jcarpent, 2019-07-15:
Dear @alaurenzi,

Thanks a lot for using our codes.
Some basic examples of using Pinocchio with CppAD are located in the unit tests (`cppad-*` files). I will try to provide a complete set of examples asap.

Concerning your code, the idea is to cast a standard model (using doubles) to your desired types as follows:
```
int main()
{

    std::string urdf_path = "/some/path/to/urdf.urdf";
   
    pinocchio::Model model_double;
    pinocchio::urdf::buildModel(urdf_path, model_double);

    pinocchio::ModelTpl<float> model = model_double.cast<float>();

    VectorXf q;
    q.resize(model.nq);

    pinocchio::DataTpl<float> data(model);

    pinocchio::forwardKinematics(model, data, q);

    return 0;
}
```

alaurenzi, 2019-07-16:
Thanks for the prompt answer. I still have compilation issues. Below you can find the exact snippet:
```c++
#define PINOCCHIO_URDFDOM_TYPEDEF_SHARED_PTR
#include <pinocchio/parsers/urdf.hpp>
#include <pinocchio/algorithm/kinematics.hpp>

int main()
{
    std::string urdf_path = "/home/arturo/Code/advr-superbuild/configs/CentauroConfig/urdf/centauro.urdf";

    pinocchio::Model model_double;
    pinocchio::urdf::buildModel(urdf_path, model_double);

    pinocchio::ModelTpl<float> model = model_double.cast<float>();

    Eigen::VectorXf q;
    q.resize(model.nq);

    pinocchio::DataTpl<float> data(model);

    pinocchio::forwardKinematics(model, data, q);

    return 0;

}
```

The error is 
```
/home/.../include/pinocchio/spatial/se3-tpl.hpp:174: error: no matching function for call to ‘pinocchio::ConstraintTpl<-1, float, 0>::se3Action(const pinocchio::SE3Tpl<float, 0>&) const’
       return d.se3Action(*this);
                               ^
```
caused by a conversion attempt
```
/home/.../include/pinocchio/multibody/constraint-generic.hpp:130: note:   no known conversion for argument 1 from ‘const pinocchio::SE3Tpl<float, 0>’ to ‘const SE3& {aka const pinocchio::SE3Tpl<double, 0>&}’
```

The strange thing is that compiling pinocchio with AUTODIFF_SUPPORT results in the same error. Any clue?

jcarpent, 2019-07-16:
Dear @alaurenzi. I will have a look this afternoon. Best

alaurenzi, 2019-07-16:
Latest news: I tried the code above (including unit tests) on the devel branch and it compiles!

jcarpent, 2019-07-16:
Good to hear this news. Sorry, finally I did have time to take a look at your issue.
Hope everything works fine for you. I will now close the issue.

Thanks again for using Pinocchio.

closed 2019-07-16 by jcarpent