stack-of-tasks/pinocchio #837: Compilation problem (boost python) of a thirdparty library that depends on `pinocchio::Model` in v2.1.5

cmastalli, 2019-07-19:
I detected a problem in v2.1.5 during the boost python compilation of a class with a `pinocchio::Model` object. This problem doesn't appears in v2.1.3.

```bash
In file included from /usr/include/boost/variant/variant.hpp:31:0,
                 from /usr/include/boost/variant.hpp:17,
                 from /local/users/cmastall/include/pinocchio/multibody/joint/joint-collection.hpp:11,
                 from /local/users/cmastall/include/pinocchio/multibody/joint/joint-generic.hpp:8,
                 from /local/users/cmastall/include/pinocchio/multibody/model.hpp:17,
                 from /devel/crocoddyl/include/crocoddyl/core/states/state-euclidean.hpp:13,
                 from /devel/crocoddyl/bindings/python/crocoddyl/core/states/state-euclidean.hpp:12,
                 from /devel/crocoddyl/bindings/python/crocoddyl/core.hpp:18,
                 from /devel/crocoddyl/bindings/python/crocoddyl/crocoddyl.cpp:14:
/usr/include/boost/variant/detail/make_variant_list.hpp: In instantiation of ‘struct boost::detail::variant::make_variant_list<pinocchio::JointModelRevoluteTpl<double, 0, 0>, pinocchio::JointModelRevoluteTpl<double, 0, 1>, pinocchio::JointModelRevoluteTpl<double, 0, 2>, pinocchio::JointModelMimic<pinocchio::JointModelRevoluteTpl<double, 0, 0> >, pinocchio::JointModelMimic<pinocchio::JointModelRevoluteTpl<double, 0, 1> >, pinocchio::JointModelMimic<pinocchio::JointModelRevoluteTpl<double, 0, 2> >, pinocchio::JointModelFreeFlyerTpl<double, 0>, pinocchio::JointModelPlanarTpl<double, 0>, pinocchio::JointModelRevoluteUnalignedTpl<double, 0>, pinocchio::JointModelSphericalTpl<double, 0>, pinocchio::JointModelSphericalZYXTpl<double, 0>, pinocchio::JointModelPrismaticTpl<double, 0, 0>, pinocchio::JointModelPrismaticTpl<double, 0, 1>, pinocchio::JointModelPrismaticTpl<double, 0, 2>, pinocchio::JointModelPrismaticUnalignedTpl<double, 0>, pinocchio::JointModelTranslationTpl<double, 0>, pinocchio::JointModelRevoluteUnboundedTpl<double, 0, 0>, pinocchio::JointModelRevoluteUnboundedTpl<double, 0, 1>, pinocchio::JointModelRevoluteUnboundedTpl<double, 0, 2>, pinocchio::JointModelRevoluteUnboundedUnalignedTpl<double, 0>, boost::recursive_wrapper<pinocchio::JointModelCompositeTpl<double, 0, pinocchio::JointCollectionDefaultTpl> > >’:
/usr/include/boost/mpl/eval_if.hpp:38:31:   required from ‘struct boost::mpl::eval_if<boost::variant<pinocchio::JointModelRevoluteTpl<double, 0, 0>, pinocchio::JointModelRevoluteTpl<double, 0, 1>, pinocchio::JointModelRevoluteTpl<double, 0, 2>, pinocchio::JointModelMimic<pinocchio::JointModelRevoluteTpl<double, 0, 0> >, pinocchio::JointModelMimic<pinocchio::JointModelRevoluteTpl<double, 0, 1> >, pinocchio::JointModelMimic<pinocchio::JointModelRevoluteTpl<double, 0, 2> >, pinocchio::JointModelFreeFlyerTpl<double, 0>, pinocchio::JointModelPlanarTpl<double, 0>, pinocchio::JointModelRevoluteUnalignedTpl<double, 0>, pinocchio::JointModelSphericalTpl<double, 0>, pinocchio::JointModelSphericalZYXTpl<double, 0>, pinocchio::JointModelPrismaticTpl<double, 0, 0>, pinocchio::JointModelPrismaticTpl<double, 0, 1>, pinocchio::JointModelPrismaticTpl<double, 0, 2>, pinocchio::JointModelPrismaticUnalignedTpl<double, 0>, pinocchio::JointModelTranslationTpl<double, 0>, pinocchio::JointModelRevoluteUnboundedTpl<double, 0, 0>, pinocchio::JointModelRevoluteUnboundedTpl<double, 0, 1>, pinocchio::JointModelRevoluteUnboundedTpl<double, 0, 2>, pinocchio::JointModelRevoluteUnboundedUnalignedTpl<double, 0>, boost::recursive_wrapper<pinocchio::JointModelCompositeTpl<double, 0, pinocchio::JointCollectionDefaultTpl> > >::is_sequence_based_, pinocchio::JointModelRevoluteTpl<double, 0, 0>, boost::detail::variant::make_variant_list<pinocchio::JointModelRevoluteTpl<double, 0, 0>, pinocchio::JointModelRevoluteTpl<double, 0, 1>, pinocchio::JointModelRevoluteTpl<double, 0, 2>, pinocchio::JointModelMimic<pinocchio::JointModelRevoluteTpl<double, 0, 0> >, pinocchio::JointModelMimic<pinocchio::JointModelRevoluteTpl<double, 0, 1> >, pinocchio::JointModelMimic<pinocchio::JointModelRevoluteTpl<double, 0, 2> >, pinocchio::JointModelFreeFlyerTpl<double, 0>, pinocchio::JointModelPlanarTpl<double, 0>, pinocchio::JointModelRevoluteUnalignedTpl<double, 0>, pinocchio::JointModelSphericalTpl<double, 0>, pinocchio::JointModelSphericalZYXTpl<double, 0>, pinocchio::JointModelPrismaticTpl<double, 0, 0>, pinocchio::JointModelPrismaticTpl<double, 0, 1>, pinocchio::JointModelPrismaticTpl<double, 0, 2>, pinocchio::JointModelPrismaticUnalignedTpl<double, 0>, pinocchio::JointModelTranslationTpl<double, 0>, pinocchio::JointModelRevoluteUnboundedTpl<double, 0, 0>, pinocchio::JointModelRevoluteUnboundedTpl<double, 0, 1>, pinocchio::JointModelRevoluteUnboundedTpl<double, 0, 2>, pinocchio::JointModelRevoluteUnboundedUnalignedTpl<double, 0>, boost::recursive_wrapper<pinocchio::JointModelCompositeTpl<double, 0, pinocchio::JointCollectionDefaultTpl> > > >’
/usr/include/boost/variant/variant.hpp:1117:17:   required from ‘class boost::variant<pinocchio::JointModelRevoluteTpl<double, 0, 0>, pinocchio::JointModelRevoluteTpl<double, 0, 1>, pinocchio::JointModelRevoluteTpl<double, 0, 2>, pinocchio::JointModelMimic<pinocchio::JointModelRevoluteTpl<double, 0, 0> >, pinocchio::JointModelMimic<pinocchio::JointModelRevoluteTpl<double, 0, 1> >, pinocchio::JointModelMimic<pinocchio::JointModelRevoluteTpl<double, 0, 2> >, pinocchio::JointModelFreeFlyerTpl<double, 0>, pinocchio::JointModelPlanarTpl<double, 0>, pinocchio::JointModelRevoluteUnalignedTpl<double, 0>, pinocchio::JointModelSphericalTpl<double, 0>, pinocchio::JointModelSphericalZYXTpl<double, 0>, pinocchio::JointModelPrismaticTpl<double, 0, 0>, pinocchio::JointModelPrismaticTpl<double, 0, 1>, pinocchio::JointModelPrismaticTpl<double, 0, 2>, pinocchio::JointModelPrismaticUnalignedTpl<double, 0>, pinocchio::JointModelTranslationTpl<double, 0>, pinocchio::JointModelRevoluteUnboundedTpl<double, 0, 0>, pinocchio::JointModelRevoluteUnboundedTpl<double, 0, 1>, pinocchio::JointModelRevoluteUnboundedTpl<double, 0, 2>, pinocchio::JointModelRevoluteUnboundedUnalignedTpl<double, 0>, boost::recursive_wrapper<pinocchio::JointModelCompositeTpl<double, 0, pinocchio::JointCollectionDefaultTpl> > >’
/local/users/cmastall/include/pinocchio/multibody/joint/joint-generic.hpp:129:10:   required from ‘struct pinocchio::JointModelTpl<double>’
/usr/include/c++/5/bits/stl_vector.h:161:9:   required from ‘std::_Vector_base<_Tp, _Alloc>::~_Vector_base() [with _Tp = pinocchio::JointModelTpl<double>; _Alloc = Eigen::aligned_allocator<pinocchio::JointModelTpl<double> >]’
/usr/include/c++/5/bits/stl_vector.h:425:33:   required from ‘std::vector<_Tp, _Alloc>::~vector() [with _Tp = pinocchio::JointModelTpl<double>; _Alloc = Eigen::aligned_allocator<pinocchio::JointModelTpl<double> >]’
/local/users/cmastall/include/pinocchio/container/aligned-vector.hpp:21:12:   [ skipping 7 instantiation contexts, use -ftemplate-backtrace-limit=0 to disable ]
/usr/include/boost/python/object/class_metadata.hpp:229:55:   required from ‘static void boost::python::objects::class_metadata<T, X1, X2, X3>::register_aux2(T2*, Callback) [with T2 = crocoddyl::StateVector; Callback = boost::integral_constant<bool, false>; T = crocoddyl::StateVector; X1 = boost::python::bases<crocoddyl::StateAbstract>; X2 = boost::python::detail::not_specified; X3 = boost::python::detail::not_specified]’
/usr/include/boost/python/object/class_metadata.hpp:219:38:   required from ‘static void boost::python::objects::class_metadata<T, X1, X2, X3>::register_aux(void*) [with T = crocoddyl::StateVector; X1 = boost::python::bases<crocoddyl::StateAbstract>; X2 = boost::python::detail::not_specified; X3 = boost::python::detail::not_specified]’
/usr/include/boost/python/object/class_metadata.hpp:205:37:   required from ‘static void boost::python::objects::class_metadata<T, X1, X2, X3>::register_() [with T = crocoddyl::StateVector; X1 = boost::python::bases<crocoddyl::StateAbstract>; X2 = boost::python::detail::not_specified; X3 = boost::python::detail::not_specified]’
/usr/include/boost/python/class.hpp:497:28:   required from ‘void boost::python::class_<T, X1, X2, X3>::initialize(const DefVisitor&) [with DefVisitor = boost::python::init_base<boost::python::init<int> >; W = crocoddyl::StateVector; X1 = boost::python::bases<crocoddyl::StateAbstract>; X2 = boost::python::detail::not_specified; X3 = boost::python::detail::not_specified]’
[... cut ...] ot_specified]’
/devel/crocoddyl/bindings/python/crocoddyl/core/states/state-euclidean.hpp:30:33:   required from here
/usr/include/boost/variant/variant.hpp:2332:43: error: using invalid field ‘boost::variant<T0, TN>::storage_’
         return internal_apply_visitor_impl(
                                           ^
/usr/include/boost/variant/variant.hpp:2334:13: error: return-statement with a value, in function returning 'void' [-fpermissive]
```

**I don't have any clue of what it's going on**

jcarpent, 2019-07-21:
I think the issue is coming from your configuration setup, not from Pinocchio itself.
The only we can advise is to build Pinocchio from a clean build directory.

closed 2019-07-21 by jcarpent

cmastalli, 2019-07-22:
This problem persists after fixing issues with my setup. I believe it's needed to include some header file whenever you want to binding a code using `pinocchio::Model`.

cmastalli, 2019-07-22:
The new error message is:

```bash
In file included from /local/users/cmastall/include/pinocchio/multibody/joint/joint-generic.hpp:8:0,
                 from /local/users/cmastall/include/pinocchio/multibody/model.hpp:17,
                 from /devel/crocoddyl/include/crocoddyl/multibody/states/state-multibody.hpp:13,
                 from /devel/crocoddyl/bindings/python/crocoddyl/multibody/states/state-multibody.hpp:12,
                 from /devel/crocoddyl/bindings/python/crocoddyl/multibody.hpp:12,
                 from /devel/crocoddyl/bindings/python/crocoddyl/crocoddyl.cpp:15:
/local/users/cmastall/include/pinocchio/multibody/joint/joint-collection.hpp:82:5: error: wrong number of template arguments (21, should be at least 0)
     > JointModelVariant;
     ^
In file included from /usr/include/boost/variant/variant.hpp:26:0,
                 from /usr/include/boost/variant.hpp:17,
                 from /local/users/cmastall/include/pinocchio/multibody/joint/joint-collection.hpp:11,
                 from /local/users/cmastall/include/pinocchio/multibody/joint/joint-generic.hpp:8,
                 from /local/users/cmastall/include/pinocchio/multibody/model.hpp:17,
                 from /devel/crocoddyl/include/crocoddyl/multibody/states/state-multibody.hpp:13,
                 from /devel/crocoddyl/bindings/python/crocoddyl/multibody/states/state-multibody.hpp:12,
                 from /devel/crocoddyl/bindings/python/crocoddyl/multibody.hpp:12,
                 from /devel/crocoddyl/bindings/python/crocoddyl/crocoddyl.cpp:15:
/usr/include/boost/variant/variant_fwd.hpp:293:53: note: provided for ‘template<class T0, class T1, class T2, class T3, class T4, class T5, class T6, class T7, class T8, class T9, class T10, class T11, class T12, class T13, class T14, class T15, class T16, class T17, class T18, class T19> class boost::variant’
 template < BOOST_VARIANT_AUX_DECLARE_PARAMS > class variant;
                                                     ^
In file included from /local/users/cmastall/include/pinocchio/multibody/joint/joint-generic.hpp:8:0,
                 from /local/users/cmastall/include/pinocchio/multibody/model.hpp:17,
                 from /devel/crocoddyl/include/crocoddyl/multibody/states/state-multibody.hpp:13,
                 from /devel/crocoddyl/bindings/python/crocoddyl/multibody/states/state-multibody.hpp:12,
                 from /devel/crocoddyl/bindings/python/crocoddyl/multibody.hpp:12,
                 from /devel/crocoddyl/bindings/python/crocoddyl/crocoddyl.cpp:15:
/local/users/cmastall/include/pinocchio/multibody/joint/joint-collection.hpp:144:5: error: wrong number of template arguments (21, should be at least 0)
     > JointDataVariant;
     ^
In file included from /usr/include/boost/variant/variant.hpp:26:0,
                 from /usr/include/boost/variant.hpp:17,
                 from /local/users/cmastall/include/pinocchio/multibody/joint/joint-collection.hpp:11,
                 from /local/users/cmastall/include/pinocchio/multibody/joint/joint-generic.hpp:8,
                 from /local/users/cmastall/include/pinocchio/multibody/model.hpp:17,
                 from /devel/crocoddyl/include/crocoddyl/multibody/states/state-multibody.hpp:13,
                 from /devel/crocoddyl/bindings/python/crocoddyl/multibody/states/state-multibody.hpp:12,
                 from /devel/crocoddyl/bindings/python/crocoddyl/multibody.hpp:12,
                 from /devel/crocoddyl/bindings/python/crocoddyl/crocoddyl.cpp:15:
/usr/include/boost/variant/variant_fwd.hpp:293:53: note: provided for ‘template<class T0, class T1, class T2, class T3, class T4, class T5, class T6, class T7, class T8, class T9, class T10, class T11, class T12, class T13, class T14, class T15, class T16, class T17, class T18, class T19> class boost::variant’
 template < BOOST_VARIANT_AUX_DECLARE_PARAMS > class variant;
                                                     ^
In file included from /local/users/cmastall/include/pinocchio/multibody/model.hpp:17:0,
                 from /devel/crocoddyl/include/crocoddyl/multibody/states/state-multibody.hpp:13,
                 from /devel/crocoddyl/bindings/python/crocoddyl/multibody/states/state-multibody.hpp:12,
                 from /devel/crocoddyl/bindings/python/crocoddyl/multibody.hpp:12,
                 from /devel/crocoddyl/bindings/python/crocoddyl/crocoddyl.cpp:15:
/local/users/cmastall/include/pinocchio/multibody/joint/joint-generic.hpp: In instantiation of ‘struct pinocchio::JointModelTpl<double>’:
/usr/include/c++/5/bits/stl_vector.h:161:9:   required from ‘std::_Vector_base<_Tp, _Alloc>::~_Vector_base() [with _Tp = pinocchio::JointModelTpl<double>; _Alloc = Eigen::aligned_allocator_indirection<pinocchio::JointModelTpl<double> >]’
/usr/include/c++/5/bits/stl_vector.h:425:33:   required from ‘std::vector<_Tp, _Alloc>::~vector() [with _Tp = pinocchio::JointModelTpl<double>; _Alloc = Eigen::aligned_allocator_indirection<pinocchio::JointModelTpl<double> >]’
/usr/include/eigen3/Eigen/src/StlSupport/StdVector.h:72:9:   required from ‘boost::python::objects::value_holder<Value>::value_holder(PyObject*, A0) [with A0 = boost::reference_wrapper<const crocoddyl::StateMultibody>; Value = crocoddyl::StateMultibody; PyObject = _object]’
/usr/include/boost/python/object/make_instance.hpp:71:48:   required from ‘static Holder* boost::python::objects::make_instance<T, Holder>::construct(void*, PyObject*, boost::reference_wrapper<const T>) [with T = crocoddyl: [... cut ...] ateMultibody; X1 = boost::python::bases<crocoddyl::StateAbstract>; X2 = boost::python::detail::not_specified; X3 = boost::python::detail::not_specified]’
/devel/crocoddyl/bindings/python/crocoddyl/multibody/states/state-multibody.hpp:32:99:   required from here
/local/users/cmastall/include/pinocchio/multibody/joint/joint-generic.hpp:129:10: error: base type ‘pinocchio::JointCollectionDefaultTpl<double>::JointModelVariant {aka int}’ fails to be a struct or class type
   struct JointModelTpl

```

jcarpent, 2019-07-23:
Which command do you run?

cmastalli, 2019-07-23:
I compiled Crocoddyl with Python bindings. The branch is cpp_devel. Maybe do you want to move this discussion here https://gepgitlab.laas.fr/loco-3d/crocoddyl/issues/200?

jcarpent, 2019-07-23:
I think it depends on the order of file inclusions.
You should first include `pincchio/fwd.hpp` before anything to correctly setup the default size of the boost::variant.

cmastalli, 2019-07-23:
I just tried this, and it doesn't solve my issue :(

Any other suggestion?

jcarpent, 2019-07-23:
The easiest way for me to debug your current setup would be to provide a docker container with the issue. Can you do it?

cmastalli, 2019-07-29:
I asked @nim65s. As soon as possible we'll share it with you.