stack-of-tasks/pinocchio #838: Error when parsing a URDF

PepMS, 2019-07-22:
When I try to load my urdf file with the following code:

```
import pinocchio as pin
import math
from pinocchio.robot_wrapper import RobotWrapper
from pinocchio.utils import *
from time import sleep

model_path = '${HOME}/robotics/test/uam_test/kinton_description'
mesh_dir = model_path
urdf_filename = "kinton_arm.urdf"
urdf_model_path = model_path + "/urdf/" + urdf_filename
robot = RobotWrapper.BuildFromURDF(urdf_model_path, ['${HOME}/robotics/test/uam_test'])
```

I get the following error:
```
File "/opt/openrobots/lib/python2.7/site-packages/pinocchio/robot_wrapper.py", line 43, in _buildGeomFromUrdf
    return pin.buildGeomFromUrdf(model, filename, dirs, geometryType, meshLoader)
ValueError: Mesh Link1.000 has a face with 2 vertices. This is not supported
Node name is: Link1_001
Mesh index: 0
Face index: 0
```
I thought I could have a wrong urdf file. I actually do not know how to check that but I load the model with [this app](https://github.com/OTL/urdf-viz) and everything seemed fine.

If you could point me where to check would be very nice. I'll keep investigating!

Thank you very much,

Pep

PS: I attach in a zip file with the urdf
[kinton_description.zip](https://github.com/stack-of-tasks/pinocchio/files/3416896/kinton_description.zip)

jcarpent, 2019-07-22:
It seems this error is coming from hpp-fcl (which is forked version of the original FCL project).
The error seems to come from your files link1.{dae,stl} which define a face with only two vertices.

Can you check it?
I think certainly @jmirabel might provide an help on this issue.

PepMS, 2019-07-22:
Ok, I think it is fixed.

I've repaired the .dae mesh files with FreeCAD and the problem is solved. Robot shown in gepetto-gui. :smile:

Thanks!

jcarpent, 2019-07-22:
Very nice. Great continuation and hope you will enjoy Pinocchio!

closed 2019-07-22 by jcarpent