stack-of-tasks/pinocchio #839: Compute CoM jacobian of subtrees

matthieuvigne, 2019-07-22:
Currently pinocchio computes the CoM of subtrees, but only the jacobian of the total model CoM. To implement tasks on a specific limb of a robot, I would like to have the jacobian of the subtree CoM as well. 

This could be done by adding a ```std::vector<Matrix3x> Jcom;``` to the Data object that would be filled during the JacobianCenterOfMassBackwardStep

jcarpent, 2019-07-22:
Adding `std::vector<Matrix3x> Jcom` would not be efficient, because I guess you only need one center of mass Jacobian for a specific subtree, not for all them.

I would suggest implementing a dedicated function called `computeCoMSubtreeJacobian` which will fill a matrix given as output argument, and doing the job you want. 

@matthieuvigne Are you willing to fulfill this task?

matthieuvigne, 2019-07-22:
I felt doing a vector was closer to what pinocchio usually does in having one algorithm compute everything as it goes and storing the results for latter use ; this is exactly what is being done for subtree CoMs for instance (in ```container::aligned_vector<Vector3> com```). So I thought it made sense to do the same thing for the Jacobian. But yes for my use case a function would be enough.

And yes I can implement this, just wanted to get you opinion before getting started.

jmirabel, 2019-07-22:
Jacobian for all subtrees consumes a lot of memory. Typically, you will flood your CPU cache. This can cause inefficiencies.

matthieuvigne, 2019-07-22:
> Typically, you will flood your CPU cache.

Good point, I hadn't thought of that. I'll go with a basic function then.

jcarpent, 2019-07-22:
> Jacobian for all subtrees consumes a lot of memory. Typically, you will flood your CPU cache. This can cause inefficiencies.

I do agree. This is also why we only store the Jacobians of all the joints in `data.J` and extract then the right Jacobians when needed.

jcarpent, 2019-07-22:
> And yes I can implement this

Sounds good. Thanks a lot!

jcarpent, 2019-07-27:
See #843 for a full and efficient solution.

jcarpent, 2019-07-27:
fixed by #843.

closed 2019-07-27 by jcarpent