stack-of-tasks/pinocchio #844: appendModel Question

Ryangupta8, 2019-07-31:
Hi, I am trying to use the library for object collision avoidance. I have had some issues when creating a model and geometry model for a robot combined with an object. Appending the object model and geometry model occurs at some frame on the base model.
appendModel( parent model, child model, parent geomModel, child geomModel, frame number, SE3, output model, output geomModel)

Initially I was appending to the last frame of the base model. I realized that this meant the base frame of the child object was attached to the last frame of the robot (in this case, the finger), so when the hand moves, so did the child object. 

When attempting to use appendModel at the 0th frame in the base model (i.e the universe), I am unable to run forward kinematics, as an assertion failed, suggesting that the model and data were not consistent. Note that this was after instantiating a new version of data for the new appended model. 

In the unittest, the only example of using appendModel is for adding a manipulator onto a humanoid, and in this case it is clear that the goal would be for the manipulator to move with some parent frame on the humanoid. Do you have any suggestions for appending the model in such a way that it does not move with the base model?

jmirabel, 2019-08-05:
I use Pinocchio in the way you mention and it works fine. You may want to have a look at the code of this function: https://github.com/humanoid-path-planner/hpp-pinocchio/blob/77296185e895b9f6e42fa1408b44bbbf89a8cdf3/src/urdf/util.cc#L195

If the base frame is 0, I do not call appendModel. I call `buildModel` with the model into which I want to append.

If appendModel with a base frame index equals to 0 fails, it is a bug. It shouldn't. Thanks for the report.

jmirabel, 2019-08-05:
> When attempting to use appendModel at the 0th frame in the base model (i.e the universe), I am unable to run forward kinematics, as an assertion failed, suggesting that the model and data were not consistent.

I couldn't reproduce the assertion you mention. I added the following lines in the unit test model.cpp, at the commit above.
```c++
    Data data1 (model1);
    Eigen::VectorXd q (neutral (model1));
    forwardKinematics(model1, data1, q);
```

jcarpent, 2019-08-05:
@jmirabel A simple check, have you checked in debug mode?

jmirabel, 2019-08-05:
> @jmirabel A simple check, have you checked in debug mode?

Yes.

jcarpent, 2019-08-05:
@Ryangupta8 Can you provide a simple example of your issue? It will be easier for us to provide guidance and help.

Ryangupta8, 2019-08-06:
`pinocchio::urdf::buildModel(filename, pinocchio::JointModelFreeFlyer(),model1)
pinocchio::urdf::buildModel(filename, pinocchio::JointModelFreeFlyer(),model2);
pinocchio::urdf::buildGeom(mode1, filename, pinocchio::COLLISION, geomModel1, packageDirs );
pinocchio::urdf::buildGeom(mode2, filename, pinocchio::COLLISION, geomModel2, packageDirs );`
- set the current configuration for each, run forward kinematics and update geometry placements
- add all collision pairs, remove those specified in an srdf
`pinocchio::appendModel(model1,model2, geomModel1,geomModel2, 0, pinocchio::SE3::Identity(), model3, geomModel3);`
- update configuration of model3
- instantiate new data, geomData for model3, geomModel3
`pinocchio::forwardKinematics(mode3l, data3, q_update);`
- goal is to here update kinematics and geometry placements for computeDistance queries to get pairs of nearest_points

jmirabel, 2019-08-06:
When do you create the data for model 3 ?

Ryangupta8, 2019-08-06:
After appendModel. In the structure we created to hold these models we hold these as pointers, so those functions like forwardKinematics(model3, *data, q_update)
`data3 = std::unique_ptr<pinocchio::Data>(new pinocchio::Data(model3));
geomData3 = std::unique_ptr<pinocchio::GeometryData>(new pinocchio::GeometryData(geomModel3));`

jmirabel, 2019-08-06:
What version of Pinocchio do you use ?

Ryangupta8, 2019-08-06:
v2.1.3

jcarpent, 2019-08-06:
@Ryangupta8 Could you try with the latest release 2.1.6?

jmirabel, 2019-08-06:
I used v2.1.3 for my test. Can you make a minimal working example (with your URDF if it is publicly available) ?

Ryangupta8, 2019-08-06:
[minimal example with urdf.zip](https://github.com/stack-of-tasks/pinocchio/files/3473290/minimal.example.with.urdf.zip)
- Here is a zip with a minimal example and two urdf files: one for a cart and one for valkyrie robot

jmirabel, 2019-08-07:
I was able to reproduce the bug. It comes from the fact that you merge two models with conflicting joint or frame names. At the moment, this issue is not checked for. I don't know why the problem appears only when using frame 0...

A workaround is to add the following code after creating `cart_model`.
```cpp
  // Apply a prefix
  std::string prefix ("cart/");
  for (pinocchio::JointIndex i = 1; i < cart_model.joints.size(); ++i) {
    cart_model.names[i] = prefix + cart_model.names[i];
  }
  for (pinocchio::FrameIndex i = 0; i < cart_model.frames.size(); ++i) {
    ::pinocchio::Frame& f = cart_model.frames[i];
    f.name = prefix + f.name;
  }
  BOOST_FOREACH(::pinocchio::GeometryObject& go, geomModel.geometryObjects) {
    go.name = prefix + go.name;
  }

```

jcarpent, 2019-08-07:
Solved by #851.

closed 2019-08-07 by jcarpent