stack-of-tasks/pinocchio #849: Build error when porting from v2.1.3 to v2.1.6

fabioexavier, 2019-08-06:
When upgrading from v2.1.3 to v2.1.6, this existing block of code fails to build:

```c++
void switchJoints(pinocchio::Model *modelInOut, uint32_t const& firstJointId, uint32_t const& secondJointId)
{
....
     pinocchio::JointModel jointTemp = modelInOut->joints[firstJointId];  // Utilities.cc line 326
....
}
``` 

```
/opt/install/GCC/include/pinocchio/multibody/joint/joint-generic.hpp:129:10: required from 'struct pinocchio::JointModelTpl<double>'
Utilities.cc:326:39: required from here
/opt/install/GCC/include/boost/variant/detail/make_variant_list.hpp:40:46: error: wrong number of template arguments (21, should be at least 0)
typedef typename mpl::list< T... >::type type;
^~~~
In file included from /opt/install/GCC/include/boost/mpl/aux_/include_preprocessed.hpp:37:0,
from /opt/install/GCC/include/boost/mpl/list.hpp:46,
from /opt/install/GCC/include/boost/math/policies/policy.hpp:9,
from /opt/install/GCC/include/boost/math/special_functions/math_fwd.hpp:29,
from /opt/install/GCC/include/boost/math/special_functions/sign.hpp:17,
from /opt/install/GCC/include/boost/lexical_cast/detail/inf_nan.hpp:34,
from /opt/install/GCC/include/boost/lexical_cast/detail/converter_lexical_streams.hpp:63,
from /opt/install/GCC/include/boost/lexical_cast/detail/converter_lexical.hpp:54,
from /opt/install/GCC/include/boost/lexical_cast/try_lexical_convert.hpp:42,
from /opt/install/GCC/include/boost/lexical_cast.hpp:32,
from /opt/install/GCC/include/urdf_model/pose.h:45,
from /opt/install/GCC/include/urdf_model/joint.h:44,
from /opt/install/GCC/include/urdf_model/link.h:46,
from /opt/install/GCC/include/urdf_model/model.h:43,
from /opt/install/GCC/include/urdf_parser/urdf_parser.h:44,
from Utilities.cc:10:
/opt/install/GCC/include/boost/mpl/aux_/preprocessed/gcc/list.hpp:22:8: note: provided for 'template<class T0, class T1, class T2, class T3, class T4, class T5, class T6, class T7, class T8, class T9, class T10, class T11, class T12, class T13, class T14, class T15, class T16, class T17, class T18, class T19> struct boost::mpl::list'
struct list;
^~~~
```

Did the API change between these two versions regarding joint model manipulation? could this be due to the recent addition of new joints types like JointMimic?

jcarpent, 2019-08-06:
@fabioexavier First of all, thanks for reporting this bug. 

By adding JointMimic, we have increased the size of the boost::vector internally defined by the the boost::variant class. In other words, we had to set BOOST_MPL_LIMIT_LIST_SIZE to 30.

This quantity is defined in this file: [src/container/boost-container-limits.hpp](https://github.com/stack-of-tasks/pinocchio/blob/master/src/container/boost-container-limits.hpp)

Can you provide a self content example to fix your issue? Indeed, the lines you have provided cannot help me to understand the inclusion of the headers you're are doing.

fabioexavier, 2019-08-06:
We narrowed it down to this code, uncommenting the first include triggers the same build error, otherwise it works.

```c++
// #include <urdf_parser/urdf_parser.h>
#include <pinocchio/multibody/model.hpp>

void switchJoints(pinocchio::Model* modelInOut, unsigned int const& firstJointId, unsigned int const& secondJointId)
{
    pinocchio::JointModel jointTemp = modelInOut->joints[firstJointId];
    modelInOut->joints[firstJointId] = modelInOut->joints[secondJointId];
    modelInOut->joints[secondJointId] = jointTemp;
}

int main(int argc, char** argv)
{
    return 0;
}
```

jcarpent, 2019-08-06:
On my OS, your code is compiling well.
Can you tell me on which distribution are you working on? and the version of UDRFDOM you are using?

jcarpent, 2019-08-06:
Can you also switch the two includes and give me the outcome?

fabioexavier, 2019-08-07:
I'm using Ubuntu 18.04.2 with urdfdom 0.3.0.

Interestingly, if I switch the two includes, the code compiles and I don't get any errors.

jmirabel, 2019-08-07:
This file https://github.com/stack-of-tasks/pinocchio/blob/master/src/container/boost-container-limits.hpp must be included before boost MPL is included.

There should be a compilation error, based on the existence of the header guard of boost MPL. @jcarpent any reason not to have included it ?

jcarpent, 2019-08-07:
@jmirabel It is already included in  https://github.com/stack-of-tasks/pinocchio/blob/master/src/fwd.hpp.

jcarpent, 2019-08-07:
@fabioexavier Could you try #852 please?

jmirabel, 2019-08-07:
> @jmirabel It is already included in https://github.com/stack-of-tasks/pinocchio/blob/master/src/fwd.hpp.

I meant why something like this:
```c++
ifdef BOOST_MPL_HEADER_GUARD
# error "you must include pinocchio before boost"
#endif
```
in https://github.com/stack-of-tasks/pinocchio/blob/master/src/container/boost-container-limits.hpp

But your patch #852 works too.

jcarpent, 2019-08-07:
Solved by #852.

jcarpent, 2019-08-08:
@fabioexavier Any news?

jmirabel, 2019-09-23:
As you can see, this issue has a very broad impact. It may even lead to unsolvable setups. I think we should do something about it.

@jcarpent the container that needs to have a larger storage capacity is `boost::variant`, right ? Do you see any other requirements ?

jcarpent, 2019-09-23:
Yes, only boost::variant. What is the current issue?

jmirabel, 2019-09-23:
1. The number of places where this has had impact.
2. The possible binary incompatibility between pinocchio and any library using boost variants with the default limit size.

jmirabel, 2019-09-23:
After looking at boost variant code, it seems that the BOOST_MPL_LIMIT_LIST_SIZE is used only when boost cannot use variadic templates. This happens either by a user request or when the compiler or C++ standard is too old. My conclusion is that it isn't necessary to set BOOST_MPL_LIMIT_LIST_SIZE to 30 for c++11 and above (though I didn't try myself). I think c++0x is sufficient.

See https://github.com/boostorg/variant/blob/c401edba70c9f3668dae55ee1cfa5d9b2c6fc2fc/include/boost/variant/variant_fwd.hpp#L66 and #if defined(__GXX_EXPERIMENTAL_CXX0X__) || (__cplusplus >= 201103L) for some pointers inside boost code.

jmirabel, 2019-09-24:
I reopened the issue because I stumbled upon this again just now. I can open another issue if desired.

A first step towards solving this is not to set BOOST_MPL_LIMIT_LIST_SIZE for C++11. @jcarpent if it works, would you agree with this ?

jcarpent, 2019-09-24:
I can take it in charge. But, this also means that we need to remove the flags from the *.pc file. Is it OK for you?

jmirabel, 2019-09-24:
Thank you.
For c++11, yes, of course.

jcarpent, 2019-09-24:
Just to be sure, for both C++98 and C++11, I will remove the forcing flags in the `*.pc` file.

jmirabel, 2019-09-24:
For c++98, you think the `pragma error` is sufficient ?

jmirabel, 2019-09-24:
Then, to make Pinocchio backward compatible, I suggest that for C++ < 11, we stick to a limit of 20 for the variant.

jcarpent, 2019-09-24:
I think this is not the correct way of doing this compatibility. We should keep the current Variant, and to set the `BOOST_MPL_LIMIT_LIST_SIZE` only for versions of C++ before C++11.

jmirabel, 2019-09-24:
*This change was not backward compatible*. Most of the source files of SoT and HPP must be changed. Being forced to include `pinocchio/fwd.hpp` before `boost/...` is bad idea.

Moreover, at the moment, the need for more than 20 elements is for mimic joints (am I wrong here ?) and, as of now, they are not ready for deployment.

I am open to other solutions if you have one to suggest.

jcarpent, 2019-09-24:
Yes, but we cannot backward as the releases have been done. So, normally, in your context, the *.pc lines should make the job.

jcarpent, 2019-09-24:
And then it means that we have to make silent the `pragma error` in the context where `BOOST_MPL_LIMIT_LIST_SIZE` is defined.

jmirabel, 2019-09-24:
The written discussion is a bit tedious. If possible, I would prefer a phone call. I am available today before 3pm or after 4:30pm.

jcarpent, 2019-09-24:
Solved by #889.

wxmerkt, 2019-10-23:
I also just ran into this: This happened despite being in a **C++11 project** - I didn't get a good error/warning pointing me to include `pinocchio/fwd.hpp` prior to any other header. Via Google I came across the original issue by Carlos, and then found the fix here. Is there a way to fix it without having to first include `fwd.hpp` for C++11 and onwards?

jmirabel, 2019-10-24:
Does catkin uses the same compilation flags as the one in the pc file ?

jcarpent, 2019-10-24:
@wxmerkt Have you tried the last release (2.1.9). They should fix your issue even on C++11

jmirabel, 2019-10-24:
Is it possible that any of the other dependencies also set BOOST_MPL_LIMIT_LIST_SIZE ?

jcarpent, 2019-10-24:
You should use `${PINOCCHIO_CFLAGS}` I think.

wxmerkt, 2019-10-24:
> Is it possible that any of the other dependencies also set BOOST_MPL_LIMIT_LIST_SIZE ?

Previously, I included a ROS header _before_ Pinocchio - that broke it. ROS on Melodic still has Boost in it, so potentially? [I don't know].

wxmerkt, 2019-10-24:
**CFLAGS_OTHER** is: 
```
-DPINOCCHIO_URDFDOM_TYPEDEF_SHARED_PTR -DPINOCCHIO_URDFDOM_USE_STD_SHARED_PTR -DBOOST_MPL_LIMIT_LIST_SIZE=30 -DPINOCCHIO_WITH_URDFDOM
```

**CFLAGS** is:
```
-DPINOCCHIO_URDFDOM_TYPEDEF_SHARED_PTR -DPINOCCHIO_URDFDOM_USE_STD_SHARED_PTR -DBOOST_MPL_LIMIT_LIST_SIZE=30 -DPINOCCHIO_WITH_URDFDOM -I/home/wxm/dev/pinocchio_ws/install/include -I/usr/include/eigen3
```
I have already taken care of the `include_directories` so `PINOCCHIO_CFLAGS_OTHERS` should suffice.

jmirabel, 2019-10-24:
Could you provide:
- the compilation command,
- the standard output of the same command with the additional gcc option `-E` ?