stack-of-tasks/pinocchio #858: Why dIntegrate creates a NaN block in the Jacobian for floating-base systems?

cmastalli, 2019-08-12:
Hi guys,

Would someone can explain me (or share with me a link) this question?

One of my goals is to compute the Jacobian of the difference operator too. As we know, we can compute this from the Jacobian of integrate operator, but this creates a the block term with NaN values. I suspect this is wrong!

It would be also important to implement dDifference, and therefore avoid any inversion of Jintegrate. @jmirabel could you help / guide me with this task?

Thanks,
Carlos

jcarpent, 2019-08-12:
If your configuration vector is not normalized, this is expected.

cmastalli, 2019-08-12:
What exactly means normalize?

jcarpent, 2019-08-12:
Quaternions part of the config vector should be normalized.

cmastalli, 2019-08-12:
This happens even when the quaternion is normalized

jmirabel, 2019-08-12:
Weird. Could you add `feenableexcept(FE_DIVBYZERO | FE_INEXACT | FE_INVALID)` before the call to dIntegrate and provide gdb backtrace of the thrown exception ?

Do you have any piece of code ? A reproducible code ?

cmastalli, 2019-08-13:
I inspected a little more this problem. It happens when I used this model:
```python
model = pinocchio.buldSampleModelHumanoid()
# model = pinocchio.buildSampleModelHumanoidRandom() this works well
```

jmirabel, 2019-08-13:
In your example, `q` has infinite values.

cmastalli, 2019-08-13:
> In your example, `q` has infinite values.

Yes, and this is another problem in `buildSampleModelHumanoid `. Do you think is the reason?

jmirabel, 2019-08-13:
The 3 first DoFs have no bounds. You must first set the upper and lower position limits.

With finite values for q and v, I don't get any errors.

closed 2019-08-13 by jcarpent