stack-of-tasks/pinocchio #859: There is Eigen memory allocation in computeABADerivatives

cmastalli, 2019-08-13:
Hi guys,

I reported that there is memory allocation by Eigen expressions in compute `computeABADerivatives`. This is piece of the code:

```cpp
 pinocchio::computeABADerivatives(model, data, q, v, u);
```

I didn't try it yet the other method with the derivatives as in/out argument.

Carlos

cmastalli, 2019-08-13:
I suspect this line creates memory allocation: https://github.com/stack-of-tasks/pinocchio/blob/master/src/algorithm/aba-derivatives.hxx#L429

jcarpent, 2019-08-13:
@cmastalli Thanks for raising it. I would say it really depends on the version of Eigen this kind of stuff. And I won't be surprised that there is some memory allocation in Pinocchio.

jcarpent, 2019-08-13:
> I suspect this line creates memory allocation: https://github.com/stack-of-tasks/pinocchio/blob/master/src/algorithm/aba-derivatives.hxx#L429

For sure, the line you mention is only a static_cast. And in this way, there is no memory allocation at all here. The memory allocation is certainly deeper in the code I would say.

cmastalli, 2019-08-13:
Do you think we should test the memory allocation in Pinocchio unit-test? I asked myself the same question in Crocoddyl.

jcarpent, 2019-08-13:
It can be useful, but again, it depends on Eigen itself which might do some memory allocation or not depending on its Version. It is hard to do it well.

gabrielebndn, 2019-09-13:
I cannot detect any allocation. On Ubuntu 16.04, (Eigen 3.2.92), this code

```
#define EIGEN_RUNTIME_NO_MALLOC

#include "pinocchio/multibody/model.hpp"
#include "pinocchio/multibody/data.hpp"
#include "pinocchio/algorithm/joint-configuration.hpp"
#include "pinocchio/algorithm/aba-derivatives.hpp"
#include "pinocchio/parsers/sample-models.hpp"

using namespace Eigen;
using namespace pinocchio;

using std::cout;
using std::endl;

int main() {
  Model model;
  buildModels::humanoidRandom(model);
  
  Data data(model);
  
  model.lowerPositionLimit.head<3>().fill(-1.);
  model.upperPositionLimit.head<3>().fill(1.);
  
  VectorXd q = randomConfiguration(model);
  VectorXd v(VectorXd::Random(model.nv));
  VectorXd tau(VectorXd::Random(model.nv));
  
  MatrixXd aba_partial_dq(model.nv,model.nv); aba_partial_dq.setZero();
  MatrixXd aba_partial_dv(model.nv,model.nv); aba_partial_dv.setZero();
  MatrixXd aba_partial_dtau(model.nv,model.nv); aba_partial_dtau.setZero();
  
  Eigen::internal::set_is_malloc_allowed(false);
  computeABADerivatives(model,data,q,v,tau,aba_partial_dq,aba_partial_dv,aba_partial_dtau);
  Eigen::internal::set_is_malloc_allowed(true);

  cout << aba_partial_dq.row(0) << endl;
  cout << endl;

  Eigen::internal::set_is_malloc_allowed(false);
  computeABADerivatives(model,data,q,v,tau);
  Eigen::internal::set_is_malloc_allowed(true);

  cout << data.ddq_dq.row(0) << endl;
}
```
compiled with this command (using Pinocchio on the current devel branch)
```
g++ -I /usr/include/eigen3/ -I /local/users/gbuondon/install/include/ test_aba_derivatives.cpp -o test_aba_derivatives
```
works perfectly fine.

jcarpent, 2019-09-13:
@gabrielebndn Thanks for the sanity check.
It may indeed depend on the eigen vesion.
Could you try with -O3 option too?

cmastalli, 2019-09-13:
I used the binary version of Ubuntu 16.04 (i.e. v3.3.2)

cmastalli, 2019-09-13:
I will do it next week.

jmirabel, 2019-09-13:
@cmastalli Are you sure the arguments were of the correct size ?

cmastalli, 2019-09-13:
I am sure that I used proper matrix dimensions.

jcarpent, 2019-09-13:
The simpler is that @cmastalli shares his code. Do you have it ready?

cmastalli, 2019-09-17:
I tested the @gabrielebndn script and it doesn't produce any memory allocation. I think we can close this issue

jcarpent, 2019-09-17:
Thanks for this check. And thanks @gabrielebndn for the nice script.

closed 2019-09-17 by jcarpent

cmastalli, 2019-09-18:
@jcarpent sorry if I come back to this issue.

When I detected this memory allocation I was using Python bindings and loading the Pinocchio Model using robot wrapper.
Do you think there is another explanation to this issue? For instance, it would be something regarding the wrapping Pinocchio model with boost python.

I just noticed the same problem but with `computeRNEADerivatives`. I have not idea what it's going on.