stack-of-tasks/pinocchio #862: namespace error in spatial/explog-quaternion.hpp

pFernbach, 2019-08-21:
https://github.com/stack-of-tasks/pinocchio/blob/11e195bb5ae57bb0d3010642608cf836620bf14e/src/spatial/explog-quaternion.hpp#L191

This line cause compilation error when I include this file :  
```
/opt/openrobots/include/pinocchio/spatial/explog-quaternion.hpp: In function ‘void pinocchio::quaternion::Jlog3(const Eigen::QuaternionBase<Derived>&, const Eigen::MatrixBase<U>&)’:
/opt/openrobots/include/pinocchio/spatial/explog-quaternion.hpp:183:7: error: ‘Jlog3’ is not a member of ‘pinocchio’
       pinocchio::Jlog3(t,w,Jlog);
       ^
/opt/openrobots/include/pinocchio/spatial/explog-quaternion.hpp:183:7: note: suggested alternative:
/opt/openrobots/include/pinocchio/spatial/explog-quaternion.hpp:175:10: note:   ‘pinocchio::quaternion::Jlog3’
     void Jlog3(const Eigen::QuaternionBase<QuaternionLike> & quat,
```

It is solved by either changing it to `pinocchio::quaternion::Jlog3` or just `Jlog3`.

jcarpent, 2019-08-22:
I think the issue is coming from your inclusions.
Can you provide a minimal example showing your issue?

jcarpent, 2019-08-22:
You should include `#include "pinocchio/spatial/explog.hpp" to get access to the signatures of exp and log functions with Quaternion inputs.

gabrielebndn, 2019-08-22:
> You should include `#include "pinocchio/spatial/explog.hpp" to get access to the signatures of exp and log functions with Quaternion inputs.

I see `pinocchio/spatial/explog.hpp` includes `pinocchio/spatial/explog-quaternion.hpp` *at the end*. So the user should never include `pinocchio/spatial/explog-quaternion.hpp` directly, they should only care about `pinocchio/spatial/explog.hpp`.

I think this is not intuitive and the error made by @pFernbach is natural. If `explog-quaternion.hpp` is not meant to be employed by the user, I suggest to take one of the following actions:
- Removing `explog-quaternion.hpp` and putting all its code into `explog.hpp`
- Renaming `explog-quaternion.hpp` to `explog-quaternion.hxx` and possibly putting the forward declarations in `explog.hpp`
- Performing a more thorough reorganization, by doing the same as above but also moving the code in `explog.hpp` to a file called `explog.hxx`

I favor this last hypothesis. @jcarpent do you agree? This is minor, but it is quick to do and it will improve the organization, preventing errors like this one.

jcarpent, 2019-08-22:
@gabrielebndn I do agree on the third option which allows a perfect clean of this part of the coder.
On my side, I have already performed a deeper splitting with log.{hpp,hxx} and exp.{hpp,hxx}.
This is on a branch which currently under development.

Before submitting this PR, I have a quick fix in mind which consists of adding these lines in the `explog-quaternion.hpp` file:
```
#ifndef __spatial_explog_hpp__
  #include "pinocchio/spatial/explog.hpp"
#endif
```
@gabrielebndn Do you agree?

jcarpent, 2019-08-22:
@gabrielebndn If you are interested to implement the complete splitting, you can go ahead if you want.

gabrielebndn, 2019-08-22:
> Before submitting this PR, I have a quick fix in mind which consists of adding these lines in the `explog-quaternion.hpp` file:
> 
> ```
> #ifndef __spatial_explog_hpp__
>   #include "pinocchio/spatial/explog.hpp"
> #endif
> ```
> 
> @gabrielebndn Do you agree?

I think the `#ifndef` is not strictly needed, the include guard inside `explog.hpp` should be enough.
Also, I don't think the fix itself is needed. The problem at hand is solved by including the correct header, that is `explog.hpp` instead of `explog-quaternion.hpp`

I am not sure the third proposal is clear. I do not want a complete split. What I propose is:

`explog.hpp`:
```
<forward explog declarations>

<forward explog quaternion declarations>

#include pinocchio/spatial/explog.hxx

#include pinocchio/spatial/explog-quaternion.hxx
```

So that the user could still do `#include <pinocchio/spatial/explog.hpp>` to have everything.
Also, I do not see the reason of splitting log and exp in two different headers.

jcarpent, 2019-08-22:
> Also, I do not see the reason of splitting log and exp in two different headers.

To simplify maintenance and to not have hundred of line of codes in the same file.

jcarpent, 2019-08-23:
Solved by #863

closed 2019-08-23 by jcarpent