stack-of-tasks/pinocchio #872: Update Installation Instructions in the documentation.

paLeziart, 2019-09-13:
There is a small update to do in the installation instructions of the documentation:
https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/md_doc_d-practical-exercises_1-directgeom.html

Trying to install the package robotpkg-gepetto-viewer-corba returns:
`robotpkg-gepetto-viewer-corba : Depends on: robotpkg-gepetto-viewer (= 4.2.0) but 4.4.0 will be installed`

You have to install robotpkg-py27-qt4-gepetto-viewer-corba which I guess is a newer and updated version of gepetto-viewer-corba

Best,

jcarpent, 2019-09-13:
Thanks @paLeziart for raising this issue.
Apparently, @nim65s already solved it few days ago in #870, which has been merged in the devel branch. 
For simplicity, I will update the Readme.md file on the master branch.

closed 2019-09-13 by jcarpent

paLeziart, 2019-09-13:
In the following there is also an issue with the creation of the RobotWrapper.

Using:
`robot = RobotWrapper(URDF, [PKG])`
Returns:
`'str' object has no attribute 'createData'`

Looking into the source code of RobotWrapper I think it is now:
`robot = RobotWrapper.BuildFromURDF(URDF, [PKG])`

jcarpent, 2019-09-13:
I would suggest opening a PR with the fix. It will be easier. I will review it.

jcarpent, 2019-09-13:
@paLeziart Can you do that?

paLeziart, 2019-09-13:
@jcarpent I'm not sure `robot = RobotWrapper.BuildFromURDF(URDF, [PKG])` actually works because the urdf of ur5-description package is broken so I get an error message with this command.

ur5-description urdf ask for package://ur-description for all meshes instead of package://ur5-description

Once it will be fixed (I reported the issue) I will be able to finish the tutorial.

nim65s, 2019-09-13:
the ur5-description package has already been fixed :)
But you might need to setup some environment variables, like `ROS_PACKAGE_PATH` or `CMAKE_PREFIX_PATH` for the `package://` thing to work.

paLeziart, 2019-09-13:
@nim65s You mean it is already fixed or that the fix will be applied to the main branch in the near future?
Because I downloaded this robotpkg-ur5-description 2 hours ago so I think I got the latest version.

nim65s, 2019-09-13:
Ok, I fixed the package yesterday, so this is another issue…