stack-of-tasks/pinocchio #879: Memory allocation in computeRNEADerivatives

cmastalli, 2019-09-18:
There is memory allocation when we pass `fext`. To be efficient, I created the following code where you can check this issue:

```cpp
#define EIGEN_RUNTIME_NO_MALLOC

#include "pinocchio/multibody/model.hpp"
#include "pinocchio/multibody/data.hpp"
#include "pinocchio/algorithm/joint-configuration.hpp"
#include "pinocchio/algorithm/rnea-derivatives.hpp"
#include "pinocchio/parsers/sample-models.hpp"

using namespace Eigen;
using namespace pinocchio;

using std::cout;
using std::endl;

int main() {
  Model model;
  buildModels::humanoidRandom(model);
  
  Data data(model);
  
  model.lowerPositionLimit.head<3>().fill(-1.);
  model.upperPositionLimit.head<3>().fill(1.);
  
  VectorXd q = randomConfiguration(model);
  VectorXd v(VectorXd::Random(model.nv));
  VectorXd a(VectorXd::Random(model.nv));

  typedef container::aligned_vector<Force> ForceVector;
  ForceVector fext((size_t)model.njoints);
  for(ForceVector::iterator it = fext.begin(); it != fext.end(); ++it)
    (*it).setRandom();

  Eigen::internal::set_is_malloc_allowed(false);
  computeRNEADerivatives(model,data,q,v,a,fext); // this allocates memory
//  computeRNEADerivatives(model,data,q,v,a); // this doesn't allocate memory (yeah!)
  Eigen::internal::set_is_malloc_allowed(true);
}
```

Remember that you need to compile in debug mode. If you name this file as `test_rnea_derivatives.cpp`, then this is a suggested compilation command:
```bash
g++ -I /usr/include/eigen3/ -I /local/users/cmastall/include/ -g test_rnea_derivatives.cpp -o test_rnea_derivatives
```

jcarpent, 2019-09-18:
Thanks for the report. I will check it now.

By the way, please use this line instead which is much more generic:
```
 g++ -I $(pkg-config --cflags pinocchio) -g test_rnea_derivatives.cpp -o test_rnea_derivatives
```

jcarpent, 2019-09-18:
I found the bug. I will provide a fix in few minutes. Thanks @cmastalli for your concerns.

cmastalli, 2019-09-18:
Awesome!

jcarpent, 2019-09-18:
Solved by #880.

closed 2019-09-18 by jcarpent