stack-of-tasks/pinocchio #883: Python warns against using np.Matrix class

blooop, 2019-09-19:
When I use ```pin.computeGeneralizedGravity()```
A numpy matrix is returned which then fires this warning:

``` PendingDeprecationWarning: the matrix subclass is not the recommended way to represent matrices or deal with linear algebra (see https://docs.scipy.org/doc/numpy/user/numpy-for-matlab-users.html). Please adjust your code to use regular ndarray.```

Are there plans to depreciate the use of Matrix?

Thanks

nim65s, 2019-09-19:
Hi @blooop,

This issue concerns [eigenpy](https://github.com/stack-of-tasks/eigenpy), and its behavior can be changed with `eigenpy.switchToNumpyArray()`

nim65s, 2019-09-19:
ref https://github.com/stack-of-tasks/eigenpy/issues/41

jmirabel, 2019-09-19:
I can reformulate the question like this.
Are there any plans to change eigenpy such that it uses arrays instead of matrices by default ?

At some point, this will be necessary. I have no idea of the timeline though.

jcarpent, 2019-09-19:
> Are there plans to depreciate the use of Matrix?

Yes, in future release of Pinocchio we plan to do it. We managed to handle both cases thanks to EigenPy.

jcarpent, 2019-09-19:
> Are there any plans to change eigenpy such that it uses arrays instead of matrices by default ?

Yes, for sure. We need to decide when? The smooth solution is already implemented, which means that we have the option to change the conversion convention online. We just need to setup the default value (this can already be performed in each package depending on EigenPy).

jmirabel, 2019-09-19:
I opened an issue on Eigenpy to discuss about that: https://github.com/stack-of-tasks/eigenpy/issues/87

jcarpent, 2019-09-19:
Dear @blooop,

First of all, thanks for raising this issue. I've just released a new release of EigenPy which will lead to a new release of Pinocchio in the very few next days.
You can already suppress your current warning by using `numpy.array`in all your code thanks to `eigenpy.switchToNumpyArray()`.

I will close this issue as it seems to be solved. Feel free to reopen it if needed.

closed 2019-09-19 by jcarpent

blooop, 2019-09-19:
Excellent, thanks!