stack-of-tasks/pinocchio #884: Installation fails on Bionic due to eigenpy, robotpkg-pinocchio, robotpkg-hpp-fcl version mismatches

blooop, 2019-09-20:
The installation outlined here is not working at the moment:
https://stack-of-tasks.github.io/pinocchio/download.html

```
sudo apt install -y robotpkg-py27-pinocchio
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies.
 robotpkg-py27-pinocchio : Depends: robotpkg-py27-eigenpy (= 1.4.3) but 1.6.0 is to be installed or
                                    robotpkg-py27-eigenpy (= 1.4.4) but 1.6.0 is to be installed or
                                    robotpkg-py27-eigenpy (= 1.4.5) but 1.6.0 is to be installed or
                                    robotpkg-py27-eigenpy (= 1.5.0) but 1.6.0 is to be installed or
                                    robotpkg-py27-eigenpy (= 1.5.1) but 1.6.0 is to be installed
                           Depends: robotpkg-pinocchio (= 2.1.3) but it is not going to be installed or
                                    robotpkg-pinocchio (= 2.1.2) but it is not going to be installed
                           Depends: robotpkg-hpp-fcl (= 1.1.1) but 1.1.3 is to be installed
E: Unable to correct problems, you have held broken packages.
```
I was able to fix this manually, but I thought I should let someone know.  Fix:
```
  sudo apt install robotpkg-py27-eigenpy=1.5.1
  sudo apt install robotpkg-hpp-fcl=1.1.1
  sudo apt install robotpkg-pinocchio=2.1.3
  sudo apt install -y robotpkg-py27-pinocchio
```

nim65s, 2019-09-20:
Hi @blooop,

Thanks for raising the issue.
This is because the buildfarm is currently rebuilding everything, to update Pinocchio & Eigenpy to their last versions.

You can follow the progression here: http://robotpkg.openrobots.org/rbulk/rbulk-dispatchd.html

I'll close the issue, please re-open it if necessary.

closed 2019-09-20 by nim65s

wxmerkt, 2019-09-20:
For understanding @nim65s: Does this mean there is no staging until all packages are rebuilt, i.e., if I pull or install at the wrong time it fails?

nim65s, 2019-09-20:
@wxmerkt that's right.

The eigenpy package is ready, but the pinocchio package is not yet available.

So the old pinocchio package is not compatible with the new eigenpy, and apt tries to upgrade eigenpy anyways, so this fails. But the workaround is straightforward, and @blooop found it: just say to apt that you want the old version.

wxmerkt, 2019-09-20:
Okay that is a bit worrying. Would it be possible to have newly built packages be in a staging and only move them to the main repository everybody is pulling from on completion of the full rebuild/release? [Given that it is a lot of packages in the queue as per your dependency diagram]

nim65s, 2019-09-20:
I agree, we should work on it.

I opened an issue there: https://git.openrobots.org/issues/214