stack-of-tasks/pinocchio #893: Pinocchio installation problem on Mac

ddliugit, 2019-09-26:
brew reinstall gepetto-viewer-corba

gepetto-gui

I found the following error:

yld: Symbol not found: __ZN5boost11basic_regexIcNS_12regex_traitsIcNS_16cpp_regex_traitsIcEEEEE9do_assignEPKcS7_j
  Referenced from: /usr/local/bin/gepetto-gui
  Expected in: /usr/local/opt/boost/lib/libboost_regex-mt.dylib
 in /usr/local/bin/gepetto-gui

Can you help me out?

jcarpent, 2019-09-27:
Dear @ddliugit 
It makes a while I did not update the packages according to new Boost versions. I would suggest reinstalling them from sources, using the `--build-from-source` option of brew.
I will generate the packages on Monday. Is it fine for you?

ddliugit, 2019-09-27:
It works fine now. Thanks!

ddliugit, 2019-11-19:
I moved to python 3 and it doesn't work well while I am using _Pinocchio_ with _gepetto-gui:_

The errors is like so:

```
  File "integratedTest.py", line 17, in <module>
    robot = model.loadHumanoidModel(isDisplay=True,urdfPath = urdfPath,meshPath = meshPath)
  File "/usr/local/lib/python3.7/site-packages/py_lqr/model.py", line 140, in __init__
    self.initDisplay()
  File "/usr/local/lib/python3.7/site-packages/py_lqr/model.py", line 150, in initDisplay
    self.robot.initViewer(loadModel=True)
  File "/usr/local/lib/python3.7/site-packages/pinocchio/robot_wrapper.py", line 236, in initViewer
    self.viz.initViewer(*args, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/pinocchio/visualize/gepetto_visualizer.py", line 19, in initViewer
    import gepetto.corbaserver
ModuleNotFoundError: No module named 'gepetto'
```

Can you help me?

jcarpent, 2019-11-19:
Yes, this is because the module is currently compiled for Python2. I don't have the time now to produce the image. It is generated now.

jcarpent, 2019-11-19:
`brew install gepetto-viewer-corba-python3`

closed 2019-11-19 by jcarpent

ddliugit, 2019-11-20:
I try it, but same error still appears.

Also, I tried:

```
brew unlink gepetto-viewer-corba
brew link gepetto-viewer-corba-python3
```
Any idea why does this happen?

ddliugit, 2019-11-26:
@jcarpent  Can you help me out?

jcarpent, 2019-11-27:
I cannot guess what happens. Please provide consistent log.

ddliugit, 2019-11-27:
10-18-226-182:~ dongdong$ python i
/usr/local/Cellar/python/3.7.5/Frameworks/Python.framework/Versions/3.7/Resources/Python.app/Contents/MacOS/Python: can't open file 'i': [Errno 2] No such file or directory
10-18-226-182:~ dongdong$ cd repos/humanoid_control/IntegratedTest/
10-18-226-182:IntegratedTest dongdong$ python integratedTest.py 
eigenpy warning: you use the deprecated class numpy.matrix without explicily asking for it. The default behaviour will change to numpy.array at next major release.
- Either call eigenpy.switchToNumpyMatrix() before using eigenpy to suppress this warning
- or call eigenpy.switchToNumpyArray() and adapt your code accordingly.
See https://github.com/stack-of-tasks/eigenpy/issues/87 for further details.
Traceback (most recent call last):
  File "integratedTest.py", line 17, in <module>
    robot = model.loadHumanoidModel(isDisplay=True,urdfPath = urdfPath,meshPath = meshPath)
  File "/usr/local/lib/python3.7/site-packages/py_lqr/model.py", line 140, in __init__
    self.initDisplay()
  File "/usr/local/lib/python3.7/site-packages/py_lqr/model.py", line 150, in initDisplay
    self.robot.initViewer(loadModel=True)
  File "/usr/local/lib/python3.7/site-packages/pinocchio/robot_wrapper.py", line 236, in initViewer
    self.viz.initViewer(*args, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/pinocchio/visualize/gepetto_visualizer.py", line 19, in initViewer
    import gepetto.corbaserver
ModuleNotFoundError: No module named 'gepetto'

jcarpent, 2019-11-27:
@ddliugit It is fixed now!

ddliugit, 2019-11-27:
I still have errors on my machine. Is this caused by the environment setting?

The error changes:
```

10-18-226-182:humanoid_control dongdong$ python IntegratedTest/integratedTest.py 
eigenpy warning: you use the deprecated class numpy.matrix without explicily asking for it. The default behaviour will change to numpy.array at next major release.
- Either call eigenpy.switchToNumpyMatrix() before using eigenpy to suppress this warning
- or call eigenpy.switchToNumpyArray() and adapt your code accordingly.
See https://github.com/stack-of-tasks/eigenpy/issues/87 for further details.
Traceback (most recent call last):
  File "IntegratedTest/integratedTest.py", line 17, in <module>
    robot = model.loadHumanoidModel(isDisplay=True,urdfPath = urdfPath,meshPath = meshPath)
  File "/usr/local/lib/python3.7/site-packages/py_lqr/model.py", line 140, in __init__
    self.initDisplay()
  File "/usr/local/lib/python3.7/site-packages/py_lqr/model.py", line 150, in initDisplay
    self.robot.initViewer(loadModel=True)
  File "/usr/local/lib/python3.7/site-packages/pinocchio/robot_wrapper.py", line 236, in initViewer
    self.viz.initViewer(*args, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/pinocchio/visualize/gepetto_visualizer.py", line 19, in initViewer
    import gepetto.corbaserver
  File "/usr/local/lib/python3.7/site-packages/gepetto/corbaserver/__init__.py", line 1, in <module>
    import omniORB
ModuleNotFoundError: No module named 'omniORB'
```

jcarpent, 2019-11-28:
You can proceed as follows:
```
brew update
brew install omniorbpy
```

jcarpent, 2019-11-28:
By default, omniORB is not built for Python3. I've fixed it by creating a new package for brew.

ddliugit, 2019-11-28:
Thanks!

I just did it and the error changes:

`10-16-22-176:IntegratedTest dongdong$ python integratedTest.py 
eigenpy warning: you use the deprecated class numpy.matrix without explicily asking for it. The default behaviour will change to numpy.array at next major release.
- Either call eigenpy.switchToNumpyMatrix() before using eigenpy to suppress this warning
- or call eigenpy.switchToNumpyArray() and adapt your code accordingly.
See https://github.com/stack-of-tasks/eigenpy/issues/87 for further details.
Traceback (most recent call last):
  File "integratedTest.py", line 17, in <module>
    robot = model.loadHumanoidModel(isDisplay=True,urdfPath = urdfPath,meshPath = meshPath)
  File "/usr/local/lib/python3.7/site-packages/py_lqr/model.py", line 140, in __init__
    self.initDisplay()
  File "/usr/local/lib/python3.7/site-packages/py_lqr/model.py", line 150, in initDisplay
    self.robot.initViewer(loadModel=True)
  File "/usr/local/lib/python3.7/site-packages/pinocchio/robot_wrapper.py", line 236, in initViewer
    self.viz.initViewer(*args, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/pinocchio/visualize/gepetto_visualizer.py", line 19, in initViewer
    import gepetto.corbaserver
  File "/usr/local/lib/python3.7/site-packages/gepetto/corbaserver/__init__.py", line 2, in <module>
    omniORB.updateModule("gepetto.corbaserver")
AttributeError: module 'omniORB' has no attribute 'updateModule'`

ddliugit, 2019-11-28:
I tried to unlink and then link again, the error is like so. Did I miss something?
```
mac:~ dongdong$ brew unlink omniorbpy
Unlinking /usr/local/Cellar/omniorbpy/4.2.3... 178 symlinks removed
mac:~ dongdong$ brew link omniorbpy
Linking /usr/local/Cellar/omniorbpy/4.2.3... 178 symlinks created
```
```
10-16-22-176:IntegratedTest dongdong$ python integratedTest.py 
eigenpy warning: you use the deprecated class numpy.matrix without explicily asking for it. The default behaviour will change to numpy.array at next major release.
- Either call eigenpy.switchToNumpyMatrix() before using eigenpy to suppress this warning
- or call eigenpy.switchToNumpyArray() and adapt your code accordingly.
See https://github.com/stack-of-tasks/eigenpy/issues/87 for further details.
Traceback (most recent call last):
  File "integratedTest.py", line 17, in <module>
    robot = model.loadHumanoidModel(isDisplay=True,urdfPath = urdfPath,meshPath = meshPath)
  File "/usr/local/lib/python3.7/site-packages/py_lqr/model.py", line 140, in __init__
    self.initDisplay()
  File "/usr/local/lib/python3.7/site-packages/py_lqr/model.py", line 150, in initDisplay
    self.robot.initViewer(loadModel=True)
  File "/usr/local/lib/python3.7/site-packages/pinocchio/robot_wrapper.py", line 236, in initViewer
    self.viz.initViewer(*args, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/pinocchio/visualize/gepetto_visualizer.py", line 19, in initViewer
    import gepetto.corbaserver
  File "/usr/local/lib/python3.7/site-packages/gepetto/corbaserver/__init__.py", line 6, in <module>
    from gepetto.corbaserver.client import Client
  File "/usr/local/lib/python3.7/site-packages/gepetto/corbaserver/client.py", line 88
    print e
          ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print(e)?
```

ddliugit, 2019-11-28:
I think the error is caused by the syntax of python 3. I changed it manually, it works!

gabrielebndn, 2019-11-28:
@ddliugit you might want to update gepetto-viewer corba. That problem seems to be fixed in the current release (and in the current binary apt release as well). I think pretty much all of our code is compatible with Python 3 by now (@nim65s can confirm).
In our team we make frequent fixes, addition, and changes, and frequent releases, particularly for Pinocchio, so please always make sure that what you have is up to date.

ddliugit, 2019-11-28:
Thanks. BTW, how to update the `gepetto-viewer-corba`

nim65s, 2019-11-28:
I can confirm that I can run pinocchio, example-robot-data & gepetto-gui all in python3 to load some robot models and display them in their initial pose, so yes, this works.

I am also taking care of any python 2 only syntax I can find, but I might miss a few lines, especially if they are added now without CI checks about that or a review from someone who cares about that… So please tell us if you find anything.

ddliugit, 2019-11-28:
I try to install gepetto-gui in python 3. The error appears in this code `/usr/local/lib/python3.7/site-packages/gepetto/corbaserver/client.py`, which is caused by `print e` in python2 syntax. I changed it to `print(e)`.That's it.

> I tried to unlink and then link again, the error is like so. Did I miss something?
>
> ```
> mac:~ dongdong$ brew unlink omniorbpy
> Unlinking /usr/local/Cellar/omniorbpy/4.2.3... 178 symlinks removed [...]

nim65s, 2019-11-28:
This was fixed 8 months ago, and included in releases v5.3.2, v5.3.1, v5.3.0 & v5.2.1: https://github.com/Gepetto/gepetto-viewer-corba/commit/21cf4c1d97160e19906a068b9aeb3327a47079be

gabrielebndn, 2019-11-28:
@ddliugit that's what I was saying. You should update gepetto-viewer-corba, because that problem is currently fixed, both in the source release and in the apt release

gabrielebndn, 2019-11-28:
> Thanks. BTW, how to update the `gepetto-viewer-corba`

How did you install `gepetto-viewer-corba`?

If bu source, then just go to the source folder, `git pull` the latest changes, and proceed to build and install.
If by apt, use `sudo apt-get install --reinstall [...]`

ddliugit, 2019-11-28:
Thanks all. I see.

jcarpent, 2019-11-28:
@gabrielebndn @nim65s. I think you did not notice, but @ddliugit is MAC OS X systems. I'm currently making the update of the packages on homebrew. Sorry for the delay, but it is hard to be on all the fronts.

gabrielebndn, 2019-11-28:
Oh! No, hadn't noticed. Sorry @ddliugit, it seems you will have to wait for the updated brew repo, or install by source

jcarpent, 2019-11-28:
@ddliugit I've updated all the packages. It should work now.

ddliugit, 2019-11-28:
Thank you. @jcarpent