stack-of-tasks/pinocchio #897: Compilation error with CppADCodeGen

cmastalli, 2019-10-01:
My setup is as follows:
  - Ubuntu 16.04
  - CppAD (v20190200)
  - CppADCodeGen (master branch > v2.3.0)
which cmake flags that enable AutoDiff, CodeGen, unittest and benchmarks.

And this is the segment of the error message:
```bash
/local/users/cmastall/include/cppad/cg/code_handler_impl.hpp:1440:122: error: ‘class CppAD::cg::CGAbstractAtomicFun<double>’ has no member named ‘afun_name’
In file included from /local/users/cmastall/include/cppad/cg/cppadcg.hpp:214:0,
                 from /local/users/cmastall/include/cppad/cg.hpp:18,
                 from /local/users/cmastall/include/cppad/cg/support/cppadcg_eigen.hpp:18,
                 from /devel/pinocchio/build/include/pinocchio/codegen/cppadcg.hpp:11,
                 from /devel/pinocchio/unittest/cppadcg-basic.cpp:5:
/local/users/cmastall/include/cppad/cg/model/functor_generic_model.hpp: In instantiation of ‘bool CppAD::cg::FunctorGenericModel<Base>::addAtomicFunction(CppAD::atomic_base<Base>&) [with Base = double]’:
/devel/pinocchio/unittest/cppadcg-basic.cpp:227:1:   required from here
/local/users/cmastall/include/cppad/cg/model/functor_generic_model.hpp:113:17: error: ‘class CppAD::atomic_base<double>’ has no member named ‘afun_name’
                 (atomic, atomic.afun_name());
                 ^
In file included from /local/users/cmastall/include/cppad/cg/cppadcg.hpp:107:0,
                 from /local/users/cmastall/include/cppad/cg.hpp:18,
                 from /local/users/cmastall/include/cppad/cg/support/cppadcg_eigen.hpp:18,
                 from /devel/pinocchio/build/include/pinocchio/codegen/cppadcg.hpp:11,
                 from /devel/pinocchio/unittest/cppadcg-basic.cpp:5:
/local/users/cmastall/include/cppad/cg/operation_stack.hpp: At global scope:
/local/users/cmastall/include/cppad/cg/operation_stack.hpp:178:13: error: ‘void CppAD::cg::depthFirstGraphNavigation(CppAD::cg::OperationNode<Base>&, std::size_t, FunctionAnalysis&, FunctionPostProcess&, bool) [with Base = double; FunctionAnalysis = CppAD::cg::CodeHandler<Base>::checkVariableCreation(CppAD::cg::CodeHandler<Base>::Node&) [with Base = double; CppAD::cg::CodeHandler<Base>::Node = CppAD::cg::OperationNode<double>]::<lambda(CppAD::cg::OperationStackData<double>&, CppAD::cg::OperationStack<double>&)>; FunctionPostProcess = CppAD::cg::CodeHandler<Base>::checkVariableCreation(CppAD::cg::CodeHandler<Base>::Node&) [with Base = double; CppAD::cg::CodeHandler<Base>::Node = CppAD::cg::OperationNode<double>]::<lambda(CppAD::cg::OperationStackData<double>&)>; std::size_t = long unsigned int]’, declared using local type ‘CppAD::cg::CodeHandler<Base>::checkVariableCreation(CppAD::cg::CodeHandler<Base>::Node&) [with Base = double; CppAD::cg::CodeHandler<Base>::Node = CppAD::cg::OperationNode<double>]::<lambda(CppAD::cg::OperationStackData<double>&, CppAD::cg::OperationStack<double>&)>’, is used but never defined [-fpermissive]
 inline void depthFirstGraphNavigation(OperationNode<Base>& root,
             ^
unittest/CMakeFiles/cppadcg-basic-test.dir/build.make:65: recipe for target 'unittest/CMakeFiles/cppadcg-basic-test.dir/cppadcg-basic.cpp.o' failed
```

How can I fix this?

cmastalli, 2019-10-01:
Thanks for the feedback. I'll close the issue

cmastalli, 2019-10-02:
it compiles the unittest with the following setup:
  - CppAD: v2019200.5 (latest release in github)
  - CppADCoGen: (master branch)

but it doesn't compile the benchmark code.

Additionally, using both master branches, it doesn't compile the unittest and benchmarks.

jcarpent, 2019-10-02:
@cmastalli There are some missing definitions in CppADCodeGen about new features of CppAD. I will try to fix them, but those details are not really related to Pinocchio.

cmastalli, 2019-10-02:
In the meantime, which versions should I use? In particularly interested in running the benchmark and possibly create some extra benchmarks for the contact dynamics.

cmastalli, 2019-10-02:
Another thing -- if I add this flag -DCMAKE_BUILD_TYPE=Release in Pinocchio, then it doesn't compile neither.

jcarpent, 2019-10-02:
I will try to fix those bugs. Unfortunately, I don't remember the correct versions as they are evolving too fast. Sorry @cmastalli

cmastalli, 2019-10-02:
Don't worry, keep me updated since this is very important for me.

jcarpent, 2019-10-02:
Fixed by #900

closed 2019-10-02 by jcarpent

cmastalli, 2019-10-02:
Thank you for this fast action

cmastalli, 2019-10-03:
Hi @jcarpent 

It might be not Pinocchio issue, but probably you know what it's going on.

If I compile Pinocchio without defining any flag in CMAKE_BUILD_TYPE, it works everything. And when I run the benchmark it prints
```bash
(the time score in debug mode is not relevant)
```bash

Then I defined `CMAKE_BUILD_TYPE=Release` but it doesn't compile. This is the message:
```bash
[  1%] Building CXX object unittest/CMakeFiles/cppadcg-basic-test.dir/cppadcg-basic.cpp.o
cd /devel/pinocchio/build/unittest && /usr/bin/c++   -DBOOST_MPL_LIMIT_LIST_SIZE=30 -DBOOST_MPL_LIMIT_VECTOR_SIZE=30 -DPINOCCHIO_URDFDOM_TYPEDEF_SHARED_PTR -DPINOCCHIO_WITH_CPPADCG_SUPPORT -DPINOCCHIO_WITH_CPPAD_SUPPORT -DPINOCCHIO_WITH_URDFDOM -I/devel/pinocchio/build -I/devel/pinocchio/build/include -I/devel/pinocchio/include -isystem /usr/include/python2.7 -isystem /usr/include/eigen3 -isystem /local/users/cmastall/include -isystem /home/cmastall/.local/lib/python2.7/site-packages/numpy/core/include -I/devel/pinocchio/unittest  -pedantic -Wno-long-long -Wall -Wextra -Wcast-align -Wcast-qual -Wformat -Wwrite-strings -Wconversion  -O3 -DNDEBUG    -DBOOST_TEST_DYN_LINK -DBOOST_TEST_MODULE=cppadcg_basicTest -std=gnu++11 -o CMakeFiles/cppadcg-basic-test.dir/cppadcg-basic.cpp.o -c /devel/pinocchio/unittest/cppadcg-basic.cpp
In file included from /usr/include/cppad/cg/cppadcg.hpp:193:0,
                 from /usr/include/cppad/cg.hpp:18,
                 from /usr/include/cppad/cg/support/cppadcg_eigen.hpp:18,
                 from /devel/pinocchio/build/include/pinocchio/codegen/cppadcg.hpp:11,
                 from /devel/pinocchio/unittest/cppadcg-basic.cpp:5:
/usr/include/cppad/cg/lang/c/language_c.hpp: In member function ‘virtual const string& CppAD::cg::LanguageC<Base>::getComparison(CppAD::cg::CGOpCode) const’:
/usr/include/cppad/cg/lang/c/language_c.hpp:2125:9: error: expected primary-expression before ‘}’ token
```

Any advice is welcome.

cmastalli, 2019-10-03:
There is bug in CppAGCodeGen. My PR solves it: https://github.com/joaoleal/CppADCodeGen/pull/26