stack-of-tasks/pinocchio #898: Error parsing urdf

proyan, 2019-10-01:
When parsing a robot through robotwrapper (BuildFromURDF function), I get the following error:
```
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/local/rbudhira/devel/install/sot/lib/python2.7/site-packages/example_robot_data/__main__.py", line 13, in <module>
    hyq = robots_loader.loadHyQ()
  File "/local/rbudhira/devel/install/sot/lib/python2.7/site-packages/example_robot_data/robots_loader.py", line 137, in loadHyQ
    robot = RobotWrapper.BuildFromURDF(modelPath + URDF_SUBPATH, [modelPath], pinocchio.JointModelFreeFlyer())
  File "/local/rbudhira/devel/install/sot/lib/python2.7/site-packages/pinocchio/robot_wrapper.py", line 17, in BuildFromURDF
    robot.initFromURDF(filename, package_dirs, root_joint, verbose, meshLoader)
  File "/local/rbudhira/devel/install/sot/lib/python2.7/site-packages/pinocchio/robot_wrapper.py", line 21, in initFromURDF
    model, collision_model, visual_model = buildModelsFromUrdf(filename, package_dirs, root_joint, verbose, meshLoader)
  File "/local/rbudhira/devel/install/sot/lib/python2.7/site-packages/pinocchio/shortcuts.py", line 42, in buildModelsFromUrdf
    geom_model = pin.buildGeomFromUrdf(model, filename, package_dirs, geometry_type)
ValueError: Unhandled bouding volume type.
```
I tried the urdf of hyq (with collada dae meshes), and of talos (with stl meshes). Both give the same error. I am on the current devel branch of hpp-fcl, and pinocchio.
Any clues on what is happening would be welcome.
Thanks.

jcarpent, 2019-10-01:
Could you provide a standalone example with robot examples and scripts?

jmirabel, 2019-10-01:
I think the library your executable is linked with at run-time is not the one generated from the source found at compile time.

In other words:
- check your LD_LIBRARY_PATH and PKG_CONFIG_PATH
- check that who gets loaded with ldd
- eventually, remove robotpkg from your environment.

jcarpent, 2019-10-01:
@proyan Apparently, robotpkg forced Pinocchio to be compiled with C++11 flags on. It might the origin of your issue.

proyan, 2019-10-01:
Thanks Joseph. I checked the installation paths of pinocchio, hpp-fcl, and urdf.
I don't have separate installations. And I am installing by source from the devel branches.

Since I'm loading from robotwrapper, I'm not sure which executable would be useful here?

jmirabel, 2019-10-01:
You should check all the so files that you find in folder $PYTHONPATH/pinocchio

proyan, 2019-10-01:
I found the problem. It seems that we changed python installations between site-packages and dist-packages. I'm surprised that I missed this change.

Thanks @jmirabel and @jcarpent. I'll close this.

closed 2019-10-01 by proyan