stack-of-tasks/pinocchio #904: Using wrapper.frameJacobian with reference frame

jviereck, 2019-10-05:
Hi folks,

I am confused by what's the best way to get jacobians and calling the compute functions. Especially, I am interested in getting the LOCAL_WORLD_ALIGNED jacobian.

One way to do this is

```
pin_robot.computeJointJacobians(q)
pin_robot.framesForwardKinematics(q)
jac = pin_robot.getFrameJacobian(frame_idx, pin.ReferenceFrame.LOCAL_WORLD_ALIGNED)
```

If I want to avoid the call to `pin_robot.computeJointJacobians(q)`, the other method is

```
pin_robot.framesForwardKinematics(q)
jac = pin_robot.frameJacobian(frame_idx, pin.ReferenceFrame.LOCAL_WORLD_ALIGNED)
```

However, in the later [1], passing the reference frame to `frameJacobian` is depricated.

Is there another (non-depricated) method on the robot wrapper that allows to get the LOCAL_WORLD_ALIGNED jacobian without calling `computeJointJacobians`?

[1]: https://github.com/stack-of-tasks/pinocchio/blob/master/bindings/python/pinocchio/robot_wrapper.py#L198

jcarpent, 2019-10-05:
Dear @jviereck,

As soon as you are dealing with WORLD or LOCAL_WORLD_ALIGNED reference frames, you have to know the world placement of the bodies (data.oMi). And then, it appears that you need to perform a forwardPass over the joints. computeJointJacobians is doing that in a very effective way (both computing the forwardKinematics and the Jacobian of all the joints).

Then, the way you proceed is almost the most optimal way of performing it.

You can replace `pin_robot.framesForwardKinematics(q)` by `pin.updateFramePlacements(robot.model,robot.data)`or simply `pin.updateFramePlacement(robot.model,robot.data,frame_idx)` for a single frame.

jcarpent, 2020-01-26:
Solved by #1029.

closed 2020-01-26 by jcarpent