stack-of-tasks/pinocchio #906: python impulse model

ddliugit, 2019-10-08:
I am working on a project of impulse model, how to use impulse model in python with Pinocchio?

jcarpent, 2019-10-09:
Dear @ddliugit,

Without more details, it is hard for Pinocchio's support team to provide guidance.
You may have a look to the tutorials of Pinocchio which seems to a good starting point: https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/md_doc_d-practical-exercises_1-directgeom.html

ddliugit, 2019-10-09:
I thought you implemented the model in rigid body dynamics algorithms. Where can I find the relevant Pinocchio code in that section?

jcarpent, 2019-10-09:
You can find the relevant details here: https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/namespacepinocchio.html#a3b3ac8f8d2b0bb7ec4aeea811d5d1800

The code in Python follows the same signature as in C++.

ddliugit, 2019-10-09:
Thanks. Really appreciate your help.

closed 2019-10-09 by ddliugit

jcarpent, 2019-10-09:
@ddliugit Your welcome.

gabrielebndn, 2019-10-09:
> The code in Python follows the same signature as in C++.

@jcarpent not exactly. For the curent version, v2.1.8, Python is using the convention of setting `q=None` in order to employ the variant with no arguments. , which was removed and replaced by a signature more similar to C++ by #902 (a non-backward-compatible change, by choice).

If you want to avoid that some people start employing the removed signature, I suggest a new release

ddliugit, 2019-10-09:
@jcarpent can you help release a new version for install on Mac? I need q for calculation.

jcarpent, 2019-10-09:
Could do that indeed.

jcarpent, 2019-10-09:
@ddliugit How do you install it on OSX?

ddliugit, 2019-10-09:
brew install pinocchio

ddliugit, 2019-10-10:
can you update homebrew formula to link to new release

current on Mac, it is still 2. 1. 8

jcarpent, 2019-10-10:
Done!

jcarpent, 2019-10-10:
It should work already.
You just need to do `brew install pinocchio --build-from-source`

ddliugit, 2019-10-10:
That's good news!
I will test it after installing Catalina.