stack-of-tasks/pinocchio #909: Safety checks in Python

gabrielebndn, 2019-10-16:
Currently, safety checks like checking the length of the array or verifying that the `Data` object is consistent with the `Model` are only done through asserts, i.e. only in debug mode.

This leaves the users completely on their own when using the Release mode, but it is consistent with what Eigen does, and I think it should stay this way: if users want, they can still compile in Debug mode.

However, users are left completely powerless in Python, if they are using the released versions and they are not compiling it themselves.
One solution would be to always enable the checks in Python, but I believe the best way is to add a global switch allowing to enable or disable the checks at will, something like `pinocchio.setAssertionsEnabled(bool)`. This might also be useful in the future if we want to implement a shared library.

If we decide to go this way, then all safety `assert` statements should be replaced by a macro, defined as something like this:
```
#ifdef PINOCCHIO_RUNTIME_ASSERT
#  define PINOCCHIO_ASSERT(cond,msg) \
if(pinocchio::isAssertionsEnabled && !(cond)) { \
  throw runtime_error(msg); \
}
#else
#  define PINOCCHIO_ASSERT(cond,msg) assert((cond) && msg)
#endif 
```

with `pinocchio::isAssertionsEnabled` initially set to false, plus some interface to properly set it.

The main motivation for this is Python, but it would be available in C++ too. Python would be compiled by default with `PINOCCHIO_RUNTIME_ASSERT` defined.

What do you think?

gabrielebndn, 2019-10-16:
Oh, I see Python actually does perform the checks! My bad, I did not remember that, I should have verified before. Still, what do you think of setting a switch?

gabrielebndn, 2019-10-16:
Also, the checks are performed, but since they are made through asserts, this actually crashes the whole program and it quits the Python interpreter. I think this is not a desirable behavior in Python. Exception would be better, in my opinion.

gabrielebndn, 2019-10-16:
> Oh, I see Python actually does perform the checks! My bad, I did not remember that, I should have verified before. Still, what do you think of setting a switch?

Actually, this is what happens: with the apt-downloaded version (v2.1.8), it seems the asserts are enabled, but with the manually compiled version (v2.1.9, but I do not think it changed) they are not (`NDEBUG` is defined).
I am I getting it right? What are the Python compilation options for the apt version? @nim65s, do you know anything about it? If I am getting it right, I find it confusing. The manually-compiled Release versions should be identical to the one actually release in apt, I think

wxmerkt, 2019-10-16:
> The manually-compiled Release versions

`assert` is, by default, deactivated (undefined) when you configure in any mode other than `CMAKE_BUILD_TYPE=Debug`. I think `-NDEBUG` or including `<cassert>` can be a workaround.

gabrielebndn, 2019-10-16:
This test
```
import pinocchio as pin

model = pin.buildSampleModelManipulator()
data = model.createData()
q = pin.utils.zero(model.nq+1)

pin.forwardKinematics(model,data,q)
```
Does not give me any error when I use the manually-compiled version. This means the assertions are disabled and `NDEBUG` is define. I can see from the compiler output that `DNDEBUG` is indeed part of the compilation command.
But, when I run it with the apt-installed version, it fails with the following error
```
python: /opt/openrobots/include/pinocchio/algorithm/kinematics.hxx:108: void pinocchio::forwardKinematics(const pinocchio::ModelTpl<Scalar, Options, JointCollectionTpl>&, pinocchio::DataTpl<Scalar, Options, JointCollectionTpl>&, const Eigen::MatrixBase<Mat>&) [with Scalar = double; int Options = 0; JointCollectionTpl = pinocchio::JointCollectionDefaultTpl; ConfigVectorType = Eigen::Matrix<double, -1, 1>]: Assertion `q.size() == model.nq && "The configuration vector is not of right size"' failed.
Aborted (core dumped)
```
then the Python interpreter crashes with no possibility of recovery. This means the assertions are enabled and `NDEBUG` is *not* defined. I see two problems here:
1) The apt-released version should not be different from the default `Release` manually-compiled version, I think
2) The fact that the whole interpreter crashes is bad, in my opinion. This should be handled through exceptions, at least in Python

Also, I think it is nice to allow users to choose whether they want the checks or not

jcarpent, 2019-10-16:
I think the best usage is to expose the checker in Python and let the user call them if they need it. This is a safe strategy that would not interfere with the current scheme.

gabrielebndn, 2019-10-16:
> I think the best usage is to expose the checker in Python and let the user call them if they need it. This is a safe strategy that would not interfere with the current scheme.

What do you mean with "expose the checker"? `model.check(data)` is already exposed, if that's what you mean. This calls `checkData(model,data)`, which is not exposed, but still the instance method is just the same.
Also, this is not the only thing one might want to check: the most common errors concern the length of the input vector.
I do not think it can get quite cumbersome if users are forced to manually set up the checks on their own, and it is not user-friendly to tell them "you are on your own, do your homework", especially in Python

cmastalli, 2019-10-16:
I would like to draw your attention in this nice assert library:
https://github.com/gpakosz/PPK_ASSERT
where at runtime you're able to ignore asserts.

cmastalli, 2019-10-16:
@gabrielebndn we could also add Python wrappers of Python bindings where:
 1. we could add Python assertions
 2. improve the automatic description of boost python which shows c++ objects (I personally hate it :)

This is very easy job, but it requires a lot of very boring work to wrap each function.

gabrielebndn, 2019-10-16:
> I would like to draw your attention in this nice assert library:
> https://github.com/gpakosz/PPK_ASSERT
> where at runtime you're able to ignore asserts.

I am looking at it. It seems nice

gabrielebndn, 2019-10-16:
> @gabrielebndn we could also add Python wrappers of Python bindings where:
> 
> 1. we could add Python assertions
> 2. improve the automatic description of boost python which shows c++ objects (I personally hate it :)
> 
> This is very easy job, but it requires a lot of very boring work to wrap each function.

Not only this would be long and cumbersome, it would also be harder to maintain. I would personally prefer to replace all `assert` with a configurable `PINOCCHIO_ASSERT` macro, which can be done very quickly, and is cleaner.
Plus, I like the automatic boost-python description :)

jcarpent, 2019-10-16:
One solution is would be to redefine the `PINOCCHIO_ASSERT` in the top header file of Python bindings. For that, we just need to do something like:
```
#ifndef PINOCCHIO_ASSERT
  #define PINOCCHIO_ASSERT assert((cond) && msg)
#endif
```
This will allow more generic features than the one you suggested before. @
What do you think?

wxmerkt, 2019-10-16:
> The fact that the whole interpreter crashes is bad, in my opinion. This should be handled through exceptions, at least in Python

I agree and it's particularly annoying in Jupyter notebooks. `assert` crashes Python (and I expect, by default, them not to be active when compiling in Release mode). If you want to be sure it's checked, perhaps an if statement (inexpensive) that throws an exception (expensive, but then otherwise you'll crash and can't catch), which is passed through the binding to Python, may be a better option?

jcarpent, 2019-10-16:
> I agree and it's particularly annoying in Jupyter notebooks. `assert` crashes Python (and I expect, by default, them not to be active when compiling in Release mode)

Indeed, assert should not appear in Release mode.

cmastalli, 2019-10-16:
With the header library that I shared, you can at runtime avoid a crash of an assert. This is 
because sometime debugging things is convenient to ignore an assert

wxmerkt, 2019-10-16:
Would `if (q.size() != nq_) throw std::runtime_error("Wrong size");` be inconvenient to replace `assert`? Is the cost of the `if` too much to add?

jcarpent, 2019-10-16:
It is not a good idea to add throwing inside efficient algorithms, it would break the predictability. This should have be done with a checker function defined for each algorithm I would suggest.

cmastalli, 2019-10-16:
It is also still possible to include the throw runtime only when we compile the Python bindings. However, it looks a bit dirty work.

I'm still not sure is this topic is important? My opinion is "if an user wants to debug its code needs to compile Pinocchio in Debug mode and its bindings".

I like pretty much plain things where users are exposed to every aspect of the development. So, a solution could be just to add a tutorial that explains how to troubleshoot your algorithm.

jmirabel, 2019-10-16:
> One solution is would be to redefine the `PINOCCHIO_ASSERT` in the top header file of Python bindings.

Wouldn't you break ABI compatibility in a case like : 
- A: a C++ only library using Pinocchio,
- A-py: Python bindings of A
- In Python, import pinocchio and A-py...

jcarpent, 2019-10-16:
I don't think so. Why do you think so? Any hints?

jmirabel, 2019-10-16:
You have no way of knowing which `pinocchio::algo` implementation will be found (in pinocchio Python bindings or in libA.so).

It shouldn't be hard to make it not break the ABI compatibility but one should be careful with such things.

wxmerkt, 2019-10-16:
If the requirement to debug your algorithm is to compile every library in Debug mode locally, the efficiency of using dependencies as binaries goes away. E.g., when I develop Crocoddyl I want to be able to install Pinocchio binaries. If I make a mistake passing the wrong vector to say `pinocchio::forwardKinematics` (e.g., state instead of configuration), I want to see a throw, realise my mistake, and fix it.

> It is not a good idea to add throwing inside efficient algorithms, it would break the predictability. This should have to be done with a checker function defined for each algorithm I would suggest.

I disagree. You will only have it throwing during development when you may get sizes wrong (and the results are garbage if you pass wrong sized vectors). For real deployment, it is just an if-check that won't throw (unless things at runtime change - and you do want to know that!). You can handle an exception, but you can't happen a segmentation fault.

jcarpent, 2019-10-16:
> I disagree. You will only have it throwing during development when you may get sizes wrong (and the results are garbage if you pass wrong sized vectors). For real deployment, it is just an if-check [...]

Sorry @wxmerkt, I do not really understand your thoughts. Can you provide more details or an example to help my understanding?

jcarpent, 2019-10-16:
I think the argument of Microsoft about not affecting the performances might be incorrect when algorithms are working at the micro-second scales like ours. I will implement a quick bench to try it when I have some free times, before doing the complete change.

jcarpent, 2019-10-16:
In all cases, I'm not against at all throwing in Python, as it might help the beginners and the development processes such as Crocoddyl or other libraries based on Pinocchio. So, we should think about such a solution.

jcarpent, 2019-10-16:
@wxmerkt Following your suggestions, it should be only done for the algorithms folder. We might have some issues in other places (like JointComposite) but this is not at the top priority.
@gabrielebndn Your solution seems to be one good starting point, except for the global variable which should be handled with a static variable inside a function, in similar it is done in Eigen for `set_is_malloc_allowed`.

jcarpent, 2019-10-16:
This is mandatory if you have several projects using Pinocchio. You did not want each one to define its own global variable, but rather to call it via a function containing a static variable. This is very similar to a singleton indeed, but simpler in the current context.

jcarpent, 2019-10-16:
No, I don't think so.

jcarpent, 2019-10-16:
In addition, and to simplify the current API in terms of Macro management of Pinocchio, I would be in favor of:
- keeping `PINOCCHIO_ASSERT` as it is done now
- adding `PINOCCHIO_THROWING_ASSERT` which will implement the throwing in Release mode and use the classic PINOCCHIO_ASSERT in the Debug mode.

gabrielebndn, 2019-10-16:
> I'm still not sure if this topic is important??? My opinion is "if an user wants to debug its code, then he/she needs to know that Pinocchio has to be compile in Debug mode (as every Christian in the planet)".

But users would *not* be able to choose Release/Debug mode if they are using the bindings from apt. This would force you to either *always* have the checks or *never* have them (and force them to compile Pinocchio from source if they want them, in this last case).

gabrielebndn, 2019-10-16:
@jcarpent, sorry, I am not following you.

> * keeping `PINOCCHIO_ASSERT` as it is done now

We do not have a macro called `PINOCCHIO_ASSERT` at the moment. We have `PINOCCHIO_ASSERT_MATRIX_SPECIFIC_SIZE`
```
#define PINOCCHIO_ASSERT_MATRIX_SPECIFIC_SIZE(type,M,nrows,ncols)              \
  EIGEN_STATIC_ASSERT(   (type::RowsAtCompileTime == Eigen::Dynamic || type::RowsAtCompileTime == nrows) \
                      && (type::ColsAtCompileTime == Eigen::Dynamic || type::ColsAtCompileTime == ncols),\
                      THIS_METHOD_IS_ONLY_FOR_MATRICES_OF_A_SPECIFIC_SIZE);    \
  assert(M.rows()==nrows && M.cols()==ncols);
```
which would not change a lot and is not systematically used (most algorithms use plain `assert`). And `PINOCCHIO_STATIC_ASSERT`
```
#define PINOCCHIO_STATIC_ASSERT(condition,msg)                                 \
  { int msg[(condition) ? 1 : -1]; /*avoid unused-variable warning*/ (void) msg; }
```

> * adding `PINOCCHIO_THROWING_ASSERT` which will implement the throwing in Release mode and use the classic PINOCCHIO_ASSERT in the Debug mode.

I really do not understand this suggestion. I thought you were against the checks in Release mode, and against exceptions in C++... And what about Python?

jcarpent, 2019-10-16:
I mean by `PINOCCHIO_ASSERT` which can be a static assert or something that the user may want to overload for personal reasons. So, using a Macro is well-defined way.
On the other side, defining `PINOCCHIO_THROWING_ASSERT` will enable us to optionally throw for certain quantities like the vector dimensions mentioned by @wxmerkt.

gabrielebndn, 2019-10-16:
> I mean by `PINOCCHIO_ASSERT` which can be a static assert or something that the user may want to overload for personal reasons. So, using a Macro is a well-defined way.
> On the other side, defining `PINOCCHIO_THROWING_ASSERT` will enable us to optionally throw for certain quantities like the vector dimensions mentioned by @wxmerkt.

I really don't see the point of having exceptions in `Release` mode and assertions in `Debug` mode. If you have exceptions in `Release` mode, I do not see the point of turning them to assertions in `Debug` mode, we might as well keep the assertions, it would get away with all complications and be more consistent. Besides, you seem to suggest that the checks will be performed in `Release` mode, which I thought you wanted to avoid.

wxmerkt, 2019-10-16:
Extra macros and definitions add complexity. 

I agree that assert as currently is the right way to go for internal quantities such as models that affect Pinocchio developers. 

For public facing methods/functions, is there an argument _against_ the `if (input.size() != member_) throw std::runtime_exception("wrong size");`. I would not define a macro and I would not switch between Debug and Release - simplicity wins. The `if` is _negligible_ if input is correct and there won't be an exception. If there is an issue with dependent code, it will be graceful.

wxmerkt, 2019-10-16:
One further argument regarding **efficiency** I'd like to add:

The robotpkg binaries currently include the `assert` - an `if` should be the same cost or faster. This means that they do perform these checks _for all of them_ at runtime already (plus whatever else is activated by `DEBUG`). On my computer there is a **3x difference in runtime** between me pulling the binaries vs compiling on the machine (for centerOfMass/centerOfMassJacobian). I think adding an if with throwing on user-input vectors/matrices and then compiling without the `DEBUG` flag should be more performant than the currently packaged version.

jcarpent, 2019-10-16:
@gabrielebndn I just want to make the distinction between internal assert as @wxmerkt mentioned and the one that might THROW if the compilation flag is active (which should be by default ON if there is no drastic overhead -- which needs to be checked). But for benchmarking the algorithms in general, I would be in favor of silent them.

@wxmerkt I think here a MACRO will simplify the reading of the code. And if one day we change our minds, it will ease the process of stepping backward.

jcarpent, 2019-10-16:
> One further argument regarding **efficiency** I'd like to add:
>
> The robotpkg binaries currently include the `assert` - an `if` should be the same cost or faster. This means that they do perform [...]

I strictly do not understand why robotpkg is doing this thing in the back of the user. @nim65s Do you have any clue on this issue?

gabrielebndn, 2019-10-16:
> @gabrielebndn I just want to make the distinction between internal assert as @wxmerkt mentioned and the one that might THROW if the compilation flag is active (which should be by default ON if there is no drastic overhead -- which needs to be checked).

But this has nothing to do with `Release` vs `Debug` mode

> But for benchmarking the algorithms in general, I would be in favor of silent them.

Yes. But then you would need to control this behavior by defining an extra macro.

> @wxmerkt I think here a MACRO will simplify the reading of the code. And if one day we change our minds, it will ease the process of stepping backward.

Agree on this: defining `PINOCCHIO_ASSERT` is easy, its use would intuitive and legible, and the macro would be more flexible.

wxmerkt, 2019-10-16:
Perhaps, prior to making a final decision, let's gather numbers on whether the `if` actually adds an overhead. If it turns out to be negligible in benchmarks, there is no need for flags to turn on/off and it can be left on by default.

nim65s, 2019-10-16:
> I strictly do not understand why robotpkg is doing this thing in the back of the user. @nim65s Do you have any clue on this issue?

I have already noticed and reported that `-O3 -DNDEBUG` is stripped from `CXX_COMPILER_FLAGS_NDEBUG` at some point, but we don't understand why yet. I'll look more deeply into this as soon as I'm back from my vacation :)

wxmerkt, 2019-10-16:
# Current devel, `MinSizeRel`

## Results
```
$ ./timings
nq = 36
RNEA = 		12.9434 us
NLE = 		12.883 us
NLE via RNEA = 		12.9513 us
CRBA = 		24.4017 us
CRBA minimal = 		16.9753 us
computeAllTerms = 		51.3335 us
Branch Induced Sparsity Cholesky = 	9.23815 us
Dense Eigen Cholesky = 	9.37849 us
Jacobian = 	4.9477 us
Jacobian Time Variation = 	10.5291 us
COM+Jcom = 	6.76057 us
COM+vCOM+aCOM = 	12.2395 us
Zero Order Kinematics = 	4.29077 us
First Order Kinematics = 	6.80566 us
Second Order Kinematics = 	10.1073 us
CCRBA = 	10.1268 us
ABA = 	31.739 us
Empty Forward Pass = 	0.2308 us
Coriolis Matrix = 	56.6405 us
Minv = 	48.1577 us
--
```

# Current devel, assert redefined as if(), `MinSizeRel`

In `macros.hpp`, this is the change:
```c++
#include <exception>

#undef assert
#define assert(cond) if (!cond) { throw std::runtime_error("sad"); }
```

## Results
```
./timings
nq = 36
RNEA = 		12.6389 us
NLE = 		12.7326 us
NLE via RNEA = 		12.4875 us
CRBA = 		24.3746 us
CRBA minimal = 		16.8823 us
computeAllTerms = 		50.7045 us
Branch Induced Sparsity Cholesky = 	9.12997 us
Dense Eigen Cholesky = 	9.41454 us
Jacobian = 	4.73732 us
Jacobian Time Variation = 	10.5936 us
COM+Jcom = 	6.75134 us
COM+vCOM+aCOM = 	12.2407 us
Zero Order Kinematics = 	4.29914 us
First Order Kinematics = 	6.75247 us
Second Order Kinematics = 	9.79773 us
CCRBA = 	9.74422 us
ABA = 	31.0411 us
Empty Forward Pass = 	0.22624 us
Coriolis Matrix = 	56.5723 us
Minv = 	48.065 us
--
```

jcarpent, 2019-10-16:
Why are you so slow?

jcarpent, 2019-10-16:
By the way, `model.check(data)` should slow your timings a lot. This is not appearing in your results.

gabrielebndn, 2019-10-16:
> I have already noticed and reported that -O3 -DNDEBUG is stripped from CXX_COMPILER_FLAGS_NDEBUG at some point, but we don't understand why yet. 

-O3 is removed too? This is Bad. BTW, I had missed that. Where did you report it? Where can we find a reference for this?

jcarpent, 2019-10-16:
@wxmerkt Could you try this code instead:
```
#undef assert
#define assert(cond) if (cond) { throw std::runtime_error("sad"); }
```
? I just changed the condition account in the `if`.
It seems that this code is not called.

wxmerkt, 2019-10-16:
Thank you for pointing this out @jcarpent. I did a search-replace on the entire workspace to define a new variable and verified it throws the exception when the if condition is flipped:

```cpp
#include <exception>
#include <cassert>

#undef assert
#define TEST_NEW_ASSERT(cond) if (!(cond)) { throw std::runtime_error("sad"); }
```
and preg-replace `assert(` with `TEST_NEW_ASSERT(`.

## `MinSizeRel`, with if() on everything including `model.check(data)`
```
./benchmark/timings
nq = 36
RNEA = 		13.4901 us
NLE = 		13.5249 us
NLE via RNEA = 		13.2356 us
CRBA = 		27.9515 us
CRBA minimal = 		20.907 us
computeAllTerms = 		56.247 us
Branch Induced Sparsity Cholesky = 	10.1582 us
Dense Eigen Cholesky = 	9.31772 us
Jacobian = 	5.74733 us
Jacobian Time Variation = 	11.8108 us
COM+Jcom = 	8.60799 us
COM+vCOM+aCOM = 	14.4325 us
Zero Order Kinematics = 	5.19017 us
First Order Kinematics = 	7.94079 us
Second Order Kinematics = 	11.0773 us
CCRBA = 	14.8684 us
ABA = 	33.0429 us
Empty Forward Pass = 	1.122 us
Coriolis Matrix = 	60.4655 us
Minv = 	50.1638 us
--
```

## `MinSizeRel`, with all `model.check(data)` removed - the rest with the above if
```
./benchmark/timings  
nq = 36
RNEA = 		12.4239 us
NLE = 		12.6691 us
NLE via RNEA = 		12.252 us
CRBA = 		27.2835 us
CRBA minimal = 		19.8913 us
computeAllTerms = 		53.505 us
Branch Induced Sparsity Cholesky = 	9.05657 us
Dense Eigen Cholesky = 	8.94269 us
Jacobian = 	4.62113 us
Jacobian Time Variation = 	10.511 us
COM+Jcom = 	6.64835 us
COM+vCOM+aCOM = 	12.1417 us
Zero Order Kinematics = 	4.2244 us
First Order Kinematics = 	6.6836 us
Second Order Kinematics = 	9.71632 us
CCRBA = 	12.886 us
ABA = 	31.1002 us
Empty Forward Pass = 	0.22336 us
Coriolis Matrix = 	59.0896 us
Minv = 	47.2666 us
--
```

## `MinSizeRel`, with condition defined as `void(0);`
```
./benchmark/timings
nq = 36
RNEA = 		12.7799 us
NLE = 		12.9566 us
NLE via RNEA = 		12.3491 us
CRBA = 		25.0207 us
CRBA minimal = 		17.0282 us
computeAllTerms = 		52.3047 us
Branch Induced Sparsity Cholesky = 	9.39966 us
Dense Eigen Cholesky = 	9.34251 us
Jacobian = 	4.70885 us
Jacobian Time Variation = 	10.7586 us
COM+Jcom = 	6.66024 us
COM+vCOM+aCOM = 	12.4961 us
Zero Order Kinematics = 	4.22962 us
First Order Kinematics = 	6.80551 us
Second Order Kinematics = 	10.078 us
CCRBA = 	10.3 us
ABA = 	31.5576 us
Empty Forward Pass = 	0.22503 us
Coriolis Matrix = 	56.1108 us
Minv = 	48.3797 us
--
```

# Summary

- *All if-checks:* RNEA gets a 5.5% slow-down if a check is carried out on all terms, including `model.check(data)`
- *All but model-data-checks with if:* RNEA is 2.8% faster _with the if_. Minv is 2.3% faster with the if. In other words, there is no performance penalty for always using the if statement with exception throwing on everything apart from the `model.check(data)`, which is good to keep as `assert`.

wxmerkt, 2019-10-16:
Cf. https://github.com/stack-of-tasks/pinocchio/pull/910 for the changes used in the benchmark above. Please take a look and re-verify in your machine. For us, we did not see a penalty - counterintuitively, it was even slightly faster with the `if` on at all times than with it being `void(0);`.

olivier-stasse, 2019-10-16:
Another solution is to package dbg version of the library. The other advantage of this is that you have the export of the symbols for debugging.

olivier-stasse, 2019-10-16:
Sorry for maybe just adding noise, but after reading this thread I looked at the two following files:
```
/opt/openrobots/share/doc/debian/pinocchio/+BUILD_INFO
/opt/openrobots/share/doc/debian/py27-pinocchio/+BUILD_INFO
```
it appears that the python wrapper is compiled with the std=c++11 flags, which I do believe is not correct.

Moreover I found out moreover this is probably due to the inclusion of the following file in the robotpkg Makefile of py-pinocchio
```
mk/language/c++11.mk
```
The corresponding line
https://github.com/nim65s/robotpkg/blob/dd06149ffec84fc5f656a7f184f8b9ef721bc4ed/mk/language/c%2B%2B11.mk#L14
is replacing the compile option CXX_COMPILER_FLAGS_NDEBUG by -std=c++11 if no alternative compiler to g++ is provided.
Removing this line from robotpkg should fix the matter.

jcarpent, 2019-10-16:
Dear @olivier-stasse,

I also found this weird C++11 forcing flag some weeks ago. It was added by maintainers of robotpkg for compatibility issues with recent versions of URDFDOM which are forcing the compatibility with C++11 sadly.
Normally and @nim65s may confirm, this flag should be removed soon from robotpkg. I just hope nobody will suffer from it.

olivier-stasse, 2019-10-17:
I did some testing.
1. Switching 
```
include ../../mk/language/c++11.mk
```
to 
```
include ../../mk/language/c++.mk
```
do fix the matter, i.e. we have -O3 -DNDEBUG compilation
2. Making the switch and adding the following lines from the c++11.mk makes the problem reappearing:
```
CXX_COMPILER_FLAGS_DEBUG+= \
  $(if $(filter g++ ccache-g++,${PKG_ALTERNATIVE.c++-compiler}),-std=c++11)
```

olivier-stasse, 2019-10-17:
And I will stop here as it is not related with the initial topic of this thread.

jcarpent, 2019-10-21:
Solved by #910. Thanks to @wxmerkt for this nice and useful contribution.

closed 2019-10-21 by jcarpent