stack-of-tasks/pinocchio #912: Model without mesh error

ddliugit, 2019-10-19:
I want to do a small experiment for a simple Urdf file without mesh etc.

see this Urdf file with only one bar:

<?xml version="1.0"?>
<robot name="model">
   <link name="base_link"/>
   <joint name="base_joint" type="fixed">
    <parent link="base_link"/>
    <child link="link1"/>
    <origin rpy="0 0 0" xyz="0 0 0"/>
  </joint>
   <link name="link1">
    <collision>
      <origin rpy="0 0 0" xyz="0 0 1.0"/>
      <geometry>
        <box size="0.1 0.1 2"/>
      </geometry>
    </collision>
    <visual>
      <origin rpy="0 0 0" xyz="0 0 1.0"/>
      <geometry>
        <box size="0.1 0.1 2"/>
      </geometry>
      <material name="orange"/>
    </visual>
    <inertial>
      <origin rpy="0 0 0" xyz="0 0 1.0"/>
      <mass value="1"/>
      <inertia ixx="1.0" ixy="0.0" ixz="0.0" iyy="1.0" iyz="0.0" izz="1.0"/>
    </inertial>
  </link>
</robot>

However, Pinocchio doesn't work .

the error is like the following:
ArgumentError: Python argument types in
    pinocchio.libpinocchio_pywrap.buildGeomFromUrdf(Model, str, JointModelFreeFlyer, GeometryType)
did not match C++ signature:
    buildGeomFromUrdf(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> Model to associate the Geometry, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > URDF filename (string), pinocchio::GeometryType Geometry type (COLLISION or VISUAL), boost::shared_ptr<hpp::fcl::MeshLoader> Mesh loader)
    buildGeomFromUrdf(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> Model to associate the Geometry, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > URDF filename (string), std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > package_dir (string), pinocchio::GeometryType Geometry type (COLLISION or VISUAL), boost::shared_ptr<hpp::fcl::MeshLoader> Mesh loader)
    buildGeomFromUrdf(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> Model to associate the Geometry, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > URDF filename (string), boost::python::list package_dirs (list of strings), pinocchio::GeometryType Geometry type (COLLISION or VISUAL), boost::shared_ptr<hpp::fcl::MeshLoader> Mesh loader)
    buildGeomFromUrdf(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> Model to associate the Geometry, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > URDF filename (string), std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > package_dirs (vector of strings), pinocchio::GeometryType Geometry type (COLLISION or VISUAL), boost::shared_ptr<hpp::fcl::MeshLoader> Mesh loader)
    buildGeomFromUrdf(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> Model to associate the Geometry, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > URDF filename (string), std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > package_dir (string), pinocchio::GeometryType Geometry type (COLLISION or VISUAL))
    buildGeomFromUrdf(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> Model to associate the Geometry, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > URDF filename (string), pinocchio::GeometryType Geometry type (COLLISION or VISUAL))
    buildGeomFromUrdf(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> Model to associate the Geometry, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > URDF filename (string), boost::python::list package_dirs (list of strings), pinocchio::GeometryType Geometry type (COLLISION or VISUAL))
    buildGeomFromUrdf(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> Model to associate the Geometry, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > URDF filename (string), std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > package_dirs (vector of strings), pinocchio::GeometryType Geometry type (COLLISION or VISUAL))

gabrielebndn, 2019-10-19:
May be related to #726. In short, right now Pinocchio's routines are only able to show meshes. This should be fixed by enhancing GepettoVisualizer https://github.com/stack-of-tasks/pinocchio/blob/master/bindings/python/pinocchio/visualize/gepetto_visualizer.py

gabrielebndn, 2019-10-22:
So, this requires python bindings for [hpp-fcl](https://github.com/humanoid-path-planner/hpp-fcl).
I was about to make them myself inside Pinocchio, but @jmirabel told me that they are being made inside hpp-fcl itself: they are basically ready on the [refactoring](https://github.com/humanoid-path-planner/hpp-fcl/tree/refactoring) branch, which should be merged in a matter of weeks, after the necessary compatibility checks.
This being the case, it would be wasteful and counter-productive to develop the bindings in Pinocchio. Therefore, we will wait for a new release of hpp-fcl.

ddliugit, 2019-10-22:
OK, I see. Thanks.

gabrielebndn, 2019-10-22:
@ddliugit, I think we can keep this issue open until it is solved

ddliugit, 2019-10-22:
OK, that's great.

gabrielebndn, 2020-01-08:
Solved by #1004

closed 2020-01-08 by gabrielebndn